inkcut/inkcut #173: Issues with odd 0,0 location

TrendMend, 2019-06-27:
Sooo, I have a cricut expression with the https://github.com/thetazzbot/FreeExpression custom firmware, and it uses the top right as 0,0 , this affects multiple things which then in sue my rage and rage quitting haha. So First off, I need inkcut to know 0,0 is the top right, and then second, when vieweing the live view it is totally off the mat, whilst the cricut is actually still on the mat, and then third, the scale is horribly off (possibly on my end?).

frmdstryr, 2019-06-27:
Can you try the current master branch? I put in some changes to fix some issues with different plot orientations. `pip3 install git+https://github.com/inkcut/inkcut.git`  

As for the scaling, which protocol is used? DMPL and HPGL apply a scale before sending but others don't so it may need changed.

TrendMend, 2019-06-27:
I will try that out shortly (just at work), I am using HPGL and the funny thing is, the plot function in inkscape (actually what is suggested) has no issues with the scaling. I will send another message when I try out the new release, also apologies for that horrendous message above, it was like 12 am for me haha.

TrendMend, 2019-06-28:
So I did a readover of the readme for the firmware, and it requires a dpi of 400 :P that was what was causing it with inkcut and when I said no issue with inkscape, was because I correctly set the dpi on my desktop, and not my laptop... so thats one thing resolved

closed 2019-07-04 by TrendMend