inkcut/inkcut #174: No DPI

TrendMend, 2019-06-28:
This ties in with #173 . There is no dpi setting for devices... This is required (unfortunately) for my craft cutter, even though it should have nothing to do with the scale, but with the resolution. Is there a remedy for this?

frmdstryr, 2019-06-28:
You can change the output scaling under devce setup. Inkscape now defaults to 96 DPI so you'll need to set the scale to 400/96.

TrendMend, 2019-06-28:
That is what I did for at the moment (I did 400/1051 because inkscape used to use it at the time of the firmware release) and it worked well enough, but you can only go two decimal places which is not optimal for things needing more precision...

DamienBurkeJJ, 2019-07-02:
Agreed, minimum of 4 decimal places would be better. My cutter, a Pixmax 720, wants somewhere between 0.91 and 0.92 for perfect scaling.

frmdstryr, 2019-07-02:
Just increased the decimal places to 6 https://github.com/inkcut/inkcut/commit/47aa6b99092cb7f574a28d49ee783373c3338464 and also pushed out a release.

Run `pip3 install --upgrade inkcut`