inkcut/inkcut #176: Inkcut - Workspace start up error

DrBwts, 2019-06-28:
Just installed Inkcut & run it from the command line & go this error...

[image]

Running on Win10 x64 installed using pip install inkcut as I couldnt get pip3 to work

frmdstryr, 2019-06-28:
The master branch has fixes for these errors on python 2. Try updating with

`pip install git+https://github.com/inkcut/inkcut.git`

DrBwts, 2019-07-01:
Yes working now altough I can find my device in the device setup, Pixmax 720. Should I open a new thread?

raboof, 2019-07-01:
> I cant find my device in the device setup, Pixmax 720. Should I open a new thread?

Yes I think that would be best!

closed 2019-07-01 by raboof