inkcut/inkcut #177: Error message

Diesel114, 2019-06-28:
If someone out there could help it would be great. I would describe my computer knowledge as basic!

My PC is running Ubuntu 18.04.2 LTS and I am running Inkscape 0.92.4 I have downloaded and installed the Inkcut extension. When I try and open up Inkcut I get a error message as follows...

Traceback (most recent call last):
File "inkcut_open.py"line 26, in <module>
import inkex
ImortError: No module named index

therefor inkcut does not open up, so if some could help this would be very helpful. I have tried looking online for more information but its difficult when you don't always understand what your doing!

kind regards

frmdstryr, 2019-06-30:
The extension must be run from the menu in inkscape.

See https://www.codelv.com/projects/inkcut/docs/installing for instructions on how to install Inkcut

Diesel114, 2019-07-01:
Thanks for response however I am running extension in menu in inkscape as per setup instruction, could offer more advice, thank you

frmdstryr, 2019-07-01:
Interesting you may need to try the inkscape forums then.  The inkex module is apart of inkscape itself so if the error is coming from that module (as it looks) I'm not sure I can help..

Diesel114, 2019-07-02:
i will go over to inkscape forums and ask, thank you

Diesel114, 2019-07-03:
the forum at inkscape have gave some advice and now left with a different error message, what does this all mean????

Traceback (most recent call last):
File "inkcut_open.py", line 66, in <module>
effect.affect()
File "/usr/share/inkscape/extensions/inkex.py", line 283, in affect
self.effect()
File "inkcut_open.py", line 61, in effect
p.stdin.write(inkex.etree.tostring(document))
IOError: [Errno 32] Broken pipe

frmdstryr, 2019-07-03:
Thanks... It most likely means Inkcut is crashing attempting to open the file. A few other's have seen this as well but I've yet to run into it.

Does the same file work when opening directly from Inkcut? Can you attach the Inkcut logs "Help->Logs...". 

Also please update to the latest version with `pip3 install --upgrade inkcut`

Diesel114, 2019-07-04:
Hi I followed instruction above i.e pip3 install --upgrade inkcut

it now gives this error see below.

Traceback (most recent call last):
File "inkcut_open.py", line 66, in <module>
effect.affect()
File "/usr/share/inkscape/extensions/inkex.py", line 283, in affect
self.effect()
File "inkcut_open.py", line 61, in effect
p.stdin.write(inkex.etree.tostring(document))
IOError: [Errno 32] Broken pipe

I have also attached log files.

[log file.zip](https://github.com/inkcut/inkcut/files/3358738/log.file.zip)

frmdstryr, 2019-07-04:
Is there any txt files in `/home/<your-user>/.config/inkcut/logs` ?  Please attach those

Diesel114, 2019-07-04:
will take a look when i get home.

rachelangelo, 2020-02-25:
Same problem here!

crash.txt is empty and inkcut.txt says:

2020-02-05 10:57:14,453 | ERROR   | inkcut | Traceback (most recent call last):
  File "c:\user\MYUSER\appdata\local\programs\python\python38-32\lib\site-packages\inkcut\app.py", line 99, in main
    workbench.run()
  File "c:\user\MYUSER\appdata\local\programs\python\python38-32\lib\site-packages\inkcut\core\workbench.py", line 146, in run
    with enaml.imports():
  File "c:\user\MYUSER\appdata\local\programs\python\python38-32\lib\site-packages\enaml\__init__.py", line 61, in imports
    from enaml.core.import_hooks import imports
  File "c:\user\MYUSER\appdata\local\programs\python\python38-32\lib\site-packages\enaml\core\import_hooks.py", line 24, in <module>
    from .enaml_compiler import EnamlCompiler, COMPILER_VERSION
  File "c:\user\MYUSER\appdata\local\programs\python\python38-32\lib\site-packages\enaml\core\enaml_compiler.py", line 11, in <module>
    from . import compiler_common as cmn
  File "c:\user\MYUSER\appdata\local\programs\python\python38-32\lib\site-packages\enaml\core\compiler_common.py", line 16, in <module>
    from . import byteplay as bp
  File "c:\user\MYUSER\appdata\local\programs\python\python38-32\lib\site-packages\enaml\core\byteplay\__init__.py", line 11, in <module>
    from .wbyteplay import *
  File "c:\user\MYUSER\appdata\local\programs\python\python38-32\lib\site-packages\enaml\core\byteplay\wbyteplay.py", line 92, in <module>
    BREAK_LOOP,
NameError: name 'BREAK_LOOP' is not defined

Any help will be appreciated :)

frmdstryr, 2020-02-25:
@rachelangelo you can either switch to python 3.7 or wait until  the next version of enaml (ui framework) is released

rachelangelo, 2020-02-25:
Thanks for the reply. How can I downgrade python from 3.8 to 3.7? Do I have to uninstall it?

frmdstryr, 2020-02-25:
Yes uninstall and reinstall.  Use 3.7.