inkcut/inkcut #178: Device setup window issue

DrBwts, 2019-07-01:
Win 10 x64  PixMax 720

From the docs its says to select 'Inkcut Generic Driver' if my device isnt listed.

I then selected the interface type, serial, as per the spreadsheet linked from the docs. I have no idea which protocol to use to I've left it on the default HPGL.

But when I click OK the setup window wont close. So I clicked 'x' to close the window then I reopned to see if my settings were still there. They weren't they had reverted to the defaults.

Upshot - nothing getting sent to my cutter.

Thanks for any help

frmdstryr, 2019-07-01:
Sounds like it's having a problem saving the settings.  Can you attach the logs (from Help->Logs)?

DrBwts, 2019-07-01:
OK done, the 'crash.txt' log was empty

[inkcut.txt](https://github.com/inkcut/inkcut/files/3346197/inkcut.txt)

frmdstryr, 2019-07-01:
Thanks for the log. Unfortunately I don't see any errors, it doesn't look like it tried to save any device settings. It looks like it's writing to a file `c:\users\nicbwt~1\appdata\local\temp\`

I think the "OK" button must have been greyed out/disabled. Try to force it to make new settings by switching the driver to a random one, then going back to the generic driver, and see if the OK button enables. If that works, I'll push an update that just removes the disabling of the button.

DrBwts, 2019-07-01:
The OK button works for every other driver except the Inkcut Generic driver

frmdstryr, 2019-07-03:
Ok thanks for confirming. If you update now `pip3 install --upgrade inkcut` the OK button should always be available even if there are no changes.

DrBwts, 2019-07-05:
Hi,

The OK is now working & it allows me to select the USB - COM3 but when I
send the file I get,

..................
................... <long list of commands>
2019-07-05 12:42:44,824 | DEBUG   | inkcut | -> COM3 | M0,0
2019-07-05 12:42:44,826 | DEBUG   | inkcut | device | finish
2019-07-05 12:42:44,827 | DEBUG   | inkcut | device | disconnect
2019-07-05 12:42:44,827 | DEBUG   | inkcut | -- COM3 | closed by request
QSocketNotifier: Invalid socket specified

The cutter is connected to USB - COM3 is this a protocol issue?

Nic

frmdstryr, 2019-07-06:
It sounds like an issue another user reported where Inkcut only sent one command and not the rest on Windows. Unfortunately I don't have a windows PC to debug this.  Could you open the terminal and try this script (simply paste it in).

```python
from serial import Serial
s = Serial('COM3')
cmds = ['IN', 'PU0,0', 'PA100,0', 'PA100,100', 'PA0,100', 'PA0,0']
for c in cmd:
   s.write(c+';')
```

It should move around in a small box.  If that does not work it means it's an issue with the underlying serial port communication library or the commands are incorrect.

DrBwts, 2019-07-06:
had to change s.write(c+';') to s.write((c+';').encode()) but the cutting
head does move a small way left then right

nic

frmdstryr, 2019-07-06:
Great if you increase the number it should move more. 

Does this also work?

```python
import time
dev = workbench.get_plugin('inkcut.device').device
cmds = ['IN', 'PU0,0', 'PA1000,0', 'PA1000,1000', 'PA0,1000', 'PA0,0']
dev.connect()
for c in cmd:
   time.sleep(0.2)
   dev.connection.write(c+';')
time.sleep(0.2)
dev.disconnect()

```

skarzzzz, 2019-12-18:
Sorry to hijack an old post but ive got this problem with the same cutter PIXMAX 720
ive get it setup as "Inkcut Generic Driver" on Serial Port "COM3 - USB" I press send to device and it plots it within the program but my cutter doesnt move at all heres what the commands look like 

2019-12-18 21:50:11,739 | DEBUG   | inkcut | Starting plugin 'inkcut.ui'
2019-12-18 21:50:11,741 | WARNING | inkcut | Plugin inkcut.ui state restored from: C:\Users\Leroy/.config/inkcut/inkcut.ui.json
2019-12-18 21:50:11,914 | DEBUG   | asyncio | Using selector: SelectSelector
2019-12-18 21:50:11,937 | DEBUG   | asyncio | Using selector: SelectSelector
2019-12-18 21:50:21,826 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2019-12-18 21:50:21,826 | INFO    | inkcut | device |
2019-12-18 21:50:21,832 | INFO    | inkcut | {}
2019-12-18 21:50:21,844 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2019-12-18 21:50:37,448 | DEBUG   | inkcut | device | submit <inkcut.job.models.Job object at 0x000001FC2C151F48>
2019-12-18 21:50:37,458 | INFO    | inkcut | device | Initializing job
2019-12-18 21:50:37,458 | DEBUG   | inkcut | device | init <inkcut.job.models.Job object at 0x000001FC2C151F48>
2019-12-18 21:50:37,462 | DEBUG   | inkcut | device | Path length: 9705.350081082
2019-12-18 21:50:41,170 | INFO    | inkcut | device | Job cancelled
2019-12-18 21:51:02,166 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2019-12-18 21:51:02,173 | INFO    | inkcut | {}
2019-12-18 21:51:02,185 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2019-12-18 21:51:04,641 | DEBUG   | inkcut | device | submit <inkcut.job.models.Job object at 0x000001FC2C151F48>
2019-12-18 21:51:04,643 | INFO    | inkcut | device | Initializing job
2019-12-18 21:51:04,644 | DEBUG   | inkcut | device | init <inkcut.job.models.Job object at 0x000001FC2C151F48>
2019-12-18 21:51:04,646 | DEBUG   | inkcut | device | Path length: 9705.350081082
2019-12-18 21:51:05,709 | INFO    | inkcut | device | Connecting to device
2019-12-18 21:51:05,710 | INFO    | inkcut | device | Processing job
2019-12-18 21:51:05,711 | DEBUG   | inkcut | device | connect
2019-12-18 21:51:05,715 | DEBUG   | inkcut | -> COM3 | IN;
2019-12-18 21:51:05,715 | DEBUG   | inkcut | COM3 | opened
2019-12-18 21:51:05,716 | INFO    | inkcut | device | Working...
2019-12-18 21:51:05,717 | DEBUG   | inkcut | -> COM3 | PU9736,819;
2019-12-18 21:51:08,116 | DEBUG   | inkcut | -> COM3 | PD9726,812;
2019-12-18 21:51:08,121 | DEBUG   | inkcut | -> COM3 | PD9724,799;
2019-12-18 21:51:08,126 | DEBUG   | inkcut | -> COM3 | PD9739,756;
2019-12-18 21:51:08,138 | DEBUG   | inkcut | -> COM3 | PD9762,718;
2019-12-18 21:51:08,150 | DEBUG   | inkcut | -> COM3 | PD9792,682;
2019-12-18 21:51:08,163 | DEBUG   | inkcut | -> COM3 | PD9831,651;
2019-12-18 21:51:08,177 | DEBUG   | inkcut | -> COM3 | PD9869,627;
2019-12-18 21:51:08,189 | DEBUG   | inkcut | -> COM3 | PD9908,606;
2019-12-18 21:51:08,201 | DEBUG   | inkcut | -> COM3 | PD9945,444;
2019-12-18 21:51:08,243 | DEBUG   | inkcut | -> COM3 | PD9966,386;
2019-12-18 21:51:08,260 | DEBUG   | inkcut | -> COM3 | PD9990,345;
2019-12-18 21:51:08,273 | DEBUG   | inkcut | -> COM3 | PD10036,296;
2019-12-18 21:51:08,291 | DEBUG   | inkcut | -> COM3 | PD10090,264;
2019-12-18 21:51:08,309 | DEBUG   | inkcut | -> COM3 | PD10136,253;
2019-12-18 21:51:08,322 | DEBUG   | inkcut | -> COM3 | PD10178,251;
2019-12-18 21:51:08,334 | DEBUG   | inkcut | -> COM3 | PD10179,251;
2019-12-18 21:51:08,336 | DEBUG   | inkcut | -> COM3 | PD10223,255;
2019-12-18 21:51:08,348 | DEBUG   | inkcut | -> COM3 | PD10260,268;
2019-12-18 21:51:08,359 | DEBUG   | inkcut | -> COM3 | PD10290,288;
2019-12-18 21:51:08,369 | DEBUG   | inkcut | -> COM3 | PD10312,316;
2019-12-18 21:51:08,379 | DEBUG   | inkcut | -> COM3 | PD10325,352;
2019-12-18 21:51:08,390 | DEBUG   | inkcut | -> COM3 | PD10327,392;
2019-12-18 21:51:08,401 | DEBUG   | inkcut | -> COM3 | PD10318,434;
2019-12-18 21:51:08,413 | DEBUG   | inkcut | -> COM3 | PD10298,476;
2019-12-18 21:51:08,426 | DEBUG   | inkcut | -> COM3 | PD10259,524;
2019-12-18 21:51:08,443 | DEBUG   | inkcut | -> COM3 | PD10209,553;
2019-12-18 21:51:08,459 | DEBUG   | inkcut | -> COM3 | PD10161,562;
2019-12-18 21:51:08,472 | DEBUG   | inkcut | -> COM3 | PD10139,559;
2019-12-18 21:51:08,479 | DEBUG   | inkcut | -> COM3 | PD10120,551;
2019-12-18 21:51:08,486 | DEBUG   | inkcut | -> COM3 | PD10118,550;
2019-12-18 21:51:08,487 | DEBUG   | inkcut | -> COM3 | PD10117,549;
2019-12-18 21:51:08,489 | DEBUG   | inkcut | -> COM3 | PD10071,573;
2019-12-18 21:51:08,501 | DEBUG   | inkcut | -> COM3 | PD10021,595;
2019-12-18 21:51:08,516 | DEBUG   | inkcut | -> COM3 | PD10002,703;
2019-12-18 21:51:08,544 | DEBUG   | inkcut | -> COM3 | PD9985,829;
2019-12-18 21:51:08,578 | DEBUG   | inkcut | -> COM3 | PD9974,943;
2019-12-18 21:51:08,607 | DEBUG   | inkcut | -> COM3 | PD9966,1043;
2019-12-18 21:51:08,633 | DEBUG   | inkcut | -> COM3 | PD10042,860;
2019-12-18 21:51:08,683 | DEBUG   | inkcut | -> COM3 | PD10085,778;
2019-12-18 21:51:08,707 | DEBUG   | inkcut | -> COM3 | PD10130,704;
[... 1832 lines cut ...]
2019-12-18 21:51:35,952 | DEBUG   | inkcut | -> COM3 | PD571,172;
2019-12-18 21:51:35,966 | DEBUG   | inkcut | -> COM3 | PD553,164;
2019-12-18 21:51:35,972 | DEBUG   | inkcut | -> COM3 | PD534,163;
2019-12-18 21:51:35,978 | DEBUG   | inkcut | -> COM3 | PD514,169;
[... cut ...]
2019-12-18 21:51:36,144 | DEBUG   | inkcut | -> COM3 | PD609,271;
2019-12-18 21:51:36,153 | DEBUG   | inkcut | -> COM3 | PD611,242;
2019-12-18 21:51:36,162 | DEBUG   | inkcut | -> COM3 | PD603,214;
2019-12-18 21:51:36,173 | DEBUG   | inkcut | -> COM3 | PU0,0;
2019-12-18 21:51:36,177 | DEBUG   | inkcut | device | finish
2019-12-18 21:51:36,178 | DEBUG   | inkcut | -> COM3 | IN;
2019-12-18 21:51:36,179 | DEBUG   | inkcut | device | disconnect
2019-12-18 21:51:36,179 | DEBUG   | inkcut | -- COM3 | closed by request
QSocketNotifier: Invalid socket specified

RootCNC, 2021-07-11:
Did you solve this? I've got the same issue? any help?

DrBwts, 2021-07-12:
No never got it sorted, so ended up buying Easy Cut Studio :)

RootCNC, 2021-07-12:
That's fair enough!

Tbh, I gave up too. For anyone else reading this thread - I also found that
Inkscape can controller the plotter directly, goto: Extensions > Export >
Plot.

This offers the basic control of the plotter and for most people, should do
the job.