inkcut/inkcut #181: No "device setup" in menu on mac

ErikEklund, 2019-07-17:
I tried inkcut on mac, High Sierra (10.13.6). All dependencies as well as Inkcut installed fine and I run it from the command line, but for some reason the "Set up device" menu option is not there. The only option under the Device menu is "Send to printer".

It defaults to another device (bluetooth serieal) and I can't find a way to access the device settings window to set device, baud speed and all of that. The menu option simply isn't available. My printer is connected through a usb serial adapter (device is called something like "/dev/cu.usbserial-XXXXXX/") and shows up fine in other cutting software, such as Sign Cut.

Any ideas what could cause this would be greatly appreciated.

frmdstryr, 2019-07-17:
https://github.com/inkcut/inkcut/issues/161#issuecomment-491455065

ErikEklund, 2019-07-17:
I checked preferences - it did not have a Device option (only things like units and other system stuff). I'm not at that computer until tomorrow so can't screenshot it currently.

What did work though was to change the hotkey in the source files. The usual Ctrl+Alt+P did not work, so I tried another (Ctrl+Y, which on mac was Cmd+Y), that opened the device window and I was able to cut.. Although with some other issues, but I will open a separate issue for that.

oskarscholander, 2020-01-20:
It is still the same. Can not use the software since I can not change device due to the "Device setup" is gone and can not be found in the "Preferences" either.

hamsterwoede, 2020-02-21:
Exactly the same problem, can't change device.

hamsterwoede, 2020-02-21:
I can confirm that changing the shortcut in devicemanifest.enaml to ctrl+y works.
 (found in (my case) in /usr/local/lib/python3.7/site-packages/inkcut/devicemanifest.enaml)
the replaced piece of code will look something like this.
```
 ActionItem:
            path = '/device/setup'
            label = tr("device", "Setup...")
            shortcut = 'Ctrl+Y'
```

frmdstryr, 2020-02-21:
@hamsterwode, can you see if changing the label to something that doesn't include "Setup", "Config", or "Preferences" makes it show up? OSX seems to think it's ok move menu's around for no reason.

hamsterwoede, 2020-02-21:
Changing the code makes it appear under Device. But if I click it (or use the shortcut), it crashes. 

`label = tr("device", "Stup...")`

> Traceback (most recent call last):
>   File "/usr/local/lib/python3.7/site-packages/enaml/qt/q_deferred_caller.py", line 42, in customEvent
>     event.callback(*event.args, **event.kwargs)
>   File "/usr/local/lib/python3.7/site-packages/enaml/workbench/ui/workbench_menus.enaml", line 24, in invoke_action
>     core.invoke_command(item.command, item.parameters, item)
>   File "/usr/local/lib/python3.7/site-packages/enaml/workbench/core/core_plugin.py", line 78, in invoke_command
>     return command.handler(event)
>   File "/usr/local/lib/python3.7/site-packages/inkcut/device/manifest.enaml", line 27, in device_setup
>     DeviceDialog(ui.window, plugin=plugin).exec_()
>   File "/usr/local/lib/python3.7/site-packages/enaml/core/enamldef_meta.py", line 79, in __call__
>     instance.__init__(parent, **kwargs)
>   File "/usr/local/lib/python3.7/site-packages/enaml/core/object.py", line 100, in __init__
>     self.set_parent(parent)
>   File "/usr/local/lib/python3.7/site-packages/enaml/core/object.py", line 160, in set_parent
>     parent.child_added(self)
>   File "/usr/local/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 185, in child_added
>     child.activate_proxy()
>   File "/usr/local/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 213, in activate_proxy
>     child.activate_proxy()
>   File "/usr/local/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 213, in activate_proxy
>     child.activate_proxy()
>   File "/usr/local/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 213, in activate_proxy
>     child.activate_proxy()
>   [Previous line repeated 6 more times]
>   File "/usr/local/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 210, in activate_proxy
>     self.activate_top_down()
>   File "/usr/local/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 226, in activate_top_down
>     self.proxy.activate_top_down()
>   File "/usr/local/lib/python3.7/site-packages/enaml/qt/qt_toolkit_object.py", line 76, in activate_top_down
>     self.init_widget()
>   File "/usr/local/lib/python3.7/site-packages/enaml/qt/qt_object_combo.py", line 95, in init_widget
>     self.refresh_items()
>   File "/usr/local/lib/python3.7/site-packages/enaml/qt/qt_object_combo.py", line 148, in refresh_items
>     if item == selected:
>   File "/usr/local/lib/python3.7/site-packages/serial/tools/list_ports_common.py", line 74, in __eq__
>     return self.device == other.device
> AttributeError: 'NoneType' object has no attribute 'device'
> Abort trap: 6

frmdstryr, 2020-02-21:
That's a different issue, see https://inkcut.org/t/crash-object-has-no-attribute-device/34/post/158/ for the fix

methiDD, 2020-03-24:
> @hamsterwode, can you see if changing the label to something that doesn't include "Setup", "Config", or "Preferences" makes it show up? OSX seems to think it's ok move menu's around for no reason.

I changed the Label to "Device-Setup..." and it shows up again. 
The Hotkey "Ctrl+Alt+P" works at MacOS as "Cmd+Option+P" - no need to change

mariusaraujo, 2020-08-04:
I just installed inkcut in OSX Catalina (all dependencies ok) and the issue persists. "Cmd+Option+P" does open the print settings but I had to find this issue to realize that. I'm using a Roland GS-24, which isn't on the list. is it similar to CX-24 or CM-24?

DeeEmm, 2020-08-16:
Similar issue here in Version 2 with OSX. No menus shown at all. - #267