inkcut/inkcut #182: Cut stops before finished

ErikEklund, 2019-07-17:
I have been able to successfully get my cutter to cut designs. However, it always stops before the design is finished. It cuts something like 90% of the design and then stops dead. There is no error message or similar, it just says it's completed. The live view shows the entire thing being printed, and it sends all commands to the printer and reaches the "disconnect" method.

What I noticed was that the cutter seems to buffer instructions. It takes something like 10 seconds from Inkcut sends a stream of commands until anything happens, and then it stops dead something like 10 seconds before the cut is completed. Could it be that the command buffer is not flushed to the cutter before the program disconnects?

I had a thought that it might be the disconnection that stops the printer before it is done, so I tried modifying the source to not disconnect after finishing the transfer, but that didn't help. It cuts flawlessly besides the fact that it never finishes the whole cut.

Any ideas of the cause? For now I "solved" it by adding another instance of the same image to cut, and just shutting down the cutter before it starts #2. That way at least the full first image is printed.

Cutter: Creation CTO630
OS: MacOS 10.13.6
Inkcut version: Latest github master branch
Interface: serial usb, 38400 baud, all settings according to the manual

frmdstryr, 2019-07-17:
It might be aborting when the port is closed at the end of the job.

Try disabling the disconnect by changing the if statement in:
https://github.com/inkcut/inkcut/blob/master/inkcut/device/plugin.py#L829

be `if False:` to see if it keeps going.  If that works I'll create an option so it keeps the port open after finishing the job

ErikEklund, 2019-07-18:
As I mentioned in the original post, I tried removing the disconnect code (those exact lines). I verified that it stayed connected, but the cut still stopped early.

frmdstryr, 2019-07-18:
Sorry I missed that... I can't think of anything at the moment that would cause this to happen. 

Some things to try:

1. Add an "IN;" in the after job commands in the settings
2. Save the plot to a file then send that with cat or some serial terminal and see if it cuts everything
3. Use wireshark to trace the output to ensure everything is being sent

Franck78, 2020-09-20:
Same king of problem with the chineese cheap plotter.

On a short job, everything is fine.
On a long job, if fails somewhere before the end.

* Can be a bug in the plotter buffering code
* something in usbstack code
* Can be write not testing for writing error

```
    def write(self, data):
        if not self.connection:
            raise IOError("{} is not opened".format(self.device_path))
        log.debug("-> {} | {}".format(self.device_path, data))
        if hasattr(data, 'encode'):
            data = data.encode()
        self.last_write = data
        self.connection.write(data)

```

Here is the output of a failed job taken from the usbmon kernel debug 

```
ffffa07e8107c600 1984493616 S Bo:1:007:2 -115 26 = 50443136 3431372c 35373133 3b504431 36343035 2c353731 323b
ffffa07e8107c600 1985362348 S Bo:1:007:2 -115 13 = 50443135 3637392c 35313039 3b
ffffa07e8107c600 1985363374 C Bo:1:007:2 0 13 >
ffffa07e8107c600 1985375389 S Bo:1:007:2 -115 13 = 50443135 3638352c 35313737 3b
ffffa07e8107c600 1985376373 C Bo:1:007:2 0 13 >
ffffa07e8107c600 1985392672 S Bo:1:007:2 -115 13 = 50443135 3638342c 35323530 3b
ffffa07e8107c600 1985394373 C Bo:1:007:2 0 13 >
ffffa07e8107c600 1985404267 S Bo:1:007:2 -115 13 = 50443135 3637362c 35333330 3b
ffffa07e8107c600 1985405372 C Bo:1:007:2 0 13 >
ffffa07e8107c600 1985421934 S Bo:1:007:2 -115 13 = 50443135 3636322c 35343134 3b
ffffa07e8107c600 1991158307 C Bo:1:007:2 0 13 >
ffffa07dfcf1e9c0 1991167625 S Bo:1:007:2 -115 4884 = 50443135 3634302c 35353035 3b504431 35363130 2c353630 313b5044 31353537
ffffa07dfcf1e9c0 2010692006 C Bo:1:007:2 0 4884 >
ffffa07dfcf1e600 2010696735 S Bo:1:007:2 -115 8192 = 50443739 37302c32 3732383b 50443739 35352c32 3734363b 50443739 35342c32
ffffa07dfcf1e600 2061766213 C Bo:1:007:2 0 8192 >
ffffa07dfcf1ef00 2061766809 S Bo:1:007:2 -115 8192 = 34383536 3b504431 31303135 2c353033 313b5044 31313132 312c3531 39363b50
ffffa07dfcf1ef00 2095738854 C Bo:1:007:2 0 8192 >
ffffa07dfcf1e180 2095739336 S Bo:1:007:2 -115 8192 = 50443130 3636362c 34383435 3b504431 30363238 2c343733 323b5044 31303539
ffffa07dfcf1e180 2159723864 C Bo:1:007:2 0 8192 >
ffffa07eb5fc0240 2159724333 S Bo:1:007:2 -115 4746 = 2c373336 303b5044 32313436 332c3733 31383b50 44323134 38382c37 3237363b
ffffa07eb5fc0240 2159727699 C Bo:1:007:2 -2 64 >
ffffa07d9666d6c0 2159728680 C Bi:1:007:1 -2 0

```

Everything goes smoothly, sometime a 26 bytes URB. Then 'the thing' switches to 4k then 8k data. If I read this correctly, only 64 byte are sent from the last 4746 block

Inkcut thinks it is doing right while not :
```
2020-09-20 16:24:01,948 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD20133,4049;
2020-09-20 16:24:01,959 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD20147,4092;
2020-09-20 16:24:01,975 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD20152,4138;
2020-09-20 16:24:01,991 | DEBUG   | inkcut | -> /dev/usb/lp0 | PU0,0;
2020-09-20 16:24:02,000 | DEBUG   | inkcut | device | finish
2020-09-20 16:24:02,001 | DEBUG   | inkcut | -> /dev/usb/lp0 | IN;
2020-09-20 16:24:02,001 | DEBUG   | inkcut | device | disconnect
2020-09-20 16:24:02,001 | DEBUG   | inkcut | -- /dev/usb/lp0 | closed by request

```

spontarelliam, 2021-02-09:
I was having this same problem with my Roland SV-8 and @Franck78 's for loop fixed it. Thanks @Franck78 !

Franck78, 2021-02-09:
Hello @spontarelliam

ah, so if your Roland is a real one (not chineese fake) , we can assume it is not buggy and this moves the bug on the computer side, somewhere....
And until that somewhere is discovered, @frmdstryr, maybe you rethink your choice !

ErikEklund, 2021-02-09:
As a note, while I have a cheap chinese plotter, it worked flawlessly with two other cutting softwares on the exact same hardware (SignCut Pro 2 and one that was included with the printer, forget the name as I tossed it out ages ago). I it seems to be inkcut specific.

spontarelliam, 2021-02-09:
It is a genuine Roland cutter. That same cutter operating on the same svg file works fine when I use TuxPlot. I also have an MH-721 that cut the same svg perfectly fine in InkCut without the patch. So many variables!

spontarelliam, 2021-03-07:
I don't have an answer yet, but I have observed that this seems to become a problem not so much with long jobs, but with jobs that require a lot of high accelerations of the tool. I made a test file to draw boxes for 45 seconds and this worked without issue, but when I made a file to draw text for 45 seconds, the preview gets ahead of the tool itself, then when the preview finishes, the machine stops early.

Franck78, 2021-03-07:
And your faulty job + the loop filler fix ? Pass or don't pass ?

1heinz, 2021-06-02:
I had the same problem with a Vevor KI-870A and bigger data.
With the fix of filling 8000 "PU0.0" it works.

amercer86, 2021-06-17:
Sorry for my ignorance - how do I apply @Franck78's loop?

Franck78, 2021-06-17:
1. find the file inkcut/device/plugin.py
2. open it with an editor
3. search inside the correct "TODO" comment
4. the type in the lines (or copy past)
5. report if it fixed it

amercer86, 2021-06-23:
@Franck78 Thank you! Step 1 is actually where I'm hung up. I found the plugin.py file that was included in the initial download. Is that the one I edit or would it have been installed somewhere else?

Franck78, 2021-06-23:
it is a simple text file. You just modify it to add the lines at the correct place.

I can't relly explain you how to use an editor, tons of youtube tuto on that.

amercer86, 2021-06-23:
I'm more than familiar w/ editing. :P I simply don't know where the installation script would have put this file. My searches haven't been fruitful.

Franck78, 2021-06-24:
Then open the installation script and read it. Or run it again.
Or just unzip inkcut and run "python3 inkcut.py" ?

amercer86, 2021-06-24:
Thanks man. I appreciate the help - as condescending as it was.

Franck78, 2021-06-24:
Oh, do you realize you don't give any clues on your system, how you installed inkcut, your searches ? 

Again, the internet is full of tutorial on how to use python on window, on linux, how to search things on hard disk, how to download zipfile from github  etc etc etc and none of this related to the bug.

ozoid, 2022-03-12:
None of the solutions above worked for me, however I found this:
https://github.com/inkcut/inkcut/issues/306
And the fix for that is to add a line to the bottom of the 'def write(self,data)' method..
self.connection.waitForBytesWritten(-1)
for inkcut this is the file:
<pythonroot>\Lib\site-packages\inkcut\device\transports\qtserialport\plugin.py
Cuts finished perfectly after this :-)

oskarscholander, 2024-06-29:
I have tried these fixes, but no luck. It still disconnects before the job is finished and plotter say "Waiting"

Plotter: Cogi PCut CP-630
Mac OS Sonoma 14.2

[image]

Feels like I have tried everything :D