inkcut/inkcut #183: Crash after finish job

methiDD, 2019-07-21:
With the last Commit of pyserial it works to plot a job. But after finishing a Job, ink cut crashes/  don't react anymore.

From the Terminal:
```
...
2019-07-21 14:14:30,299 | DEBUG   | inkcut | -> /dev/cu.usbserial | PD794,2089;
2019-07-21 14:14:30,319 | DEBUG   | inkcut | -> /dev/cu.usbserial | PU0,0;
2019-07-21 14:14:30,323 | DEBUG   | inkcut | device | finish
2019-07-21 14:14:30,324 | DEBUG   | inkcut | -> /dev/cu.usbserial | IN;
2019-07-21 14:14:30,325 | DEBUG   | inkcut | device | disconnect
2019-07-21 14:14:30,326 | DEBUG   | inkcut | -- /dev/cu.usbserial | closed by request
```

Furthermore, I have to restart my Mac to use it properly again (keyboard don't work)
Maybe some problems closing the serial connection?

Anything I can do or test?

frmdstryr, 2019-07-21:
Sounds like an issue with osx's USB serial driver. Another user saw the same in #155.

See https://apple.stackexchange.com/questions/227956/usb-to-serial-adaptor-works-once-for-every-reboot

methiDD, 2019-07-21:
and again:... Thanks for your advice.
I installed the driver from https://www.mac-usb-serial.com/ and it works!

closed 2019-07-21 by methiDD