inkcut/inkcut #186: Doesn't seem to generate GPGL?

procastino, 2019-09-09:
Hi, 
when I try to cut something on inkcut send the job to the device, it doesn't work. The duration for the job appears as "0 min 0 sec" , so I'm guessing it doesn't generate the GPGL. If I go to "live" tab, the progress bar, well, progresses up to 100%, no errors shown, but the machine does not move. I'm using ubuntu 18.04 and my cutter is a Silhouette Portrait. I've used inkcut time ago, and never had a problem :/

raboof, 2019-09-09:
I'm pretty sure the Silhouette Portrait expects HPGL (and I though was automatically configured as such if you select it from the presets). Does that help?

procastino, 2019-09-09:
Thanks a lot, Arnout, but I believe it's GPGL that first "G" should stand
for "Graphtec", the brand's name. On the device settings, if I choose
"Silhouette Portrait" in "driver", it only allows me to pick GPGL on the
"protocol" tab :/

raboof, 2019-09-10:
I misremembered, sorry about the noise (caa0d4111c1e28523de9ae2b71ba569498bb26d5).

Did you add your printer to CUPS with the 'Raw Queue' driver as suggested at https://github.com/inkcut/inkcut/blob/master/docs/device-setup.md#cutters-with-a-parallel-interface ?

procastino, 2019-09-10:
Ups! I had not added it with the "Raw Queue" driver, I should have read the
docs slowlier...
Now Inkcut shows a time for the job, but I can't cut, though.
These are the settings that I need for Inkcut to show time for cutting:
General > Driver > Inkcut Generic Driver
Connection > Serial Port
And, also, I have to add the printer on Ubuntu with Raw Queue Driver.
If I change any of these, even if it's choosing "silhouette portrait"
driver, that inkcut happens to have or *Connection > Printer*, it goes back
to show "0 min 0 sec" when I try to print the job.
When I choose Connection > Serial Port, the menu doesn't show any port on
that window...
Thanks a lot for the help

procastino, 2019-09-10:
So, I followed this issue <https://github.com/inkcut/inkcut/issues/168> and
got it to work! Just to sum up, these are my settings:
Ubuntu printers:
printer added with *Raw Queue* driver
Inkcut Device Setup:
*General > Driver > Inkcut Generic Driver*

*Connection > Parallel Port *
I had to add my user to lp group with:
*sudo usermod -a -G lp user*
Thanks a lot for your help!

O mar., 10/09/2019 ás 20:28, tucho mendez (<tuchodefresa@gmail.com>)
escribiu:

> Ups! I had not added it with the "Raw Queue" driver, I should have read
> the docs slowlier...
> Now Inkcut shows a time for the job, but I can't cut, though.
> These are the settings that I need for Inkcut to show time for cutting:
> General > Driver > Inkcut Generic Driver
> Connection > Serial Port
> And, also, I have to add the printer on Ubuntu with Raw Queue Driver.
> If I change any of these, even if it's choosing "silhouette portrait"
> driver, that inkcut happens to have or *Connection > Printer*, it goes
> back to show "0 min 0 sec" when I try to print the job.
> When I choose Connection > Serial Port, the menu doesn't show any port on
> that window...
> Thanks a lot for the help
>
> escribiu:
>
>> I misremembered, sorry about the noise (caa0d41
>> ).
>>
>> Did you add your printer to CUPS with the 'Raw Queue' driver as suggested
>> at
>> https://github.com/inkcut/inkcut/blob/master/docs/device-setup.md#cutters-with-a-parallel-interface
>> ?
>>
>>
>
>
> Tucho méndez
> ===========
> *http://about.me/tuchomendez <http://about.me/tuchomendez>*
> <http://sites.google.com/site/sitiotuchomendez/>
> http://sites.google.com/site/sitiotuchomendez/

closed 2019-09-10 by procastino