inkcut/inkcut #190: Installation of InkCut on Win 10 64 Bit fails

vmario89, 2019-10-22:
Hi, i tried to install InkCut from Masterbranch but after running it fails. I have no idea how to solve. Maybe you can help. I attached my installation manual (PDF, sorry its in german) i created while installing this (feel free to use to add to your own documentation if you like).

[SK-InkCutSetup-221019-2343-8.pdf](https://github.com/inkcut/inkcut/files/3758792/SK-InkCutSetup-221019-2343-8.pdf)

The log so far:

```
Microsoft Windows [Version 10.0.17763.805]
(c) 2018 Microsoft Corporation. Alle Rechte vorbehalten.
 
C:\Users\mario>pip3 install git+https://github.com/codelv/inkcut.git
Collecting git+https://github.com/codelv/inkcut.git
  Cloning https://github.com/codelv/inkcut.git to c:\users\mario\appdata\local\temp\pip-req-build-x5v4kggv
  Running command git clone -q https://github.com/codelv/inkcut.git 'C:\Users\mario\AppData\Local\Temp\pip-req-build-x5v4kggv'
Requirement already satisfied: enaml>=0.10 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from inkcut==2.1.1.dev0) (0.10.4)
Requirement already satisfied: twisted in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from inkcut==2.1.1.dev0) (19.7.0)
Requirement already satisfied: enamlx>=0.4.1 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from inkcut==2.1.1.dev0) (0.4.1)
Requirement already satisfied: pyqtgraph in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from inkcut==2.1.1.dev0) (0.10.0)
Requirement already satisfied: qtconsole in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from inkcut==2.1.1.dev0) (4.5.5)
Requirement already satisfied: pyserial>=3.4 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from inkcut==2.1.1.dev0) (3.4)
Requirement already satisfied: jsonpickle in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from inkcut==2.1.1.dev0) (1.2)
Requirement already satisfied: lxml in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from inkcut==2.1.1.dev0) (4.4.1)
Requirement already satisfied: qt-reactor in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from inkcut==2.1.1.dev0) (0.6)
Requirement already satisfied: pywin32 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from inkcut==2.1.1.dev0) (225)
Requirement already satisfied: setuptools in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from enaml>=0.10->inkcut==2.1.1.dev0) (41.4.0)
Requirement already satisfied: atom>=0.4.2 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from enaml>=0.10->inkcut==2.1.1.dev0) (0.4.3)
Requirement already satisfied: qtpy>=1.3 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from enaml>=0.10->inkcut==2.1.1.dev0) (1.9.0)
Requirement already satisfied: kiwisolver>=1.0.0 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from enaml>=0.10->inkcut==2.1.1.dev0) (1.1.0)
Requirement already satisfied: ply>=3.4 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from enaml>=0.10->inkcut==2.1.1.dev0) (3.11)
Requirement already satisfied: zope.interface>=4.4.2 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from twisted->inkcut==2.1.1.dev0) (4.6.0)
Requirement already satisfied: constantly>=15.1 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from twisted->inkcut==2.1.1.dev0) (15.1.0)
Requirement already satisfied: incremental>=16.10.1 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from twisted->inkcut==2.1.1.dev0) (17.5.0)
Requirement already satisfied: Automat>=0.3.0 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from twisted->inkcut==2.1.1.dev0) (0.8.0)
Requirement already satisfied: hyperlink>=17.1.1 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from twisted->inkcut==2.1.1.dev0) (19.0.0)
Requirement already satisfied: PyHamcrest>=1.9.0 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from twisted->inkcut==2.1.1.dev0) (1.9.0)
Requirement already satisfied: attrs>=17.4.0 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from twisted->inkcut==2.1.1.dev0) (19.3.0)
Requirement already satisfied: numpy in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from pyqtgraph->inkcut==2.1.1.dev0) (1.17.3)
Requirement already satisfied: ipython-genutils in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from qtconsole->inkcut==2.1.1.dev0) (0.2.0)
Requirement already satisfied: ipykernel>=4.1 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from qtconsole->inkcut==2.1.1.dev0) (5.1.3)
Requirement already satisfied: pygments in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from qtconsole->inkcut==2.1.1.dev0) (2.4.2)
Requirement already satisfied: jupyter-core in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from qtconsole->inkcut==2.1.1.dev0) (4.6.0)
Requirement already satisfied: jupyter-client>=4.1 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from qtconsole->inkcut==2.1.1.dev0) (5.3.4)
Requirement already satisfied: traitlets in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from qtconsole->inkcut==2.1.1.dev0) (4.3.3)
Requirement already satisfied: six in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from Automat>=0.3.0->twisted->inkcut==2.1.1.dev0) (1.12.0)
Requirement already satisfied: idna>=2.5 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from hyperlink>=17.1.1->twisted->inkcut==2.1.1.dev0) (2.8)
Requirement already satisfied: tornado>=4.2 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from ipykernel>=4.1->qtconsole->inkcut==2.1.1.dev0) (6.0.3)
Requirement already satisfied: ipython>=5.0.0 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from ipykernel>=4.1->qtconsole->inkcut==2.1.1.dev0) (7.8.0)
Requirement already satisfied: python-dateutil>=2.1 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from jupyter-client>=4.1->qtconsole->inkcut==2.1.1.dev0) (2.8.0)
Requirement already satisfied: pyzmq>=13 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from jupyter-client>=4.1->qtconsole->inkcut==2.1.1.dev0) (18.1.0)
Requirement already satisfied: decorator in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from traitlets->qtconsole->inkcut==2.1.1.dev0) (4.4.0)
Requirement already satisfied: backcall in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut==2.1.1.dev0) (0.1.0)
Requirement already satisfied: pickleshare in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut==2.1.1.dev0) (0.7.5)
Requirement already satisfied: prompt-toolkit<2.1.0,>=2.0.0 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut==2.1.1.dev0) (2.0.10)
Requirement already satisfied: jedi>=0.10 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut==2.1.1.dev0) (0.15.1)
Requirement already satisfied: colorama; sys_platform == "win32" in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut==2.1.1.dev0) (0.4.1)
Requirement already satisfied: wcwidth in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from prompt-toolkit<2.1.0,>=2.0.0->ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut==2.1.1.dev0) (0.1.7)
Requirement already satisfied: parso>=0.5.0 in c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages (from jedi>=0.10->ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut==2.1.1.dev0) (0.5.1)
Building wheels for collected packages: inkcut
  Building wheel for inkcut (setup.py) ... done
[... cut ...]
  File "c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages\enaml\core\compiler_common.py", line 16, in <module>
    from . import byteplay as bp
  File "c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages\enaml\core\byteplay\__init__.py", line 11, in <module>
    from .wbyteplay import *
  File "c:\users\mario\appdata\local\programs\python\python38-32\lib\site-packages\enaml\core\byteplay\wbyteplay.py", line 92, in <module>
    BREAK_LOOP,
NameError: name 'BREAK_LOOP' is not defined
 
C:\Users\mario>
```

regards, Mario

frmdstryr, 2019-10-22:
Python 3.8 is not yet supported by the graphics library wrapper Inkcut uses. 

Try uninstalling python 3.8 and installing python 3.6 or 3.7, then install inkcut again.

vmario89, 2020-01-19:
hi. thanks for feedback. closing this issue. maybe you should update your documentation to be more precise what libs to be used

closed 2020-01-19 by vmario89

Malczyk69, 2020-03-01:
Hello
after many attempts I was able to install inkcut (python 3.7 amd64) but when I try to run I have an error:

2020-03-01 11:55:11,864 | INFO    | inkcut | ========================================
2020-03-01 11:55:11,880 | INFO    | inkcut | Inkcut launched
2020-03-01 11:55:11,880 | INFO    | inkcut | Version: 2.1.1
2020-03-01 11:55:11,880 | INFO    | inkcut | Python: 3.7.6 (tags/v3.7.6:43364a7ae0, Dec 19 2019, 00:42:30) [MSC v.1916 64 bit (AMD64)]
2020-03-01 11:55:11,880 | INFO    | inkcut | System: uname_result(system='Windows', node='ekmar_2015', release='8.1', version='6.3.9600', machine='AMD64', processor='AMD64 Family 21 Model 19 Stepping 1, AuthenticAMD')
2020-03-01 11:55:11,880 | INFO    | inkcut | Executable: c:\users\marek\appdata\local\programs\python\python37\python.exe
2020-03-01 11:55:11,880 | INFO    | inkcut | Args: ['C:\\Users\\Marek\\AppData\\Local\\Programs\\Python\\Python37\\Scripts\\inkcut']
2020-03-01 11:55:12,364 | INFO    | inkcut | Qt: 5.14.1 Api:
2020-03-01 11:55:12,364 | INFO    | inkcut | ========================================
2020-03-01 11:55:22,099 | DEBUG   | inkcut | Inkcut loaded.
2020-03-01 11:55:22,177 | DEBUG   | inkcut | CLI | Commands loaded
2020-03-01 11:55:22,177 | DEBUG   | inkcut | CLI | No cli command was given.
2020-03-01 11:55:23,833 | DEBUG   | inkcut | CLI | Commands loaded
2020-03-01 11:55:23,833 | DEBUG   | inkcut | CLI | No cli command was given.
2020-03-01 11:55:23,927 | INFO    | inkcut | Loading plugin inkcut.device
2020-03-01 11:55:24,208 | DEBUG   | inkcut | Starting plugin 'inkcut.device'
2020-03-01 11:55:28,067 | INFO    | inkcut | Loading plugin inkcut.preview
2020-03-01 11:55:28,083 | DEBUG   | inkcut | Starting plugin 'inkcut.preview'
2020-03-01 11:55:28,099 | INFO    | inkcut | Loading plugin inkcut.monitor
2020-03-01 11:55:28,114 | DEBUG   | inkcut | Starting plugin 'inkcut.monitor'
2020-03-01 11:55:28,145 | INFO    | inkcut | Loading plugin inkcut.joystick
2020-03-01 11:55:28,161 | DEBUG   | inkcut | Starting plugin 'inkcut.joystick'
2020-03-01 11:55:28,239 | INFO    | inkcut | Loading plugin inkcut.console
2020-03-01 11:55:33,333 | INFO    | inkcut | Loading plugin inkcut.job
2020-03-01 11:55:33,349 | DEBUG   | inkcut | Starting plugin 'inkcut.job'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x000000D4F02DE6C8>.auto_approve = False'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x000000D4F02DE6C8>.count = 0'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x000000D4F02DE6C8>.created = 2020-02-29 16:20:52.533698'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x000000D4F02DE6C8>.data = '
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x000000D4F02DE6C8>.duration = 0:00:00'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x000000D4F02DE6C8>.ended = None'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x000000D4F02DE6C8>.length = 0.0'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x000000D4F02DE6C8>.progress = 0'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x000000D4F02DE6C8>.speed = 0.0'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x000000D4F02DE6C8>.started = None'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x000000D4F02DE6C8>.status = staged'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x000000D4F02DE6C8>.units = in'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x000000D4F02DE788>.color = #000000'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x000000D4F02DE788>.cost = 1.0'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x000000D4F02DE788>.force = 10'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x000000D4F02DE788>.is_roll = False'
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x000000D4F02DE788>.name = '
2020-03-01 11:55:33,364 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x000000D4F02DE788>.padding = [10.0, 10.0, 10.0, 10.0]'
2020-03-01 11:55:33,380 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x000000D4F02DE788>.size = [1800.0, 2700.0]'
2020-03-01 11:55:33,380 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x000000D4F02DE788>.speed = 10'
2020-03-01 11:55:33,380 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x000000D4F02DE788>.use_force = False'
2020-03-01 11:55:33,380 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x000000D4F02DE788>.use_speed = False'
2020-03-01 11:55:33,380 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x000000D4F02DE788>.used = [0.0, 0.0]'
[... cut ...]
  File "c:\users\marek\appdata\local\programs\python\python37\lib\site-packages\jupyter_client\client.py", line 118, in start_channels
    self.control_channel.start()
  File "c:\users\marek\appdata\local\programs\python\python37\lib\site-packages\jupyter_client\client.py", line 200, in control_channel
    socket, self.session, self.ioloop
TypeError: ChannelABC() takes no arguments</pre>

<p> and include the logs from <a href="C:\Users\Marek/.config/inkcut\logs">C:\Users\Marek/.config/inkcut\logs</a>.
</p>
</html>

frmdstryr, 2020-03-01:
@Malczyk69 Please run `pip install jupyter-client==5.3.4` it's a bug recently introduced in the latest version of that package. See #231

Malczyk69, 2020-03-01:
Thanks for the help but I have another problem. inkcut works but does not send the project to the plotter.
I have Creation PCut 630, USB Serial Port (COM2) configured, inkcut set to PCut - COM2.

2020-03-01 20:13:10,548 | INFO    | inkcut | Opening D:/00_projekty_bieżące/rysunek.svg
2020-03-01 20:13:10,548 | INFO    | inkcut | {'type': 'update', 'object': <inkcut.job.models.Job object at 0x000000F3D44CC588>, 'name': 'model', 'oldvalue': None, 'value': <PyQt5.QtGui.QPainterPath object at 0x000000F3E1C57668>}
2020-03-01 20:13:10,564 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2020-03-01 20:13:30,320 | DEBUG   | inkcut | device | submit <inkcut.job.models.Job object at 0x000000F3D44CC588>
2020-03-01 20:13:30,335 | INFO    | inkcut | device | Initializing job
2020-03-01 20:13:30,336 | DEBUG   | inkcut | device | init <inkcut.job.models.Job object at 0x000000F3D44CC588>
2020-03-01 20:13:30,336 | DEBUG   | inkcut | device | Path length: 2488.6190563301584
2020-03-01 20:13:36,700 | INFO    | inkcut | device | Connecting to device
2020-03-01 20:13:36,700 | INFO    | inkcut | device | Processing job
2020-03-01 20:13:36,701 | DEBUG   | inkcut | device | connect
2020-03-01 20:13:36,732 | DEBUG   | inkcut | -> COM2 | IN;
2020-03-01 20:13:36,732 | DEBUG   | inkcut | COM2 | opened
2020-03-01 20:13:36,733 | INFO    | inkcut | device | Working...
2020-03-01 20:13:36,736 | DEBUG   | inkcut | -> COM2 | PU4580,1101;
2020-03-01 20:13:37,890 | DEBUG   | inkcut | -> COM2 | PD4577,1050;
2020-03-01 20:13:37,903 | DEBUG   | inkcut | -> COM2 | PD4568,1000;
2020-03-01 20:13:37,917 | DEBUG   | inkcut | -> COM2 | PD4534,901;
2020-03-01 20:13:37,944 | DEBUG   | inkcut | -> COM2 | PD4479,807;
2020-03-01 20:13:37,971 | DEBUG   | inkcut | -> COM2 | PD4404,716;
.
.
.
.
2020-03-01 20:13:43,723 | DEBUG   | inkcut | -> COM2 | PD6674,1564;
2020-03-01 20:13:43,738 | DEBUG   | inkcut | -> COM2 | PD6677,1511;
2020-03-01 20:13:43,753 | DEBUG   | inkcut | -> COM2 | PU0,0;
2020-03-01 20:13:43,755 | DEBUG   | inkcut | device | finish
2020-03-01 20:13:43,755 | DEBUG   | inkcut | -> COM2 | IN;
2020-03-01 20:13:43,756 | DEBUG   | inkcut | device | disconnect
2020-03-01 20:13:43,756 | DEBUG   | inkcut | -- COM2 | closed by request
2020-03-01 20:14:05,513 | INFO    | inkcut | Closing D:/00_projekty_bieżące/rysunek.svg
2020-03-01 20:14:05,513 | INFO    | inkcut | {'type': 'update', 'object': <inkcut.job.plugin.JobPlugin object at 0x000000F3D41D5A08>, 'name': 'job', 'oldvalue': <inkcut.job.models.Job object at 0x000000F3D44CC588>, 'value': <inkcut.job.models.Job object at 0x000000F3DDDB0408>}
2020-03-01 20:14:05,523 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2020-03-01 20:14:05,526 | INFO    | inkcut | Saving state due to change: {'type': 'update', 'object': <inkcut.job.plugin.JobPlugin object at 0x000000F3D41D5A08>, 'name': 'job', 'oldvalue': <inkcut.job.models.Job object at 0x000000F3D44CC588>, 'value': <inkcut.job.models.Job object at 0x000000F3DDDB0408>}
2020-03-01 20:14:18,790 | INFO    | inkcut | ========================================
2020-03-01 20:14:18,790 | INFO    | inkcut | Inkcut exited cleanly
2020-03-01 20:14:18,790 | INFO    | inkcut | ========================================