inkcut/inkcut #193: Cutter does not respond with IN; unless I fiddle with the language settings.

MatthewCroughan, 2019-11-17:
I've tested this, and it is reproducible. I can submit some footage to demonstrate this soon, but I'm away from the cutter. The cutter is a Roland CX-24 CAMM-1 connected via Parallel 

Basically, this is what happens:

Launch Inkcut
Load a file and attempt to cut via Ctrl + P
It is falsely sent to the cutter, `lsof /dev/usb/lp0` shows that inkcut thinks it is indeed sending data to the cutter
This is only fixed by going to settings and changing the language to something else, and then back again
After I have done this, connecting the printer via the connect button in Inkcut shows "IN;" via the console
Prior to doing this, connections would show nothing, as the printer is not sending anything back

raboof, 2019-11-18:
That's surprising, I've been happily using inkcut with this cutter and haven't seen this issue yet. Unfortunately I don't have things correctly set up right now, but would like to further diagnose this.

MatthewCroughan, 2019-11-18:
@raboof If you get the time, maybe we can talk on Matrix or some platform more realtime and then concatenate the details here? I'll be in with the cutter tonight and able to demonstrate the issue in more detail. @chunkypuffs:matrix.org

raboof, 2019-11-19:
I'm happy to chat on IRC (I'm raboof on freenode, looks like you're there as well?), though I'm not sure if I have anything more useful than "huh that is weird" to share at this point :smile:.

aaujon, 2019-11-20:
Hi, I'm experiencing the same issue with a Roland Camm1 GS-24, I need to change some settings and save again the correct /dev/usb/lp0 to make it work. I didn't reproduce this issue when using it as a printer. My guess is that someting is not properly saved the .config file. Maybe comparing the .config file before and after the tweak can show what's wrong.

MatthewCroughan, 2019-11-20:
@aaujon A brilliant suggestion, I gave up when I noticed there was no debug occurring on the CLI during the interaction here. I'll try this next time I'm at the cutter. Though this is only fixed for the time the application is open, so there may never be a state when the config file is correctly written to.

MatthewCroughan, 2019-11-22:
Alright, so I've tested this on a different cutter with Serial, and the same issue does not occur. This might very well just be an issue with using parallel ports, I'm not running cups, maybe this is the issue?

aaujon, 2020-01-31:
I reproduce this issue, looks like the command IN is not send, the first commandf sent to the machine is a MOVE. Maybe the method connection_made is not called, I will dig more to fix this.

aaujon, 2020-01-31:
The method is called, the problem is that in : 
`
def write(self, data):
        """ Call this to write data using the underlying transport.
        This should typically not be overridden.
        """
        if self.transport is not None:
            self.transport.write(data)
`
transport is null when starting inkcut

MatthewCroughan, 2020-01-31:
Thanks so much for taking the time to reproduce this. I'm glad you could do it with what little terminology and specifics I was able to provide. @aaujon

aaujon, 2020-01-31:
Can you try to use parallel port connexion instead of 'raw device descriptor' and inkcut with today commit ? It's working for me without having to tweak the settings, so you can wait for a real fix.

aaujon, 2020-02-01:
Just tried again, and the problem is also present with parallel port.

aaujon, 2020-02-06:
@frmdstryr I tried to understand the issue but I'm facing another one, I can't use pdb because of https://github.com/nucleic/enaml/issues/376, that you fixed recently, do you know if a release of enaml is coming ?

frmdstryr, 2020-02-06:
@aaujon is this using blade offset? 

Can you capture a log? I want to see if the 0's are actually being written to the cutter or if the cutter itself is not able to parse the input and just treating it as a 0.

aaujon, 2020-02-06:
I wil not have access to the cutter for maybe 2 weeks, I will come back to you with the log.

aaujon, 2020-02-24:
> 2020-02-24 13:33:10,823 | INFO    | inkcut | ========================================
2020-02-24 13:33:10,823 | INFO    | inkcut | Inkcut launched
2020-02-24 13:33:10,823 | INFO    | inkcut | Version: 2.1.2dev
2020-02-24 13:33:10,823 | INFO    | inkcut | Python: 3.7.6 (default, Dec 19 2019, 22:52:49) 
[GCC 9.2.1 20190827 (Red Hat 9.2.1-1)]
2020-02-24 13:33:10,826 | INFO    | inkcut | System: uname_result(system='Linux', node='localhost.localdomain', release='5.4.14-200.fc31.x86_64', version='#1 SMP Thu Jan 23 13:06:12 UTC 2020', machine='x86_64', processor='x86_64')
2020-02-24 13:33:10,827 | INFO    | inkcut | Executable: /usr/bin/python
2020-02-24 13:33:10,827 | INFO    | inkcut | Args: ['main.py']
2020-02-24 13:33:10,838 | INFO    | inkcut | Qt: 5.12.2 Api:
2020-02-24 13:33:10,838 | INFO    | inkcut | ========================================
2020-02-24 13:33:11,854 | DEBUG   | inkcut | Starting plugin 'inkcut.core'
2020-02-24 13:33:11,854 | DEBUG   | inkcut | Inkcut loaded.
2020-02-24 13:33:11,858 | DEBUG   | inkcut | CLI | Commands loaded
2020-02-24 13:33:11,859 | DEBUG   | inkcut | CLI | No cli command was given.
2020-02-24 13:33:11,860 | WARNING | inkcut | Setting locale: fr_FR
2020-02-24 13:33:12,267 | DEBUG   | inkcut | CLI | Commands loaded
2020-02-24 13:33:12,268 | DEBUG   | inkcut | CLI | No cli command was given.
2020-02-24 13:33:12,277 | INFO    | inkcut | Loading plugin inkcut.joystick
2020-02-24 13:33:12,279 | DEBUG   | inkcut | Starting plugin 'inkcut.joystick'
2020-02-24 13:33:12,358 | INFO    | inkcut | Loading plugin inkcut.device
2020-02-24 13:33:12,412 | DEBUG   | inkcut | Starting plugin 'inkcut.device'
2020-02-24 13:33:12,413 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.absolute = False'
2020-02-24 13:33:12,413 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.commands_after = '
2020-02-24 13:33:12,413 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.commands_before = '
2020-02-24 13:33:12,413 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.commands_connect = '
2020-02-24 13:33:12,413 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.commands_disconnect = '
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.custom_rate = -1.0'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.force = 40.0'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.force_enabled = False'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.force_units = g'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.interpolate = False'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.mirror_x = False'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.mirror_y = False'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.quality_factor = 1.0'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.rotation = 0'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.sample_rate = 100'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.scale = [1.0, 1.0]'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.speed = 4.0'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.speed_enabled = False'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.speed_units = in/s'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.spooled = False'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.step_size = 3.5433070866'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.step_time = 35.0'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.swap_xy = False'
2020-02-24 13:33:12,414 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x7fa97e6360a0>.test_mode = False'
2020-02-24 13:33:12,416 | DEBUG   | inkcut | Restoring state '<inkcut.device.transports.parallelport.plugin.ParallelConfig object at 0x7fa97e6363c0>.device_path = /dev/usb/lp0'
2020-02-24 13:33:12,416 | DEBUG   | inkcut | Restoring state '<inkcut.device.transports.parallelport.plugin.ParallelConfig object at 0x7fa97e6363c0>.mode = r+b'
[... cut ...]
`diff --git a/inkcut/device/plugin.py b/inkcut/device/plugin.py
index 740fc1f..5b76770 100644
--- a/inkcut/device/plugin.py
+++ b/inkcut/device/plugin.py
@@ -181,6 +181,7 @@ class DeviceProtocol(Model):
         This should typically not be overridden.

         """
+        log.debug("write: {} - {}".format(self.transport, data))
         if self.transport is not None:
             self.transport.write(data)
`

So, nothing is written tot he cutter because transport isn't initialized properly when starting inkcut

Franck78, 2020-04-04:
Hello,
I confirm this bug is not fixed in inkut 2.1.2 (master Latest commit adc64eb on 22 Feb )

Setup as usual, cheap chineese Vevor cutter
Using Paralell port

You absolutely need to switch to some protocol, validate then switch back to HPGL.

If you do, not the log is
```
device connect
--/dev/usb/lp0 opened
device working
device finish
device disconnect

```

when correct re-switched then it is
```
device connect
--/dev/usb/lp0 opened
->/dev/usb/lp0 IN;
device working
->/dev/usb/lp0 PU113..............;
....
device finish
->/dev/usb/lp0 IN;
device disconnect

```

How do can we fix this ?

Thank you

Franck78, 2020-04-09:
Ok, I have greatly narrowed the problem.

Some init code is not called as it should
or some "Saving state due to change: {'type': 'request'}"
does not save everything or the readback is....

It will take two days for me to 'comprehend the logic'
and two minutes for you to fix it.

            
nevertheless        
        
        
        
     "transport" is no restaured
     in Class DeviceProtocol

     so no actual writes
     
my custom log when not working:     
2020-04-09 23:09:37,422 | INFO    | inkcut | ====HPGL MOVE======01134
2020-04-09 23:09:37,422 | INFO    | inkcut | ======TRANSPORT=None
and when working:     
2020-04-09 23:10:13,820 | INFO    | inkcut | ====HPGL MOVE======00
2020-04-09 23:10:13,820 | INFO    | inkcut | ======TRANSPORT=/dev/usb/lp0

IN file DEVICE/plugin.py

from class DeviceProtocol(Model):

def write(self, data):
        """ Call this to write data using the underlying transport.

        This should typically not be overridden.

        """
        log.info ("======TRANSPORT=" + str(self.transport))
        if self.transport is not None:
            self.transport.write(data)

            
            
from  class DeviceTransport(Model):
           
    when the protocol change is ignored, then really the fix is not working        
  
    def _observe_protocol(self, change):
        """ Whenever the protocol changes update the transport reference

        """
        return  DO NOTHING, WON't FIX (as proof)

        if change['type'] == 'update' and change['value']:
            self.protocol.transport = self     <= FIX THINGS

aaujon, 2020-04-10:
Hi @Franck78,
I think you pointed the bug, by changing condition from DeviceTransport to :+1:
 ` if (change['type'] == 'update' or change['type'] == 'create') and change['value']:`
It's working for me, can you check that ?

Franck78, 2020-04-10:
Arnaud, c'est absolument réparé en ajoutant 'create' !!

@anyone, yes adding the 'create' case correctly initializes the transport.

aaujon, 2020-04-10:
Great ! Thanks a lot for pointing the problem in  _observe_protocol.
I will prepare a PR today

commit 19517567 (2020-04-10): fix #193 init transport on startup
commit 27edc7dd (2020-04-10): Merge pull request #241 from aaujon/bugfix/193-initialise-transport

fix #193 init transport on startup
commit 5af23009 (2020-06-05): Fixes #193  the init bug
commit 1c82eb75 (2020-06-09): Fixes #193  the init bug
closed 2020-04-10 by frmdstryr