inkcut/inkcut #194: PixMax windows issue COM3 access denied

Adriachampion, 2019-11-19:
Win 10 x64 \ PixMax 720

When I press "Start" for printing some shape the program says me:

`WARNING | inkcut | could not open port 'COM3': PermissionError(13, 'Acceso denegado.', None, 5)`

How can I solve this problem?
[image]

Thanks!

skarzzzz, 2019-11-28:
Have you found a solution to this as I have the same issue and I’m using a pixmax 721

Adriachampion, 2019-11-28:
> Have you found a solution to this as I have the same issue and I’m using a pixmax 721

Not yet, I'm waiting for a response...

anITiot, 2019-12-01:
Check the permissions for the device COM3 ! 
Have you tested to write something from another program to the PixMax?

skarzzzz, 2019-12-01:
I can successfully use it from Inkscape using the pixmax with no problems but I can’t from inkcut so I’m not sure what’s going on ?

skarzzzz, 2019-12-02:
so ive disabled and renabled the com3 port and now i dont get the error it proceeds to plot but doesnt actually use the cutter it just does it on the program? anyone ?