inkcut/inkcut #196: Pixmax 720/721 Not in the setup!

skarzzzz, 2019-11-27:
Ive finnaly get inkcut to work and now i cant set uyp my cutter because it doesnt exist in the meny so  is it possible to get the Pixmax added to the device setup please???

frmdstryr, 2019-11-28:
Most cutters work with the generic driver. Try that one.

skarzzzz, 2019-11-28:
I’ve tried that driver and still doesn’t seem to work I get an error about the port (com3) access denied?? I don’t really have a clue

splext, 2020-09-22:
Can I bump this?
I have a pixmax 721 and inkcut setup on Windows 10. 
I have managed to get inkcut to the point where it doesn't reject or deny the cutter. But the cutter doesn't move when live is.

So far I'm on generic cutter, 9600 baud, using hdpl. Everything else I try with rejected the connection so I feel this is the right track. Just need Someone to help me troubleshoot this further who knows the backend. Thanks