inkcut/inkcut #197: Roland CAMM-1 GS-24 cutting wrong -> Compiling generates random hops to origin?

elfoebbro, 2019-11-30:
I run Inkcut on Ubuntu 18.04
inkcut is installed as standalone (but also as inkscape extension does not work)

svg file generated via inkscape is loaded fine and generates a reasonable path to be cut
[image]

when I launch the cut via "send to device", the cut is bad.
apparently some some path are included (either in the compiling phase or in the communication phase). 
As a matter of fact the blade hops to the origin in unwanted places (the blade is not lifted), cutting through the outlines.
[image]

Thanks for your help

frmdstryr, 2019-11-30:
Can you attach the cut file being used used?

aaujon, 2020-01-30:
Hi,
I use the same cutter, I run into that issue when I was using the cutter as a printer, switching to serial solved the problem for me, is it the same for you ?

aaujon, 2020-01-31:
Sorry, I mean switching to parallelport, with today commit it's working great.
@frmdstryr for me the bug is reproductible with every file with this cutter and using the cutter as a printer on linux

frmdstryr, 2020-01-31:
@aaujon thanks for verifying this. It sounds flow control is not working. Does it do this with hardware flow control on (check RTS/CTS)?

aaujon, 2020-01-31:
Ma machine isn't connected on a serial port so I can't try the flow control option, it's connected on usb.

aaujon, 2020-02-24:
I compared the log when using printer and serial port, the generated code is exactly the same. So I guess you are right about some flow control issue, but I really don't know how to check that with a printer on linux

aaujon, 2020-02-24:
I will try some settings in cups to see if it can be fixed that way

Zaubentrucker, 2020-03-17:
I had the exact same issue with the GS-24 but resolved it by setting the Printer Model from "Generic Text-Only" to "Raw" within the cups browser. Afterwards the additional cuts to the origin were gone.

I did this by going to http://localhost:631/admin (cups-browsed has to be running) and then choosing:
Manage Printers -> GS-24 -> (pick Modify Printer from the list-picker that says "Administration") -> Continue -> Continue -> Select another Make/Manufacturer -> Raw -> Continue -> Raw Queue -> Modify Printer

aaujon, 2020-06-05:
@Zaubentrucker thanks a lot for the tip, I will be able to test it soon and I'm pretty sure that's the problem.

aaujon, 2020-07-06:
@Zaubentrucker It works, thanks ! @elfoebbro can you give it a try ?

valvelifter1951, 2021-01-31:
Zaubentrucker
Thanks, I was having the same trouble with my Roland CAMM-1 (cutting movement path lines) and I set it to RAW as you suggested and it's working fine now.

DamianLeuthold, 2021-04-01:
Setting the Camm-1 to RAW in ubuntu 20.04 worked for me as well.  Thanks!