inkcut/inkcut #198: Ubuntu 18.04 Installation failed

reparator, 2019-12-02:
Tried to install inkcut under ubuntu 18.04.
After 
`pip3 install inkcut `
I got the following result:
```
pip3 install inkcut
Collecting inkcut
  Using cached https://files.pythonhosted.org/packages/f5/f2/76b5ff7cba74764d18fa0b0ca57a4009c35d6f227fe8f6dd95ea52647947/inkcut-2.1.0-py2.py3-none-any.whl
Collecting twisted (from inkcut)
  Using cached https://files.pythonhosted.org/packages/88/e2/0c21fadf0dff02d145db02f24a6ed2c24993e7242d138babbca41de2f5a2/Twisted-19.10.0-cp36-cp36m-manylinux1_x86_64.whl
Collecting enamlx>=0.4.1 (from inkcut)
  Using cached https://files.pythonhosted.org/packages/cb/f7/48efe7f1b91d91e879e9b762d2cdcbe166edaf03a8989f46a43c938d868a/enamlx-0.4.1-py2.py3-none-any.whl
Collecting pyqtgraph (from inkcut)
Collecting qt-reactor (from inkcut)
  Using cached https://files.pythonhosted.org/packages/83/77/506a082b80c43fc50509f13c1aed9b63716295f3f0f983035b8bb49f6e23/qt_reactor-0.6-py2.py3-none-any.whl
Collecting pyserial>=3.4 (from inkcut)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting pycups; sys_platform == "linux" (from inkcut)
Collecting enaml>=0.10 (from inkcut)
  Using cached https://files.pythonhosted.org/packages/2b/86/79067440fb47bc83a28704e07ba90c1fa57efd7e0ac068213345d1af974e/enaml-0.10.4-cp36-cp36m-manylinux1_x86_64.whl
Collecting jsonpickle (from inkcut)
  Using cached https://files.pythonhosted.org/packages/07/07/c157520a3ebd166c8c24c6ae0ecae7c3968eb4653ff0e5af369bb82f004d/jsonpickle-1.2-py2.py3-none-any.whl
Collecting qtconsole (from inkcut)
  Using cached https://files.pythonhosted.org/packages/7c/57/3528b84ffa753e2089908bbf74bb5ae60653eb7a63797b6234e88b847d67/qtconsole-4.6.0-py2.py3-none-any.whl
Collecting lxml (from inkcut)
  Using cached https://files.pythonhosted.org/packages/68/30/affd16b77edf9537f5be051905f33527021e20d563d013e8c42c7fd01949/lxml-4.4.2-cp36-cp36m-manylinux1_x86_64.whl
Collecting attrs>=17.4.0 (from twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/a2/db/4313ab3be961f7a763066401fb77f7748373b6094076ae2bda2806988af6/attrs-19.3.0-py2.py3-none-any.whl
Collecting constantly>=15.1 (from twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/b9/65/48c1909d0c0aeae6c10213340ce682db01b48ea900a7d9fce7a7910ff318/constantly-15.1.0-py2.py3-none-any.whl
Collecting zope.interface>=4.4.2 (from twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/16/1c/d9e4d1e4eb9777ae675c5ac01290e70012498944d5e743bd2777d1096ad7/zope.interface-4.7.1-cp36-cp36m-manylinux1_x86_64.whl
Collecting PyHamcrest>=1.9.0 (from twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/9a/d5/d37fd731b7d0e91afcc84577edeccf4638b4f9b82f5ffe2f8b62e2ddc609/PyHamcrest-1.9.0-py2.py3-none-any.whl
Collecting Automat>=0.3.0 (from twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/e5/11/756922e977bb296a79ccf38e8d45cafee446733157d59bcd751d3aee57f5/Automat-0.8.0-py2.py3-none-any.whl
Collecting incremental>=16.10.1 (from twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/f5/1d/c98a587dc06e107115cf4a58b49de20b19222c83d75335a192052af4c4b7/incremental-17.5.0-py2.py3-none-any.whl
Collecting hyperlink>=17.1.1 (from twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/7f/91/e916ca10a2de1cb7101a9b24da546fb90ee14629e23160086cf3361c4fb8/hyperlink-19.0.0-py2.py3-none-any.whl
Collecting numpy (from pyqtgraph->inkcut)
  Using cached https://files.pythonhosted.org/packages/d2/ab/43e678759326f728de861edbef34b8e2ad1b1490505f20e0d1f0716c3bf4/numpy-1.17.4-cp36-cp36m-manylinux1_x86_64.whl
Collecting qtpy (from qt-reactor->inkcut)
  Using cached https://files.pythonhosted.org/packages/cd/fd/9972948f02e967b691cc0ca1f26124826a3b88cb38f412a8b7935b8c3c72/QtPy-1.9.0-py2.py3-none-any.whl
Collecting kiwisolver>=1.0.0 (from enaml>=0.10->inkcut)
  Using cached https://files.pythonhosted.org/packages/f8/a1/5742b56282449b1c0968197f63eae486eca2c35dcd334bab75ad524e0de1/kiwisolver-1.1.0-cp36-cp36m-manylinux1_x86_64.whl
Collecting atom>=0.4.2 (from enaml>=0.10->inkcut)
  Using cached https://files.pythonhosted.org/packages/b4/9c/bf9f593b96f0e539c03723335388f01c347d698d3482d3e9705a80560edc/atom-0.4.3-cp36-cp36m-manylinux1_x86_64.whl
Collecting setuptools (from enaml>=0.10->inkcut)
  Using cached https://files.pythonhosted.org/packages/54/28/c45d8b54c1339f9644b87663945e54a8503cfef59cf0f65b3ff5dd17cf64/setuptools-42.0.2-py2.py3-none-any.whl
Collecting ply>=3.4 (from enaml>=0.10->inkcut)
  Using cached https://files.pythonhosted.org/packages/a3/58/35da89ee790598a0700ea49b2a66594140f44dec458c07e8e3d4979137fc/ply-3.11-py2.py3-none-any.whl
Collecting pygments (from qtconsole->inkcut)
  Using cached https://files.pythonhosted.org/packages/be/39/32da3184734730c0e4d3fa3b2b5872104668ad6dc1b5a73d8e477e5fe967/Pygments-2.5.2-py2.py3-none-any.whl
Collecting traitlets (from qtconsole->inkcut)
  Using cached https://files.pythonhosted.org/packages/ca/ab/872a23e29cec3cf2594af7e857f18b687ad21039c1f9b922fac5b9b142d5/traitlets-4.3.3-py2.py3-none-any.whl
Collecting ipykernel>=4.1 (from qtconsole->inkcut)
  Using cached https://files.pythonhosted.org/packages/e1/92/8fec943b5b81078399f969f00557804d884c96fcd0bc296e81a2ed4fd270/ipykernel-5.1.3-py3-none-any.whl
Collecting jupyter-core (from qtconsole->inkcut)
  Using cached https://files.pythonhosted.org/packages/fb/82/86437f661875e30682e99d04c13ba6c216f86f5f6ca6ef212d3ee8b6ca11/jupyter_core-4.6.1-py2.py3-none-any.whl
Collecting ipython-genutils (from qtconsole->inkcut)
  Using cached https://files.pythonhosted.org/packages/fa/bc/9bd3b5c2b4774d5f33b2d544f1460be9df7df2fe42f352135381c347c69a/ipython_genutils-0.2.0-py2.py3-none-any.whl
Collecting jupyter-client>=4.1 (from qtconsole->inkcut)
  Using cached https://files.pythonhosted.org/packages/13/81/fe0eee1bcf949851a120254b1f530ae1e01bdde2d3ab9710c6ff81525061/jupyter_client-5.3.4-py2.py3-none-any.whl
Collecting six (from PyHamcrest>=1.9.0->twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/65/26/32b8464df2a97e6dd1b656ed26b2c194606c16fe163c695a992b36c11cdf/six-1.13.0-py2.py3-none-any.whl
Collecting idna>=2.5 (from hyperlink>=17.1.1->twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/14/2c/cd551d81dbe15200be1cf41cd03869a46fe7226e7450af7a6545bfc474c9/idna-2.8-py2.py3-none-any.whl
Collecting decorator (from traitlets->qtconsole->inkcut)
  Using cached https://files.pythonhosted.org/packages/8f/b7/f329cfdc75f3d28d12c65980e4469e2fa373f1953f5df6e370e84ea2e875/decorator-4.4.1-py2.py3-none-any.whl
[... 3 lines cut ...]
Collecting pyzmq>=13 (from jupyter-client>=4.1->qtconsole->inkcut)
  Using cached https://files.pythonhosted.org/packages/94/07/cee3d328a2e13f9de1c2b62cced7a389b61ac81424f2e377f3dc9d668282/pyzmq-18.1.1-cp36-cp36m-manylinux1_x86_64.whl
Collecting python-dateutil>=2.1 (from jupyter-client>=4.1->qtconsole->inkcut)
  Using cached https://files.pythonhosted.org/packages/d4/70/d60450c3dd48ef87586924207ae8907090de0b306af2bce5d134d78615cb/python_dateutil-2.8.1-py2.py3-none-any.whl
Collecting pexpect; sys_platform != "win32" (from ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut)
  Using cached https://files.pythonhosted.org/packages/0e/3e/377007e3f36ec42f1b84ec322ee12141a9e10d808312e5738f52f80a232c/pexpect-4.7.0-py2.py3-none-any.whl
Collecting pickleshare (from ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut)
  Using cached https://files.pythonhosted.org/packages/9a/41/220f49aaea88bc6fa6cba8d05ecf24676326156c23b991e80b3f2fc24c77/pickleshare-0.7.5-py2.py3-none-any.whl
Collecting prompt-toolkit!=3.0.0,!=3.0.1,<3.1.0,>=2.0.0 (from ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut)
  Using cached https://files.pythonhosted.org/packages/7f/1f/e145dd467dc9b0e6f1e64232c03119498dfec497e383f1e8be9f83eaa97e/prompt_toolkit-3.0.2-py3-none-any.whl
Collecting jedi>=0.10 (from ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut)
  Using cached https://files.pythonhosted.org/packages/55/54/da994f359e4e7da4776a200e76dbc85ba5fc319eefc22e33d55296d95a1d/jedi-0.15.1-py2.py3-none-any.whl
Collecting backcall (from ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut)
Collecting ptyprocess>=0.5 (from pexpect; sys_platform != "win32"->ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut)
  Using cached https://files.pythonhosted.org/packages/d1/29/605c2cc68a9992d18dada28206eeada56ea4bd07a239669da41674648b6f/ptyprocess-0.6.0-py2.py3-none-any.whl
[... cut ...] xml-4.4.2 numpy-1.17.4 parso-0.5.1 pexpect-4.7.0 pickleshare-0.7.5 ply-3.11 prompt-toolkit-3.0.2 ptyprocess-0.6.0 pycups-1.9.74 pygments-2.5.2 pyqtgraph-0.10.0 pyserial-3.4 python-dateutil-2.8.1 pyzmq-18.1.1 qt-reactor-0.6 qtconsole-4.6.0 qtpy-1.9.0 setuptools-42.0.2 six-1.13.0 tornado-6.0.3 traitlets-4.3.3 twisted-19.10.0 wcwidth-0.1.7 zope.interface-4.7.1
```
This seemed to be ok. But when I try to start the program in terminal, there is no inkcut available:
`inkcut: file not found`
Any Ideas?

frmdstryr, 2019-12-02:
Try running with `python3 -m inkcut`

reparator, 2019-12-02:
Wow, very fast answer! But unfortunately it does not start inkcut:
`:~$ python3 -m inkcut`
`/usr/bin/python3: No module named inkcut.__main__; 'inkcut' is a package and cannot be directly executed`

reparator, 2019-12-02:
OK, found something:
inkcut can be started with`
/home/me/.local/bin/inkcut`
So it seems to be installed locally in my home directory and not globally. Why? Dont know...

closed 2019-12-02 by reparator

frmdstryr, 2019-12-02:
Sorry it I remember now, it should be  `python3 -m inkcut.app` but it sounds like you got it to work.