inkcut/inkcut #201: Switch language

AntoineTurmel, 2019-12-17:
I have installed Inkut 2.1.0 using pip on a french version of Windows 10 (x64).
But the software remain in English.
How do I switch to french ? I have notice the InkCut folder was missing res/translations :

C:\Users\Antoine\AppData\Local\Programs\Python\Python37\Lib\site-packages\inkcut\res\

I did add the translations folder along with fr_FR files, but it's still in english.

AntoineTurmel, 2019-12-17:
Just to be sure, I have verified this:
```
>>> import enaml
>>> from enaml.qt import QT_API, QtCore, QtWidgets, QtGui
>>> QtCore.QLocale.system().name()
'fr_FR'
```

Crepin1989, 2019-12-28:
[image]
I installed Inkscape and the extension of Inkcut. When I want to cut out something, the extension does not open.

Crepin1989, 2019-12-28:
"Traceback (most recent call last):
  File "inkcut_cut.py", line 73, in <module>
    effect.affect()
  File "inkex.py", line 283, in affect
    self.effect()
  File "inkcut_cut.py", line 67, in effect
    close_fds=True)
  File "C:\Program Files\Inkscape\lib\python2.7/subprocess.py", line 278, in __init__
    raise ValueError("close_fds is not supported on Windows "
ValueError: close_fds is not supported on Windows platforms if you redirect stdin/stdout/stderr"

I have the text above as comment, but the extension does not link my computer to Inkcut. 

What should do? I use Windows Pro 10, the French version. 

Thanks for any proposal of solution.

aaujon, 2020-01-21:
Hi, about the language issue, I believe it has been fixed in 8234cc9, the issue was reported in bug #195

aaujon, 2020-02-01:
Can you try again with the latest developpement version ?

aaujon, 2020-06-05:
@frmdstryr you can close this

closed 2020-07-29 by frmdstryr

Crepin1989, 2021-08-31:
Hello Sir, please can I send AnyDesk to you so that you install Inkcut on
my computer?

I tried many times, but unfortunately I didn't succeed.

Have a Nice Day!