inkcut/inkcut #202: Installation on MX Linux 19 Unsuccessful

BigDon55, 2019-12-18:
Ran the installation as directed in the instructions. Received the following error at end of install:

-------------------------------------- BEGIN CAPTURE ------------------------------------
 pip3 install inkcut
Collecting inkcut
  Using cached https://files.pythonhosted.org/packages/f5/f2/76b5ff7cba74764d18fa0b0ca57a4009c35d6f227fe8f6dd95ea52647947/inkcut-2.1.0-py2.py3-none-any.whl
Collecting enamlx>=0.4.1 (from inkcut)
  Using cached https://files.pythonhosted.org/packages/cb/f7/48efe7f1b91d91e879e9b762d2cdcbe166edaf03a8989f46a43c938d868a/enamlx-0.4.1-py2.py3-none-any.whl
Collecting twisted (from inkcut)
  Using cached https://files.pythonhosted.org/packages/06/28/2a433e147de68c8416aa0179c45e67b67161f5c0f24aaaf1723f6229f574/Twisted-19.10.0-cp37-cp37m-manylinux1_x86_64.whl
Collecting lxml (from inkcut)
  Using cached https://files.pythonhosted.org/packages/95/60/552fc6e027bc1158ba4691cccfdf6eb77f206f2f21d3c1c5f23b89f68a0e/lxml-4.4.2-cp37-cp37m-manylinux1_x86_64.whl
Collecting qt-reactor (from inkcut)
  Using cached https://files.pythonhosted.org/packages/83/77/506a082b80c43fc50509f13c1aed9b63716295f3f0f983035b8bb49f6e23/qt_reactor-0.6-py2.py3-none-any.whl
Collecting jsonpickle (from inkcut)
  Using cached https://files.pythonhosted.org/packages/07/07/c157520a3ebd166c8c24c6ae0ecae7c3968eb4653ff0e5af369bb82f004d/jsonpickle-1.2-py2.py3-none-any.whl
Requirement already satisfied: pycups; sys_platform == "linux" in /usr/lib/python3/dist-packages (from inkcut) (1.9.73)
Collecting pyqtgraph (from inkcut)
  Using cached https://files.pythonhosted.org/packages/cd/ad/307e0280df5c19986c4206d138ec3a8954afc722cea991f4adb4a16337d9/pyqtgraph-0.10.0.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ModuleNotFoundError: No module named 'setuptools'

--------------------------------------------- END OF CAPTURE -------------------------------

I tried to run the application both from the "run application" window and from the command line and it was a no go.

All of the prerequisites installed with no errors shown.

frmdstryr, 2019-12-18:
Run
`sudo apt-get install python3-setuptools`
Then repeat the install command.

BigDon55, 2019-12-18:
Ran your command and it completed without error. However, rerunning the inkcut install again, gave several errors:

---------------------------------- BEGIN CAPTURE --------------------------
error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for pyqtgraph
  Running setup.py clean for pyqtgraph
  Running setup.py bdist_wheel for tornado ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-l29isl_8/tornado/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-e9igjss9 --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for tornado
  Running setup.py clean for tornado
  Running setup.py bdist_wheel for backcall ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-l29isl_8/backcall/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-236jpeg8 --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for backcall
  Running setup.py clean for backcall
Failed to build pyqtgraph tornado backcall
Installing collected packages: decorator, ipython-genutils, traitlets, jupyter-core, pygments, tornado, pyzmq, python-dateutil, jupyter-client, parso, jedi, wcwidth, prompt-toolkit, backcall, pickleshare, ipython, ipykernel, qtconsole, jsonpickle, kiwisolver, qtpy, ply, atom, enaml, enamlx, incremental, PyHamcrest, attrs, Automat, zope.interface, hyperlink, constantly, twisted, qt-reactor, lxml, numpy, pyqtgraph, pyserial, inkcut
  The scripts jupyter, jupyter-migrate and jupyter-troubleshoot are installed in '/home/Don/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  The script pygmentize is installed in '/home/Don/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  Running setup.py install for tornado ... done
  The scripts jupyter-kernel, jupyter-kernelspec and jupyter-run are installed in '/home/Don/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  Running setup.py install for backcall ... done
  The scripts iptest, iptest3, ipython and ipython3 are installed in '/home/Don/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  The scripts enaml-compileall and enaml-run are installed in '/home/Don/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  The script automat-visualize is installed in '/home/Don/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  The scripts cftp, ckeygen, conch, mailmail, pyhtmlizer, tkconch, trial, twist and twistd are installed in '/home/Don/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  The scripts f2py, f2py3 and f2py3.7 are installed in '/home/Don/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  Running setup.py install for pyqtgraph ... done
  The script inkcut is installed in '/home/Don/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed Automat-0.8.0 PyHamcrest-1.9.0 atom-0.4.3 attrs-19.3.0 backcall-0.1.0 constantly-15.1.0 decorator-4.4.1 enaml-0.10.4 enamlx-0.4.1 hyperlink-19.0.0 incremental-17.5.0 inkcut-2.1.0 ipykernel-5.1.3 ipython-7.10.2 ipython-genutils-0.2.0 jedi-0.15.1 jsonpickle-1.2 jupyter-client-5.3.4 jupyter-core-4.6.1 kiwisolver-1.1.0 lxml-4.4.2 numpy-1.17.4 parso-0.5.2 pickleshare-0.7.5 ply-3.11 prompt-toolkit-3.0.2 pygments-2.5.2 pyqtgraph-0.10.0 pyserial-3.4 python-dateutil-2.8.1 pyzmq-18.1.1 qt-reactor-0.6 qtconsole-4.6.0 qtpy-1.9.0 tornado-6.0.3 traitlets-4.3.3 twisted-19.10.0 wcwidth-0.1.7 zope.interface-4.7.1

------------------------------------ END OF CAPTURE ---------------------------
$ inkcut
bash: inkcut: command not found

frmdstryr, 2019-12-18:
Hmm... Run
`sudo apt-get install python3-wheel`
Then repeat the install command (again).

Seems like different versions of linux have different packages by default...

BigDon55, 2019-12-18:
That did it! I had to run the commands and the installation as sudo however.

Once completed the application will run as a regular user.

Thanks a bunch.

frmdstryr, 2019-12-18:
Great thanks!

closed 2019-12-18 by frmdstryr