inkcut/inkcut #203: Inkcut's extension doesn't open on my Windows10

Crepin1989, 2019-12-28:
[image]

I downloaded Inkcut's pulgrin from inkscape.org and copied the files in in share/extensions. 
I opened Inkscape and when I wanted to cut out my document, Inkcut doesn't open, but I see its extension en in the list in Inkscape. 

I use Windows 10 pro. Please, I am waiting for your explanation to get out of that situation.

As coment, Inkscape displays this: 

"Traceback (most recent call last):
  File "inkcut_cut.py", line 73, in <module>
    effect.affect()
  File "inkex.py", line 283, in affect
    self.effect()
  File "inkcut_cut.py", line 67, in effect
    close_fds=True)
  File "C:\Program Files\Inkscape\lib\python2.7/subprocess.py", line 278, in __init__
    raise ValueError("close_fds is not supported on Windows "
ValueError: close_fds is not supported on Windows platforms if you redirect stdin/stdout/stderr".

frmdstryr, 2019-12-31:
I uploaded the update inkscape plugins to https://inkscape.org/~frmdstryr/%E2%98%85inkcut, can you please re-download it and see if it works? Thanks!

vmario89, 2020-01-19:
hi. this works for me. Your update in InkScape gallery is missing in the git repository. Please update this repo to be equal to https://inkscape.org/~frmdstryr/%E2%98%85inkcut. regards and thanks for your work :-)