inkcut/inkcut #205: Mac Inkcut has no "device setup"

artfulnomad, 2020-01-01:
First of all I'm a rookie diving into this world, so my apologies if this seems trivial but I'll be grateful for any help! I've loaded Inkscape and Inkcut to use my Roland GX-24 cutter. Unfortunately I'm unable to view or open the "device setup" in Inkcut or view any mention of it on the application. Even when I go to print there's no option to open it.

On a separate note the plugin/extension is not sending paths to Inkcut from Inkscape. 

Currently using Mac Mojave 10.14.6

Over all would just like to use Inkscape and sent to the GX24 from my Mac. If there's a better way I'm all ears.

Thx!!

[image]

DeeEmm, 2020-01-24:
Can confirm. Same issue here.

However you can still access the dialog by using the keyboard shortcut opt+cmd+p

artfulnomad, 2020-01-24:
Unfortunately I’m not able to for some reason. Decide to go to Linux and it works great here.

frmdstryr, 2020-03-05:
Duplicate of #181

closed 2020-03-05 by frmdstryr

DeeEmm, 2020-08-15:
You have closed this as being a duplicate of #181 but yet you have closed #181 as being an issue of this issue.

So I have opened #267  as this issue has not been fixed and is still present.