inkcut/inkcut #206: InkCut Windows binary compile - ready to download

vmario89, 2020-01-19:
Hi,
i made a short description on how to compile InkCut native executable using PyInstaller.

https://wiki.fablabchemnitz.de/display/FCT/Setup+InkCut+Plotter+Software (end of the page)

I also uploaded an executable to my cloud kitchen: 
https://cloud.leyghis.de/d/47539a1682fe40a18472/?p=%2FInkcut&mode=list

I had no time to test most of the stuff so this might be unstable. Feel free to test it and/or to compile it yourself and/or adjust documentation for Windows users.

regards, Mario

frmdstryr, 2020-01-20:
Great thanks for sharing!

frmdstryr, 2020-01-20:
Also I might point out that using the "shortest path" cutting order can save a lot of cutting time if you see a lot of blue lines (as in a few of your pictures) if you haven't already tried it..

vmario89, 2020-01-20:
hey. i updated some things. Inkcut really requires a lot of modules to work. Hopefully less or no stack trace errors are now thrown. I am going to check Inkcut with my cutter fully tomorrow :-)

by the way the build command now looks like this:
`del /Q build\ & del /Q dist\ & py -3.7 -m PyInstaller app.py -n=inkcut --clean -i=res\icons\inkcut.ico --noconfirm --add-data="..\enaml;enaml" --add-data="..\enamlx;enamlx" --add-data="..\qtconsole;qtconsole" --add-data="..\zmq;zmq" --add-data="..\ipykernel;ipykernel" --add-data="..\future;future" --add-data="..\serial;serial" --add-data="..\twisted\internet;twisted\internet" --add-data="cli;inkcut\cli" --add-data="console;inkcut\console" --add-data="core;inkcut\core" --add-data="device;inkcut\device" --add-data="job;inkcut\job" --add-data="joystick;inkcut\joystick" --add-data="monitor;inkcut\monitor" --add-data="preview;inkcut\preview" --add-data="res;inkcut\res" --add-data="ui;inkcut\ui" --hidden-import qreactor --hidden-import jsonpickle --hidden-import win32print --onefile --windowed`

vmario89, 2020-02-02:
new compile to test: https://cloud.leyghis.de/d/47539a1682fe40a18472/files/?p=%2FInkcut%2Finkcut_v2.2.1dev_pull223_commit_aebd46a4f98586edf4ba754f5a5fcf62afea9492.exe

approx. next week i get access to some Sacebo plotter. There i will test if it works correctly^^

vmario89, 2020-02-04:
updated the link to fablabchemnitz.de/y/compilerkitchen

aaronaverill, 2020-06-30:
Unfortunately it fails to run.

  File "c:\users\mobius\appdata\local\programs\python\python37-32\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\mobius\appdata\local\programs\python\python37-32\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\mobius\AppData\Local\Programs\Python\Python37-32\Scripts\inkcut.exe\__main__.py", line 7, in <module>
  File "c:\users\mobius\appdata\local\programs\python\python37-32\lib\site-packages\inkcut\app.py", line 99, in main
    workbench.run()
  File "c:\users\mobius\appdata\local\programs\python\python37-32\lib\site-packages\inkcut\core\workbench.py", line 147, in run
    from enaml.workbench.core.core_manifest import CoreManifest
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "c:\users\mobius\appdata\local\programs\python\python37-32\lib\site-packages\enaml\core\import_hooks.py", line 151, in exec_module
    code, _ = self.get_code()
  File "c:\users\mobius\appdata\local\programs\python\python37-32\lib\site-packages\enaml\core\import_hooks.py", line 412, in get_code
    return self.compile_code()
  File "c:\users\mobius\appdata\local\programs\python\python37-32\lib\site-packages\enaml\core\import_hooks.py", line 376, in compile_code
    ast = parse(self.read_source(), file_info.src_path)
  File "c:\users\mobius\appdata\local\programs\python\python37-32\lib\site-packages\enaml\core\parser\__init__.py", line 42, in parse
    return _parser.parse(enaml_source, filename)
  File "c:\users\mobius\appdata\local\programs\python\python37-32\lib\site-packages\enaml\core\parser\base_parser.py", line 206, in parse
    return self.parser.parse(source, debug=0, lexer=lexer)
  File "c:\users\mobius\appdata\local\programs\python\python37-32\lib\site-packages\ply\yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
  File "c:\users\mobius\appdata\local\programs\python\python37-32\lib\site-packages\ply\yacc.py", line 1120, in parseopt_notrack
    p.callable(pslice)
  File "c:\users\mobius\appdata\local\programs\python\python37-32\lib\site-packages\enaml\core\parser\parser36.py", line 118, in p_atom10
    p[0] = m.body[0].value
IndexError: list index out of range

splext, 2020-09-21:
All the links in this thread are dead. Can anyone provide a working link to a windows installer?
@vmario89 ?

mjsir911, 2022-06-14:
Bump on this. Additionally, the link to [the store](https://www.codelv.com/store/) does not even include inkcut on it.

Would be really helpful to ship bundled executables.