inkcut/inkcut #208: USB Serial fatal error / crash

vmario89, 2020-01-20:
hmmm i am sitting next to our Roland PNC 950 and trying to configure it. I am trying to select serial interface but i receive an error which i am not able to fix :/ 
[image]

```
2020-01-20 15:43:17,148 | DEBUG   | asyncio | Using selector: SelectSelector
Traceback (most recent call last):
  File "Lib\site-packages\enaml\qt\qt_object_combo.py", line 117, in on_index_changed
  File "Lib\site-packages\enaml\core\declarative_meta.py", line 68, in declarative_change_handler
  File "Lib\site-packages\enaml\core\expression_engine.py", line 217, in write
  File "Lib\site-packages\enaml\core\standard_handlers.py", line 82, in __call__
  File "C:\Users\mario\AppData\Local\Programs\Python\Python37\Lib\site-packages\inkcut\dist\inkcut\inkcut\device\dialogs.enaml", line 124, in f
    device.connection = declaration.factory(driver, declaration, protocol)
  File "C:\Users\mario\AppData\Local\Programs\Python\Python37\Lib\site-packages\inkcut\dist\inkcut\inkcut\device\transports\serialport\manifest.enaml", line 13, in transport_factory
    from .plugin import SerialTransport, SerialConfig
  File "C:\Users\mario\AppData\Local\Programs\Python\Python37\Lib\site-packages\inkcut\dist\inkcut\inkcut\device\transports\serialport\plugin.py", line 21, in <module>
    from twisted.internet.serialport import SerialPort
  File "C:\Users\mario\AppData\Local\Programs\Python\Python37\Lib\site-packages\inkcut\dist\inkcut\twisted\internet\serialport.py", line 22, in <module>
    from serial import PARITY_NONE, PARITY_EVEN, PARITY_ODD
ImportError: cannot import name 'PARITY_NONE' from 'serial' (C:\Users\mario\AppData\Local\Programs\Python\Python37\Lib\site-packages\inkcut\dist\inkcut\serial\__init__.py)
```

pyserial says:
```
C:\Users\mario\AppData\Local\Programs\Python\Python37\Lib\site-packages\inkcut>pip3 install --upgrade pyserial
Requirement already up-to-date: pyserial in c:\users\mario\appdata\local\programs\python\python37\lib\site-packages (3.4)
```

This happens when i run Inkcut from Python as well as the Windows executable (https://github.com/inkcut/inkcut/issues/206) i am working an. Do you have an idea? Might it belong to some python module versioning? The following stuff i installed at the moment:

```
Package          Version    
---------------- -----------
altgraph         0.17       
atom             0.4.3      
attrs            19.3.0     
Automat          0.8.0      
backcall         0.1.0      
Click            7.0        
colorama         0.4.3      
constantly       15.1.0     
decorator        4.4.1      
dill             0.3.1.1    
enaml            0.10.4     
enamlx           0.4.1      
future           0.18.2     
hyperlink        19.0.0     
idna             2.8        
incremental      17.5.0     
inkcut           2.1.1      
ipykernel        5.1.3      
ipython          7.11.1     
ipython-genutils 0.2.0      
iso8601          0.1.12     
jedi             0.15.2     
jsonpickle       1.2        
jupyter-client   5.3.4      
jupyter-core     4.6.1      
kiwisolver       1.1.0      
klepto           0.1.8      
lxml             4.4.2      
Markdown         3.1.1      
mpmath           1.1.0      
mystic           0.3.5      
numpy            1.17.4     
parso            0.5.2      
pefile           2019.4.18  
pickleshare      0.7.5      
pip              19.3.1     
ply              3.11       
pox              0.2.7      
prompt-toolkit   3.0.2      
pybin            0.3.4      
pycairo          1.18.2     
pyclipper        1.1.0.post1
pycparser        2.19       
Pygments         2.5.2      
PyHamcrest       2.0.0      
PyInstaller      3.6        
PyQt5            5.14.1     
PyQt5-sip        12.7.0     
pyqtgraph        0.10.0     
pyserial         3.4        
python-dateutil  2.8.1      
python-gettext   4.0        
pywin32          227        
pywin32-ctypes   0.2.0      
PyYAML           5.3        
pyzmq            18.1.1     
qt-reactor       0.6        
qtconsole        4.6.0      
QtPy             1.9.0      
scipy            1.3.2      
serial           0.0.97     
setuptools       44.0.0     
six              1.14.0     
sympy            1.4        
tornado          6.0.3      
traitlets        4.3.3      
Twisted          19.10.0    
wcwidth          0.1.8      
wheel            0.33.6     
zmq              0.0.0      
zope.interface   4.7.1
```

frmdstryr, 2020-01-20:
You can try importing in `serial` in the Console and type `serial.` to see if it shows those fields. Sounds like either something changed or something is missing from the build.

vmario89, 2020-01-20:
[image]
same inside the app :-/ mmmmhh

vmario89, 2020-01-20:
could you maybe print our your pip module list? so i can check against. I could try to install older versions maybe

frmdstryr, 2020-01-20:
Does it only give the error when using the built version?

frmdstryr, 2020-01-20:
It looks like twisted isn't importing the fields properly on windows.

Edit `inkcut\twisted\internet\serialport.py` and change the failing lines from `from serial import` to `from serial.serialutil import`

vmario89, 2020-01-20:
haaaar fixed it! You don't need to change it. Instead i uninstalled serial. 

i did the following in short:

```
pip3 uninstall serial
pip3 uninstall pyserial
pip3 install pyserial
```
[image]

closed 2020-01-20 by vmario89