inkcut/inkcut #210: Feature request: better weeding lines

vmario89, 2020-01-20:
Hi,

i want to swap over from commercial Roland CutStudio which is old fashioned. I think Inkcut has enough power to replace it completely. The only thing i am missing is a really cool weeding line feature like present in Roland CutStudio, eCut or SignCut 2.

In CS it looks like that:
[image]

In eCut 6 it looks that way:
[image]

in SignCut:
[image]

i think this piece of algorithm is a little harder to code. 

At the moment when doing open source i am do it like described here (using InkScape):
https://wiki.fablabchemnitz.de/display/FCT/Weeding+lines+with+InkScape

frmdstryr, 2020-01-20:
Agree this is an area that needs improvement. The original inkcut had options for copy based weedlines (horizontal and vertical) but not within the graphic.  

The signcut weedlines makes no sense to me but something like the others could be useful.

vmario89, 2020-01-20:
hi, the signcut from my side example is not good (messed up). i just fiddled around. the way how to set them is quite good in signcut. https://signcutpro.com/ you can play around with test / evaluation copy as i did.

you can view that short youtube video to see how they do:
https://www.youtube.com/watch?v=BVsdpNGQ7Is

in general i think it would be cool to have the option to auto-generate weeding lines (horizontal, vertical, ballon) and the option to set them manually maybe . so it would be a combination. the best of two worlds so to say.