inkcut/inkcut #211: Fatal error /crash when setting "FFT"

vmario89, 2020-01-20:
hi,

another thing i found out when playing with Inkcut:

[image]

```
2020-01-20 16:22:53,703 | INFO    | inkcut | Saving state due to change: {'type': 'update', 'object': <inkcut.job.plugin.JobPlugin object at 0x0000017E365AE8C8>, 'name': 'recent_documents', 'oldvalue': ['C:/Users/mario/Seafile/cloud.leyghis.de/Meine Bibliothek/NichtDafür.svg'], 'value': ['C:/Users/mario/Seafile/cloud.leyghis.de/Meine Bibliothek/NichtDafür.svg', 'C:/Users/mario/Desktop/test.svg']}
Traceback (most recent call last):
  File "c:\users\mario\appdata\local\programs\python\python37\lib\site-packages\pyqtgraph\graphicsItems\PlotItem\PlotItem.py", line 891, in updateSpectrumMode
    c.setFftMode(b)
AttributeError: 'PainterPathPlotItem' object has no attribute 'setFftMode'
```

looks like some missing default value or so

regards, Mario

frmdstryr, 2020-01-20:
Those are added by the http://www.pyqtgraph.org/ and not really relevant. They should be disabled (at least they are on linux) I'm not sure why they are there in the first place.

This ideally would be solved by #105

vmario89, 2020-01-20:
Hi, for me it is not clear what's the solution in #105. What should i do? if i remove numpy or pyqtgraph package inkcut fails to start

installing pyqtpgraph with --user option brings no difference

as i understand the right click menu should just not appear right? thats how it looks like right now:

[image]

frmdstryr, 2020-01-20:
> What should i do?

Just ignore/don't use those menu items.

vmario89, 2020-01-20:
hmmm :-D okay i will mark it in documentation. But isn't there some way to remove them? Mayby by some kind of reflection/override? (i am no python developer^^)