inkcut/inkcut #212: Detect plotter over USB

vmario89, 2020-01-20:
Hi,
after fixing https://github.com/inkcut/inkcut/issues/208 i was hoping so see my plotter appear under parallel or serial driver menu (because i thought maybe our USB interface could appear under serial). This is not the case.

I have installed our plotter by using an USB interface (Parallel to USB Adapter) which i connected to the parallel interface of the Roland PNC 950

[image]

We run it with this:
https://www.ebay.de/p/1857757145?iid=362303682329&chn=ps&norover=1&mkevt=1&mkrid=707-134425-41852-0&mkcid=2&itemid=362303682329&targetid=859716921970&device=c&mktype=pla&googleloc=9068430&poi=&campaignid=7782956145&mkgroupid=90355854198&rlsatarget=pla-859716921970&abcId=1140506&merchantid=7558652&gclid=EAIaIQobChMIwJO38cGS5wIVWeN3Ch2FHw8vEAQYAyABEgI97_D_BwE

is there any way to overcome the usb bottleneck?

frmdstryr, 2020-01-20:
In linux it's setup as a printer then used from inkcut. I'm not sure how windows handles usb to parallel adapters.

vmario89, 2020-01-20:
hm okay it just might me the problem that the plotter driver itself is not installed on my machine. I was used to start a virtual machine containing windows xp and the compatible driver for pnc 950. that works fine. 

at windows 10 native OS the interface itself is visible in windows device manager, but the plotter is an unknown device below \USB001

thought that inkcut itself contains drivers to control those legacy machines (like CoCut does with its software > https://www.cocut.com/index.dml?sprache=deu&unten=products/deu/CoCut+Starter+-+Treiber~CCSA-T.html&von=html)

hmm so i will try what happens running Inkcut in Windows XP machine. That will give a good legacy feeling too xD

regards, Mario

vmario89, 2020-01-20:
add: https://www.computing.net/answers/windows-vista/drivers-for-roland-pnc-950/7358.html for overview of PN 950.

i wonder how works PNC 900 and PNC 1000 with Inkcut? Those machine are same age. Does Linux provide working drivers for them?

frmdstryr, 2020-01-20:
> but the plotter is an unknown device below \USB001

Sounds like you need the driver for the adapter.

> Does Linux provide working drivers for them?

Yes linux has drivers builtin.

vmario89, 2020-01-20:
the adapter is installed properly and works (i can route that interface into virtual box with windows xp machine and it works). the problem should be the plotter driver itself for Windows 10. there's just none. so a custom driver solution would be cool. I think that's what CoCut did with their product.

lkvapil, 2020-08-06:
@vmario89 Could you please help me setting up pnc950 on windows 10 ?
I installed windowsXP using Virtualbox and I got two RS232 adapters to USB but without any success :/

Thank you!

vmario89, 2020-08-10:
hi,

did you route the USB adapters correctly into the VM? you need to allow access from VM to the device to make it work. And you need to install old drivers inside the virtual machine correctly: https://www.rolanddga.com/support/products/cutting/camm-1-pnc-950-24-vinyl-cutter

additionally check the DIP switches at the side of your plotter if your interface is configured properly

MauriceNorden, 2023-11-19:
Hey a friend of mine have a similar vinyl cutter. (PNC0-1000)
First we've tried using a serialcable with inkcut but without any succes.
After that we've tried the paralel port with inkcut under linux but the scaling was way off for some reason.

Next approach was using windows XP vm with the paralel adapter since serial only works if you have a serial crossover cable.

Then we could send designs to the cutter but only the coordinates got plottet instead of the design.

We still are not done with the project so we'll keep you updated.

raboof, 2023-11-20:
> After that we've tried the paralel port with inkcut under linux but the scaling was way off for some reason.

You should be able to adjust the scaling in the device settings