inkcut/inkcut #213: Language / translations

vmario89, 2020-01-20:
Hey,
i did a lot of translation work for other FOSS and like to do this for Inkcut (german lang). I stumpled upon
https://github.com/inkcut/inkcut/blob/master/docs/translating.md.

At the moment i am missing the res/translations/ files. If i install it with pip the translations/ dir is missing. I created it manually and downloaded french lang files. 

and i am missing a button to switch lang in user interface. It's not a good idea to auto detect this. From my point of view when having a FabLab, like we do, we want to be multi-lingual without nationality detection. Users come and go and should be able to switch between their desired langs

found this in workbench.py
```
        locale = QtCore.QLocale.system().name()
        qtTranslator = QtCore.QTranslator()
        if qtTranslator.load("inkcut/res/translations/" + locale):
            self.application._qapp.installTranslator(qtTranslator)
```

so far... after dropping french translations and restarting inkcut nothing changes.

i found Qt Linguist for Windows: https://github.com/thurask/Qt-Linguist/releases
this allows to edit *.ts files. Now i need way to build up the pattern for lupdate enaml on Windows.

Can you please provide some thoughts about UI and so on? 

regards, Mario :-)

vmario89, 2020-01-20:
Add. question:
i run the command
`lupdate.exe -recursive -extensions enaml C:\Users\mario\AppData\Local\Programs\Python\Python37\lib\site-packages\inkcut -ts de_DE.ts -target-language de_DE`. That does not generate a ts file. instead a lot of errors, e.g.

```
... some more lines on top

C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/monitor/manifest.enaml:28: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/monitor/manifest.enaml:32: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/monitor/view.enaml:1: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/monitor/view.enaml:11: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/manifest.enaml:1: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/manifest.enaml:11: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/manifest.enaml:40: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/manifest.enaml:46: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/manifest.enaml:68: tr() cannot be called without context
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/settings.enaml:1: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/settings.enaml:11: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/settings.enaml:22: tr() cannot be called without context
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/settings.enaml:30: tr() cannot be called without context
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/settings.enaml:33: tr() cannot be called without context
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/settings.enaml:36: tr() cannot be called without context
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/view.enaml:1: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/view.enaml:11: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/view.enaml:22: tr() cannot be called without context
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/view.enaml:33: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/view.enaml:34: Unterminated C++ character
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/view.enaml:37: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/view.enaml:47: tr() cannot be called without context
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/preview/view.enaml:33: Unbalanced opening parenthesis in C++ code (or abuse of the C++ preprocessor)
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/dock.enaml:1: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/dock.enaml:11: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/dock.enaml:39: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/dock.enaml:41: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/manifest.enaml:1: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/manifest.enaml:11: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/manifest.enaml:85: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/manifest.enaml:91: Unterminated C++ string
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/manifest.enaml:99: Unterminated C++ character
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/manifest.enaml:138: tr() cannot be called without context
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/manifest.enaml:147: tr() cannot be called without context
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/manifest.enaml:152: tr() cannot be called without context
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/manifest.enaml:156: tr() cannot be called without context
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/manifest.enaml:161: tr() cannot be called without context
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/manifest.enaml:171: tr() cannot be called without context
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/manifest.enaml:176: tr() cannot be called without context
[... cut ...]
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/settings.enaml:66: tr() cannot be called without context
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/settings.enaml:74: Unterminated C++ character
C:/Users/mario/AppData/Local/Programs/Python/Python37/lib/site-packages/inkcut/ui/settings.enaml:74: Unbalanced opening bracket in C++ code (or abuse of the C++ preprocessor)

```

I cannot imagine that 
`lupdate-qt5 inkcut/*/*.enaml -I inkcut/*/*/*.enaml -ts inkcut/res/translations/fr_FR.ts`
will generate proper file output without warnings

vmario89, 2020-01-21:
[image]

this is the rest of missing translations i found after my last pull request some minutes ago. going to bed now :-P

aaujon, 2020-01-21:
Hi,
Thanks for your work on the translation, I think there is an issue with the use of tr(), which is an alias to QApplication.translate(). Looks like lupdates search through the files without using a preprocessor, so it doesn't see tr as a valid sentence to translate (see https://www.qtcentre.org/threads/56042-lupdate-problem).

vmario89, 2020-01-21:
yes i played with tr-function-alias but i did not get it :-( i think generating them by lupdate would be the cleanest way to improve overall translation progress. doing it by hand hardly sucks :-D

aaujon, 2020-01-21:
It works when the code was using QApplication.translate() directly. I generated the ts file with lupdate at that time.
I try again this morning and I can't make it work using the tr(), neither with lupdate-qt5 or pylupdate. I will try to find a workaround, if there is nothing to do I guess we'll have to remove the alias.

vmario89, 2020-01-22:
i dont know much about Qt and those aliases. For me it still unclear how to really apply that tr-function-alias parameter to the lupdate tool. Hopefully some easy to run command pops out which generated a recent ts file :-) The translation and releasing is easy then. A good starting point also might be just to create the english base file (include them in the repo) which can be used as template to generate other language translations against.

maybe the translation stuff could be centralized to a platform similar to https://poeditor.com

aaujon, 2020-01-30:
@frmdstryr the only way I see to fix this is to use QApplication.translate() directly again instead of tr(), if that's okay with you I can prepare a PR

frmdstryr, 2020-01-30:
Yeah go for it, I should have just left it like that in the first place.

aaujon, 2020-01-30:
PR is here : https://github.com/inkcut/inkcut/pull/218

aaujon, 2020-01-30:
@vmario89 This fix the ts file generation, I tested it on the french translation, you should be able to use the tool now. i also added the missing strings.

vmario89, 2020-02-02:
thanks alot! i was able to run lupdate succesfully. i made some pull request to update that stuff

aaujon, 2020-06-05:
@frmdstryr you can close this, the PR has been merged