inkcut/inkcut #217: Problem with sending to device from Windows - cutter does nothing

Petr-200, 2020-01-25:
I have installed and run LinkCut on Ubuntu 16.04 LTS and tried it with cutter MH-721 - connected by USB. Everything worked OK.
Than I tried to do the same on Windows machines - I have tested it on 3 computers - all with windows 10. I have tested different version of Python 3 (3.7, 3.6 and 3.5). I can run InkCut. I set the device as I set it on Ubuntu. Than I send data to device. I can see logs - It looks as InkCut correctly communicates with cutter. But cutter did nothing. Only one difference from linux version is that after sending all commands to USB-serial there is error:
QSocketNotifier: Invalid socket specified

I've attached whole log: 

[inkcut_log_windows.txt](https://github.com/inkcut/inkcut/files/4112984/inkcut_log_windows.txt)

vmario89, 2020-02-02:
hi, i made some native compile for windows: https://github.com/inkcut/inkcut/issues/206
does it occure with it too?
regards, Mario

Petr-200, 2020-02-03:
Thanks, I've tried it, but the result is the same. 
First sending to cutter (according logs) looks good, but cutter did nothing. Second sending to cutter ends with permission error. I am attaching log - start of incut application and two sending to cutter. 
[inkcut_log_windows2.txt](https://github.com/inkcut/inkcut/files/4149300/inkcut_log_windows2.txt)

frmdstryr, 2020-02-03:
Can you open Inkcut, then in the `Console` tab paste in

```python
from serial import Serial
with Serial('COM9') as s:
    s.write(b"IN;PU0,1000;PU1000,1000;PU1000,0;PU0,0;")
```

It should move around in a box or give an error..

Petr-200, 2020-02-09:
Hi, I'm sorry I was few day out of place with cutter to test it.
Now I've test it from console from executable inkcut, you've sent. Cutter did not move but no error was show:

Jupyter QtConsole 4.6.0
Python 3.7.6 (tags/v3.7.6:43364a7ae0, Dec 19 2019, 00:42:30) [MSC v.1916 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 7.11.1 -- An enhanced Interactive Python. Type '?' for help.
from serial import Serial
with Serial('COM9') as s:
    s.write(b"IN;PU0,1000;PU1000,1000;PU1000,0;PU0,0;")
with Serial('COM9') as s:
    s.write(b"IN;PU0,1000;PU1000,1000;PU1000,0;PU0,0;")

Than I tried it in different way:

Jupyter QtConsole 4.6.0
Python 3.7.6 (tags/v3.7.6:43364a7ae0, Dec 19 2019, 00:42:30) [MSC v.1916 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 7.11.1 -- An enhanced Interactive Python. Type '?' for help.
from serial import Serial
s = Serial('COM9')
s.write(b"IN;PU0,1000;PU1000,1000;PU1000,0;PU0,0;")
Out[3]: 39
s.close()
s2 = Serial('COM9')
s2.write(b"IN;PU0,1000;PU1000,1000;PU1000,0;PU0,0;")
Out[6]: 39
s2.close()

And cutter moved correctly in a square.

ArminHobbyProjects, 2020-02-10:
Hi,
I am trying to use Inkcut with Win10 and a cheap Chinese Rabbit-type cutter connect via a CH340.
I also see the error that the plotter does not respond and I get a 'QSocketNotifier: Invalid socket -1' message when trying to disconnect.

This behavior can be seen with with Mario's binary install, the installed version from the repository and
the master branch source code.

Unfortunately I am a total Python newbie, however, I managed to trace down the problem to the Twisted framework's implementation of the serial port (or its application).

I modified the serial port plugin, disabling the write and disconnect methods and played around with connect method: If I issue a write command directly after the creation of the serial port, the plotter moves correctly (self.connection.write(b"U1000,1000;U0,0;").
I can pass multiple moves in this command, however, a second write command is ignored, no exception thrown, just no more response from the plotter:

self.connection = SerialPort(....)
self.connection.write(b"U1000,1000;U0,0;")	<-- plotter moves
self.connection.write(b"U1000,1000;U0,0;") <-- no more response

The app does not hang, no stack traces...
To me, it seem that either the port is blocked by waiting for a reponse, or it is automatically closed.

I did not find a way around this issue, but maybe someone with more experience can give a hint.

My next attempt with Inkcut will be creating a dedicated serial plugin based on the file output driver and the pySerial serial interface (which works flawlessly on my machine).

Regards,
Armin

perryc, 2020-09-01:
Tried inkcut  current 2.1.2dev on a Railsail 800C and it's still not send data out the serial port when trying to plot in win10.  Same build plots to the device with no problem in unbuntu. 

From console this moves the plotter
`from serial import Serial
s = Serial('COM14')
cmd = ['IN', 'PU0,0', 'PA100,0', 'PA100,100', 'PA0,100', 'PA0,0']
for c in cmd:
   s.write((c+';').encode())
s.close()`
but this does not
`
import time
dev = workbench.get_plugin('inkcut.device').device
cmd = ['IN', 'PU0,0', 'PA1000,0', 'PA1000,1000', 'PA0,1000', 'PA0,0']
dev.connect()
for c in cmd:
   time.sleep(0.2)
   dev.connection.write((c+';').encode())
time.sleep(0.2)
dev.disconnect()
`
Only error I see is "invalid socket" at the end of the plot:
2020-09-01 12:16:14,409 | DEBUG   | inkcut | -> COM14 | PU0,0;
2020-09-01 12:16:14,416 | DEBUG   | inkcut | device | finish
2020-09-01 12:16:14,417 | DEBUG   | inkcut | -> COM14 | IN;
2020-09-01 12:16:14,423 | DEBUG   | inkcut | device | disconnect
2020-09-01 12:16:14,423 | DEBUG   | inkcut | -- COM14 | closed by request
QSocketNotifier: Invalid socket specified

ArminHobbyProjects, 2020-09-07:
I tried again the lastest version yesterday, but still the same error.
Has anybody ever sucessfully used Inkcut on a Win10 machine?

Ducsi, 2020-10-23:
I am also on current 2.1.2dev with Python: 3.7.9  and windows 10.
My plotter is a Sekonic SPL-450 with HPGL.
My observation is that **only the very first command** goes out on the serial port and then nothing else.
The first command is: **IN;**
If I change the connection from serial port to save to file and send that file to serial port eg with hercules.exe then the plotter moves perfectly!
I’m ready to help with tests, but I don’t know what and where I should change in the code...

Ducsi, 2020-10-26:
Please check this out! [https://inkcut.org/t/inkcut-works-on-win10-with-a-new-serial-port-implementation/67/](url)
I was able to make it work by writing a new Serialport connection based on QSerialPort.