inkcut/inkcut #220: Stacktrace using parallel port settings

aaujon, 2020-01-31:
When trying to configure a machine using parallel port, inkcut crashes and I get the stacktrace  :

> Traceback (most recent call last):
  File "/usr/local/lib64/python3.7/site-packages/enaml/qt/qt_abstract_button.py", line 77, in on_clicked
    self.declaration.clicked(checked)
  File "/usr/local/lib64/python3.7/site-packages/enaml/core/declarative_meta.py", line 68, in declarative_change_handler
    engine.write(owner, change['name'], change)
  File "/usr/local/lib64/python3.7/site-packages/enaml/core/expression_engine.py", line 217, in write
    pair.writer(owner, name, change)
  File "/usr/local/lib64/python3.7/site-packages/enaml/core/standard_handlers.py", line 82, in __call__
    call_func(func, (), {}, scope)
  File "/home/bobine2/app/inkcut/inkcut/device/transports/parallelport/settings.enaml", line 50, in f
    clicked :: model.refresh()
  File "/home/bobine2/app/inkcut/inkcut/device/transports/parallelport/plugin.py", line 122, in refresh
    self.ports = self._default_ports()
  File "/usr/local/lib64/python3.7/site-packages/enaml/core/standard_tracer.py", line 60, in __call__
    engine.update(owner, self.name)
  File "/usr/local/lib64/python3.7/site-packages/enaml/core/expression_engine.py", line 248, in update
    setattr(owner, name, pair.reader(owner, name))
  File "/usr/local/lib64/python3.7/site-packages/enaml/core/declarative_meta.py", line 68, in declarative_change_handler
    engine.write(owner, change['name'], change)
  File "/usr/local/lib64/python3.7/site-packages/enaml/core/expression_engine.py", line 217, in write
    pair.writer(owner, name, change)
  File "/usr/local/lib64/python3.7/site-packages/enaml/core/standard_handlers.py", line 82, in __call__
    call_func(func, (), {}, scope)
  File "/home/bobine2/app/inkcut/inkcut/device/transports/parallelport/settings.enaml", line 42, in f
    model.port = port.device
AttributeError: 'ParallelConfig' object has no attribute 'port'

frmdstryr, 2020-01-31:
Thanks for reporting this, seems it broke during a refactor. Can you see if it works with the fix above?

aaujon, 2020-01-31:
Yes it's working perfectly, thanks for the quick fix.

closed 2020-01-31 by aaujon