inkcut/inkcut #225: No WM_CLASS, causing windows to separate on Ubuntu

MatthewCroughan, 2020-02-07:
This causes the same issue I solved here https://github.com/Ultimaker/Cura/pull/6429

When inspecting with `xprob WM_CLASS` and clicking on the Inkcut window, you're greeted with an empty string.

```
doesuser@does-vinylcutter:/usr/share/applications$ xprop WM_CLASS
WM_CLASS:  not found.
```

Searching for WM_NAME it seems some part of the code has been successful in setting that.

```
doesuser@does-vinylcutter:/usr/share/applications$ xprop WM_CLASS
WM_CLASS:  not found.
```

After some digging I feel I've narrowed it down to these lines in this file.

https://github.com/inkcut/inkcut/blob/853fa8ee7e6df0f6ea5639b573d321b360e11f52/inkcut/ui/plugin.py#L268-L275

I do not have enough knowledge of QT to understand how to manipulate this code for the desired outcome. Although here are two examples of projects discussing the same issue and fixing it:

#### Example 1
Issue: 
https://github.com/spyder-ide/spyder/issues/2207
Fix:
https://github.com/spyder-ide/spyder/commit/d91a3220562f211c3e372526f2ff422be0eeecec

#### Example 2

https://github.com/bitcoin/bitcoin/issues/10064

MatthewCroughan, 2020-02-07:
This should be an easy fix if you know how QT works, any ideas? @goatchurchprime

frmdstryr, 2020-02-10:
It looks like the fix needs to go into the .desktop file?

MatthewCroughan, 2020-02-10:
@frmdstryr No, the problem is that pyqt5 is not setting the WM_CLASS variable, which means you cannot group windows by their class using the .desktop file, as the string is empty.

aaujon, 2020-04-10:
Looks like there is 2 different issues, the .desktop is missing StartupWMClass AND WM_CLASS isn't set (I check using xprop), both are needed I think. the only way to set WM_CLASS using pyqt is to set it when creating Application, something like 
`  parser.add_argument('--qt-name', help="Set the window name.", metavar="inkcut")
    args = parser.parse_args()
    app = app.Application(args)`
An exemple of a fix here : https://github.com/qutebrowser/qutebrowser/commit/617cd8977b7221d7d417ecc4bb4ed72e16fa354e#diff-d5908e8f4776d031dbd9c0d9dd403690