inkcut/inkcut #226: update needed for Inkscape version 1.0

brynnd, 2020-02-16:
Hello,
Thank you for providing this extension for Inkscape users!

This is just to let you know that most 3rd party Inkscape extensions, like this one, probably will not work with the upcoming new Inkscape version, the long-awaited version 1.0.

Here is the info you need to update this extension, so that it will work with 1.0 and future versions.  
https://wiki.inkscape.org/wiki/index.php?title=Updating_your_Extension_for_1.0

If you have further questions, you can contact Inkscape developers via [mailing lists](https://lists.inkscape.org/postorius/lists/?all-lists), [forum](https://inkscape.org/forums/extensions/), or the [chatroom](https://chat.inkscape.org/channel/inkscape_extensions).

If you have already updated it, please disregard this message.

All best,
brynn

aaujon, 2020-02-17:
duplicates #216

aaujon, 2020-04-10:
@frmdstryr you can close this duplicated issue

closed 2020-04-10 by frmdstryr