inkcut/inkcut #231: Problem in the installation of inkcut

jlberzal, 2020-02-26:
Hello and thanks for providing inkcut to the community,
I am having problems installing inkcut in Ubuntu, I have the same problem in Mac.
After following the installation tutorial, I am able to launch the application, but it then blocks with the following message:

Sorry, an error prevented Inkcut from starting
This is either a bug or a required library may be missing on your system. As always you can request help on the forums at https://inkcut.org. 
First try to restart the application. If the error persists you can try clearing the saved settings in /home/jlberzal/.config/inkcut and restart the application again. 
If that still doesn't work please report a new issue at https://github.com/codelv/inkcut with the following error: 
kernel_client.start_channels()
  File "/home/jlberzal/.virtualenvs/inkcut/lib/python3.6/site-packages/qtconsole/kernel_mixins.py", line 47, in start_channels
    super(QtKernelClientMixin, self).start_channels(*args, **kw)
  File "/home/jlberzal/.virtualenvs/inkcut/lib/python3.6/site-packages/ipykernel/inprocess/client.py", line 64, in start_channels
    super(InProcessKernelClient, self).start_channels()
  File "/home/jlberzal/.virtualenvs/inkcut/lib/python3.6/site-packages/jupyter_client/client.py", line 118, in start_channels
    self.control_channel.start()
  File "/home/jlberzal/.virtualenvs/inkcut/lib/python3.6/site-packages/jupyter_client/client.py", line 200, in control_channel
    socket, self.session, self.ioloop
TypeError: object() takes no parameters 

I have tried to installing in a virtual environment and at system level, also on a mac I get the same message.
 
Log file is attached, thanks in advance

[inkcut.txt](https://github.com/inkcut/inkcut/files/4254477/inkcut.txt)

erselbst, 2020-02-26:
I have the same error on Linux Mint 19.2 Cinnamon (Ubuntu 18.04) (Kernel 4.15.0-88-generic). The installation was successfuly but when I'm start inkcut I get the same error.

My inkcut installation:

> 
root:~# pip3 install --upgrade inkcut
Requirement already up-to-date: inkcut in /usr/local/lib/python3.6/dist-packages
Requirement already up-to-date: lxml in /usr/local/lib/python3.6/dist-packages (from inkcut)
Requirement already up-to-date: enamlx>=0.4.1 in /usr/local/lib/python3.6/dist-packages (from inkcut)
Requirement already up-to-date: jsonpickle in /usr/local/lib/python3.6/dist-packages (from inkcut)
Requirement already up-to-date: pycups; sys_platform == "linux" in /usr/local/lib/python3.6/dist-packages (from inkcut)
Requirement already up-to-date: qtconsole in /usr/local/lib/python3.6/dist-packages (from inkcut)
Requirement already up-to-date: twisted in /usr/local/lib/python3.6/dist-packages (from inkcut)
Requirement already up-to-date: pyserial>=3.4 in /usr/local/lib/python3.6/dist-packages (from inkcut)
Requirement already up-to-date: enaml>=0.10; python_version > "3.0" in /usr/local/lib/python3.6/dist-packages (from inkcut)
Requirement already up-to-date: qt-reactor in /usr/local/lib/python3.6/dist-packages (from inkcut)
Requirement already up-to-date: pyqtgraph in /usr/local/lib/python3.6/dist-packages (from inkcut)
Requirement already up-to-date: ipykernel>=4.1 in /usr/local/lib/python3.6/dist-packages (from qtconsole->inkcut)
Requirement already up-to-date: jupyter-client>=4.1 in /usr/local/lib/python3.6/dist-packages (from qtconsole->inkcut)
Requirement already up-to-date: jupyter-core in /usr/local/lib/python3.6/dist-packages (from qtconsole->inkcut)
Requirement already up-to-date: traitlets in /usr/local/lib/python3.6/dist-packages (from qtconsole->inkcut)
Requirement already up-to-date: pygments in /usr/local/lib/python3.6/dist-packages (from qtconsole->inkcut)
Requirement already up-to-date: ipython-genutils in /usr/local/lib/python3.6/dist-packages (from qtconsole->inkcut)
Requirement already up-to-date: hyperlink>=17.1.1 in /usr/local/lib/python3.6/dist-packages (from twisted->inkcut)
Requirement already up-to-date: attrs>=17.4.0 in /usr/local/lib/python3.6/dist-packages (from twisted->inkcut)
Requirement already up-to-date: Automat>=0.3.0 in /usr/local/lib/python3.6/dist-packages (from twisted->inkcut)
Requirement already up-to-date: PyHamcrest>=1.9.0 in /usr/local/lib/python3.6/dist-packages (from twisted->inkcut)
Requirement already up-to-date: constantly>=15.1 in /usr/local/lib/python3.6/dist-packages (from twisted->inkcut)
Requirement already up-to-date: zope.interface>=4.4.2 in /usr/local/lib/python3.6/dist-packages (from twisted->inkcut)
Requirement already up-to-date: incremental>=16.10.1 in /usr/local/lib/python3.6/dist-packages (from twisted->inkcut)
Requirement already up-to-date: qtpy>=1.3 in /usr/local/lib/python3.6/dist-packages (from enaml>=0.10; python_version > "3.0"->inkcut)
Requirement already up-to-date: ply>=3.4 in /usr/local/lib/python3.6/dist-packages (from enaml>=0.10; python_version > "3.0"->inkcut)
Requirement already up-to-date: setuptools in /usr/local/lib/python3.6/dist-packages (from enaml>=0.10; python_version > "3.0"->inkcut)
Requirement already up-to-date: kiwisolver>=1.0.0 in /usr/local/lib/python3.6/dist-packages (from enaml>=0.10; python_version > "3.0"->inkcut)
Requirement already up-to-date: atom>=0.4.2 in /usr/local/lib/python3.6/dist-packages (from enaml>=0.10; python_version > "3.0"->inkcut)
Requirement already up-to-date: numpy in /usr/local/lib/python3.6/dist-packages (from pyqtgraph->inkcut)
Requirement already up-to-date: tornado>=4.2 in /usr/local/lib/python3.6/dist-packages (from ipykernel>=4.1->qtconsole->inkcut)
Requirement already up-to-date: ipython>=5.0.0 in /usr/local/lib/python3.6/dist-packages (from ipykernel>=4.1->qtconsole->inkcut)
Requirement already up-to-date: pyzmq>=13 in /usr/local/lib/python3.6/dist-packages (from jupyter-client>=4.1->qtconsole->inkcut)
Requirement already up-to-date: python-dateutil>=2.1 in /usr/local/lib/python3.6/dist-packages (from jupyter-client>=4.1->qtconsole->inkcut)
Requirement already up-to-date: six in /usr/local/lib/python3.6/dist-packages (from traitlets->qtconsole->inkcut)
Requirement already up-to-date: decorator in /usr/local/lib/python3.6/dist-packages (from traitlets->qtconsole->inkcut)
Requirement already up-to-date: idna>=2.5 in /usr/local/lib/python3.6/dist-packages (from hyperlink>=17.1.1->twisted->inkcut)
Requirement already up-to-date: jedi>=0.10 in /usr/local/lib/python3.6/dist-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut)
Requirement already up-to-date: prompt-toolkit!=3.0.0,!=3.0.1,<3.1.0,>=2.0.0 in /usr/local/lib/python3.6/dist-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut)
Requirement already up-to-date: backcall in /usr/local/lib/python3.6/dist-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut)
Requirement already up-to-date: pickleshare in /usr/local/lib/python3.6/dist-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut)
Requirement already up-to-date: pexpect; sys_platform != "win32" in /usr/local/lib/python3.6/dist-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut)
Requirement already up-to-date: parso>=0.5.2 in /usr/local/lib/python3.6/dist-packages (from jedi>=0.10->ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut)
Requirement already up-to-date: wcwidth in /usr/local/lib/python3.6/dist-packages (from prompt-toolkit!=3.0.0,!=3.0.1,<3.1.0,>=2.0.0->ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut)
Requirement already up-to-date: ptyprocess>=0.5 in /usr/local/lib/python3.6/dist-packages (from pexpect; sys_platform != "win32"->ipython>=5.0.0->ipykernel>=4.1->qtconsole->inkcut)

frmdstryr, 2020-02-26:
Can one or both of you post the output of 

```sh
pip3 freeze | grep jupyter
```

Mine says:
```
jupyter-client==5.3.1
jupyter-core==4.5.0
```

You can try to install that specific version if it's different.

erselbst, 2020-02-26:
On my computer I get these version of jupyter-*

`
jupyter-client==6.0.0
`

`
jupyter-core==4.6.3
`

frmdstryr, 2020-02-26:
Try to downgrade, it's the same issue as https://github.com/jupyter/jupyter_client/issues/270 . 

You may need to downgrade ipykernel as well
```
ipykernel==5.1.2
```

frmdstryr, 2020-02-26:
I can confirm this is broken with jupyter_client 6.0

```
kernel_client.start_channels()
  File "/home/jrm/miniconda3/lib/python3.7/site-packages/qtconsole/kernel_mixins.py", line 47, in start_channels
    super(QtKernelClieEditntMixin, self).start_channels(*args, **kw)
  File "/home/jrm/miniconda3/lib/python3.7/site-packages/ipykernel/inprocess/client.py", line 64, in start_channels
    super(InProcessKernelClient, self).start_channels()
  File "/home/jrm/miniconda3/lib/python3.7/site-packages/jupyter_client/client.py", line 118, in start_channels
    self.control_channel.start()
  File "/home/jrm/miniconda3/lib/python3.7/site-packages/jupyter_client/client.py", line 200, in control_channel
    socket, self.session, self.ioloop
TypeError: ChannelABC() takes no arguments 
```

erselbst, 2020-02-26:
I have downgrade the packages as following:

> pip3 install --upgrade jupyter-client==5.3.1
> pip3 install --upgrade jupyter-core==4.5.0
> pip3 install --upgrade ipykernel==5.1.2

When I start inkcut I get these error (like the same, see logfile).

[inkcut.txt](https://github.com/inkcut/inkcut/files/4258704/inkcut.txt)

erselbst, 2020-02-26:
Ohh wait! When I see the package version, I have the same before:
`
pip3 freeze | grep jupyter

jupyter-client==6.0.0

jupyter-core==4.6.3
`

How can I downgrade the packages?

frmdstryr, 2020-02-26:
Running just `pip3 install jupyter-client==5.3.1` worked for me, try just that command.

erselbst, 2020-02-26:
Okay, now I have downgrade the package but the error still exits.

> pip3 freeze | grep ipykernel
> ipykernel==5.1.2

> pip3 freeze | grep jupyter
> jupyter-client==5.3.1
> jupyter-core==4.5.0

frmdstryr, 2020-02-26:
Here's a full env
[requirements.txt](https://github.com/inkcut/inkcut/files/4258772/requirements.txt)

jlberzal, 2020-02-27:
Now it works. 
Thanks!

closed 2020-02-27 by jlberzal

erselbst, 2020-02-27:
My installation from requirements.txt breaks on git clone -q git@github.com:codelv/inkcut.git ...

> git@github.com: Permission denied (publickey).

Where I can put my ssh public key to use git clone?

erselbst, 2020-02-27:
I have add my public key and now I can install from requirements.txt, but the error on inkcut start still exists.

[inkcut.txt](https://github.com/inkcut/inkcut/files/4260941/inkcut.txt)
[install-out.txt](https://github.com/inkcut/inkcut/files/4260943/install-out.txt)

Should I open a new ticket because it was closed?

frmdstryr, 2020-02-27:
You can just replace that line with just `inkcut`

erselbst, 2020-02-28:
when I start inkcut on cmd with root, it runs, when I start incut as user it chrash. inkcut needs root privileges?

erselbst, 2020-02-28:
Is there a step by step installation manuel also for graphic designer? It seems you have to be a developer to run inkcut ;)