inkcut/inkcut #233: Inkcut fails to start in Ubuntu 18.04

boropaul99, 2020-03-01:
I have installed Inkcut under Ubuntu 18.04 running Python 3.6.9. Inkcut fails to start with the following error messages:

kernel_client.start_channels()
  File "/home/paul/.local/lib/python3.6/site-packages/qtconsole/kernel_mixins.py", line 47, in start_channels
    super(QtKernelClientMixin, self).start_channels(*args, **kw)
  File "/home/paul/.local/lib/python3.6/site-packages/ipykernel/inprocess/client.py", line 64, in start_channels
    super(InProcessKernelClient, self).start_channels()
  File "/home/paul/.local/lib/python3.6/site-packages/jupyter_client/client.py", line 118, in start_channels
    self.control_channel.start()
  File "/home/paul/.local/lib/python3.6/site-packages/jupyter_client/client.py", line 200, in control_channel
    socket, self.session, self.ioloop
TypeError: object() takes no parameters 
[inkcut.txt](https://github.com/inkcut/inkcut/files/4272209/inkcut.txt)

majda107, 2020-03-01:
Same issue on newest arch linux

frmdstryr, 2020-03-01:
See https://github.com/inkcut/inkcut/issues/232#issuecomment-592093490

boropaul99, 2020-03-01:
Fantastic! That has worked for me. Thank you for your fast response.

closed 2020-03-01 by boropaul99