inkcut/inkcut #236: Support GCC Spirit laser cutters

hassanbot, 2020-03-11:
Hi!

I'm a volunteer at a [FabLab](https://www.fablabs.io/) where we have two laser cutters:
1. GCC LaserPro Spirit
2. GCC LaserPro Spirit GLS

They come with a plugin that is currently used with Adobe Illustrator, but it would be great to get it to work with Inkscape since it is free and would be much more accessible to our users.

I've done some googling to try and see what work has been done before, and found these resources:

- [Reversed engineered commands](http://www.wiki.cl.cam.ac.uk/rowiki/CompArch/HardwareLab/LaserCutter) (found through [this](https://github.com/fellesverkstedet/gcc-laserpro-driver) repo)
- [Command implementation](https://github.com/fellesverkstedet/fabmodules/blob/master/src/core/path_gcc.c)

The first source is stating that "The raw data output by the driver appears to be HPGL-2 embedded in a PCL wrapper".

I might try to do this myself but I still wanted to make an issue if anyone wants to help or give some input. I have never worked with low-level commands of machines before, but I can see some similarities between the hpgl protocol file of this repo and the commands in the second source so that might get me started.

hassanbot, 2020-03-24:
I have had to put a hold on this since my Fablab has closed down due to Corona :c

lawrence-jeff, 2022-09-19:
@gris-martin  - Do you have a copy of the referenced repo? I have been working on a Gcode to GCC Laserpro HPGL converter and have been having some success (Goal being to be able to consume Gcode generated by Lightburn) but having some issues with certain operations.

gris-martin, 2022-09-19:
Hi! Which repo do you mean? The links still work for me. Sadly I have moved and no longer have access to the laser cutter, but it's great to see this getting picked up :)

enly1, 2023-04-30:
> @gris-martin - Do you have a copy of the referenced repo? I have been working on a Gcode to GCC Laserpro HPGL converter and have been having some success (Goal being to be able to consume Gcode generated by Lightburn) but having some issues with certain operations.

Has any progress been made with regards the GCC lasers and decoding of the protocol / wrappers being used ?

lawrence-jeff, 2023-04-30:
I got semi-working translation to gcode working that could handle cuts -but I could never figure out their encoding of raster information. Eventually I just converted my machine (Mercury 1) to using a Ruida controller