inkcut/inkcut #237: qtreactor qt5reactor

Franck78, 2020-04-04:
Hello,

for some reason I don't really understand, the module "qtreactor" was no more available after some tries install/uninstall between 2.1.1 & 2.1.2

I ended up changing the file "inkcut/core/manifest.enaml" :

```
import qt5reactor
qt5reactor.install()
```
and then Inkcut loaded again.

I don't know python language but I think it can handle loading a module or another without crashing.

frmdstryr, 2020-04-04:
qt5reactor will work fine. 

Inkcut uses [qt-reactor](https://pypi.org/project/qt-reactor/) because it works with more Qt wrappers (eg PySide2) and older versions of python.

Franck78, 2020-04-04:
yep , messy code to handle python 2.x, 3.4, 3.x and qt qt5 reactor.

https://stackoverflow.com/questions/14050281/how-to-check-if-a-python-module-exists-without-importing-it

But terminating abruptly is not good either if it is just a name mismatch ;)

closed 2020-05-07 by frmdstryr