inkcut/inkcut #238: ModuleNotFoundError: No module named 'atom.compat'

flying4fun, 2020-04-06:
After installing via pip3, I get the following on both Linux (Fedora Core 31) and microsoft 10.
`ModuleNotFoundError: No module named 'atom.compat'`

I cannot find any info about atom.compat anywhere.  Didn't get any errors during the install via pip.

flying4fun, 2020-04-06:
```
$ pip3 list|grep atom
atom               0.5.0

$ pip3 list|grep enaml 
enaml              0.10.4    
enamlx             0.4.1    
```

I assume it's part of enaml since I get the following:

`  File "/usr/local/lib64/python3.7/site-packages/enaml/compat.py", line 11, in <module>
    from atom.compat import *
ModuleNotFoundError: No module named 'atom.compat'
`

chulo601, 2020-04-06:
im having the same problem with ModuleNotFoundError: No module named 'atom.compat' so if you solve the problem can you please post it. thank you

frmdstryr, 2020-04-06:
Run `pip install atom==0.4.3` the new release drops python 2 support and apparently breaks every old version of enaml...

chulo601, 2020-04-06:
thanks @frmdstryr it work for ubuntu but on windows get a error NameError: name 'BREAK_LOOP' is not defined

frmdstryr, 2020-04-06:
You can either downgrade to python 3.7 and reinstall everything (with atom==0.4.3) or if you want to use python 3.8, you'll need to upgrade back to `atom==0.5.0` then install enaml from the master branch on github (or wait until enaml is released).

chulo601, 2020-04-06:
thanks again. i will test it tomorrow to see if it works on windows,

MatthieuDartiailh, 2020-04-06:
The new release of enaml compatible with ataom 0.5.0 is now available on PyPI.

flying4fun, 2020-04-06:
Very good!  Thanks for the quick reply.  I was about to try to start downgrading some of the modules until I got it working.  

I have verified that the updated enaml is indeed compatible with atom 0.5.0 and now inkcut works.  
(pip3 install enaml --upgrade)

Thanks again!

closed 2020-04-06 by flying4fun

chulo601, 2020-04-07:
ok thanks @MatthieuDartiailh

chulo601, 2020-04-07:
I upgrade the enaml on windows an now i get AttributeError: module 'time' has no attribute 'clock' on windinws 10

frmdstryr, 2020-04-07:
Please create a new issue with the full error.