inkcut/inkcut #239: AttributeError: module 'time' has no attribute 'clock'

chulo601, 2020-04-07:
I upgrade the enaml on windows 10 an now i get 
AttributeError: module 'time' has no attribute 'clock'

anyone have fix? 

thanks for any help

frmdstryr, 2020-04-08:
Can you attach a log? I don't see any uses of `time.clock` in Inkcut's code...

chulo601, 2020-04-08:
File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\inkcut\core\plugin.py", line 66, in start_default_workspace
    ui.select_workspace('inkcut.workspace')
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\enaml\workbench\ui\ui_plugin.py", line 157, in select_workspace
    new_workspace.start()
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\inkcut\ui\workspace.py", line 46, in start
    self.workbench.get_plugin('inkcut.ui')
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\enaml\workbench\workbench.py", line 151, in get_plugin
    plugin.start()
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\inkcut\ui\plugin.py", line 87, in start
    self._refresh_dock_items()
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\inkcut\ui\plugin.py", line 181, in _refresh_dock_items
    plugin = workbench.get_plugin(plugin_id)
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\enaml\workbench\workbench.py", line 151, in get_plugin
    plugin.start()
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\inkcut\job\plugin.py", line 62, in start
    if self.job.document:
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\inkcut\job\plugin.py", line 46, in _default_job
    return Job(material=self.material)
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\inkcut\job\plugin.py", line 152, in _observe_material
    job = self.job
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\inkcut\job\plugin.py", line 166, in _refresh_preview
    preview_plugin = self.workbench.get_plugin('inkcut.preview')
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\enaml\workbench\workbench.py", line 144, in get_plugin
    plugin = manifest.factory()
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\inkcut\preview\manifest.enaml", line 23, in plugin_factory
    from .plugin import PreviewPlugin
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\inkcut\preview\plugin.py", line 12, in <module>
    import pyqtgraph as pg
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\pyqtgraph\__init__.py", line 204, in <module>
    from .graphicsItems.VTickGroup import *
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\pyqtgraph\graphicsItems\VTickGroup.py", line 7, in <module>
    from .. import functions as fn
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\pyqtgraph\functions.py", line 17, in <module>
    from . import debug
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\pyqtgraph\debug.py", line 11, in <module>
    from . import ptime
  File "c:\users\appdata\local\programs\python\python38-32\lib\site-packages\pyqtgraph\ptime.py", line 24, in <module>
    cstart = systime.clock()  ### Required to start the clock in windows
AttributeError: module 'time' has no attribute 'clock'

frmdstryr, 2020-04-08:
Do you have pyqtgraph `0.10.0`?

That line was commented out so there must have been an issue at one point:
https://github.com/pyqtgraph/pyqtgraph/blob/8d2c16901b92813f8bebcd6d6cc988fd37bff9f3/pyqtgraph/ptime.py#L24

chulo601, 2020-04-08:
yes i have pyqtgraph 0.10.0

aaujon, 2020-04-10:
I understand that this is caused because pyqtgraph 0.10 doesn't work with python 3, this has been fixed by https://github.com/pyqtgraph/pyqtgraph/commit/73c440a4db6cef93fa47e8aefd7d0f502e281e7e#diff-8272f6892941949b98bbb19cc59d3e0d
So updating to 0.11rc0 should fix this issue I think

artemidya, 2020-05-22:
have the same issue.
there are still no solution?

aaujon, 2020-07-06:
pyqtgraph 0.11 is available, this should be fixed by update it with pip3 install --upgrade pyqtgraph

ink-s75, 2020-07-15:
Hi i have this problem.  pyqtgraph 0.11.0 

self._event_puller = ZMQStream(pipe_in, self.io_loop)
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\zmq\eventloop\zmqstream.py", line 127, in __init__
    self._init_io_state()
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\zmq\eventloop\zmqstream.py", line 546, in _init_io_state
    self.io_loop.add_handler(self.socket, self._handle_events, self.io_loop.READ)
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\tornado\platform\asyncio.py", line 100, in add_handler
    self.asyncio_loop.add_reader(fd, self._handle_events, fd, IOLoop.READ)
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\asyncio\events.py", line 501, in add_reader
    raise NotImplementedError
NotImplementedError

solution?

Dan018, 2020-07-24:
I have the same problem as ink-s75 above. pyqtgraph 0.11.0 is installed.

frmdstryr, 2020-08-19:
Should be fixed by https://github.com/inkcut/inkcut/commit/4b39517c5d113a01ec7d1420fd4514c2ec363f13

closed 2020-08-19 by frmdstryr