inkcut/inkcut #244: Request: svg to hpgl

MrKyr, 2020-05-04:
I was wondering if someone could point me in the right direction for a standalone script or inkcut cli command for svg to hpgl convertor. Thanks in advance

enyachoke, 2020-05-08:
@MrKyr I thin if you select the generic inkscut driver and output to file you can get hpgl

MrKyr, 2020-05-09:
> @MrKyr I thin if you select the generic inkscut driver and output to file you can get hpgl

yes I am aware of that. 

My request is for a command line tool to take two arguments, input file (.svg) and output (.hpgl) based in inkcut's hpgl code.  I don't want to interact with the gui. All i want is a command line tool so I can call it from a custom script, like `svg2hpgl input.svg output.hpgl` or `inkcut input.svg output.hpgl`

mgmax, 2024-05-11:
Inkscape itself can also export to HPGL, without using InkCut:
`inkscape /tmp/a.svg -o /tmp/a.hpgl`