inkcut/inkcut #245: Double linked list corruption in Qt code -> crash

cg9999, 2020-05-07:
Inkcut pretty reliably crashes here during cutting operation with any svg file I've tried.
Inkcut versions tried:
inkcut 2.1.1 (From pypi)
current git master (commit d0e7cf2d65e124cfba00834e960333dc27c663f4)

System is arch linux with latest updates, inkcut is installed in virtualenv, I had to also install pyqt5 from pypi in the virtualenv for the program to start, version PyQt5 5.14.2

Error message varies a little from time to time, but seems to always involve linked list corruption in Qt graphicscene drawing functions

for example: 
...
2020-05-07 13:27:24,019 | DEBUG   | inkcut | -> /dev/ttyUSB0 | PD8504,3207;
2020-05-07 13:27:24,026 | DEBUG   | inkcut | -> /dev/ttyUSB0 | PD8523,3197;
2020-05-07 13:27:24,034 | DEBUG   | inkcut | -> /dev/ttyUSB0 | PD8557,3154;
malloc(): unsorted double linked list corrupted
[1]    21845 abort (core dumped)  ./main.py

some bt:s
```(gdb) bt full
#0  0x00007f2fa6ad1ce5 in raise () from /usr/lib/libc.so.6
#1  <signal handler called>
#2  0x00007f2fa6ad1ce5 in raise () from /usr/lib/libc.so.6
#3  0x00007f2fa6abb857 in abort () from /usr/lib/libc.so.6
#4  0x00007f2fa6b152b0 in __libc_message () from /usr/lib/libc.so.6
#5  0x00007f2fa6b1c74a in malloc_printerr () from /usr/lib/libc.so.6
#6  0x00007f2fa6b1f8fc in _int_malloc () from /usr/lib/libc.so.6
#7  0x00007f2fa6b20fb9 in malloc () from /usr/lib/libc.so.6
#8  0x00007f2fa51ddada in operator new (sz=16) at /build/gcc/src/gcc/libstdc++-v3/libsupc++/new_op.cc:50
        p = <optimized out>
#9  0x00007f2fa1817e3d in QRegion::QRegion(QRect const&, QRegion::RegionType) () from /usr/lib/libQt5Gui.so.5
#10 0x00007f2fa181c87d in QRegion::operator+=(QRect const&) () from /usr/lib/libQt5Gui.so.5
#11 0x00007f2fa1f4007c in QGraphicsViewPrivate::updateRect(QRect const&) () from /usr/lib/libQt5Widgets.so.5
#12 0x00007f2fa1f1671b in ?? () from /usr/lib/libQt5Widgets.so.5
#13 0x00007f2fa1f169c9 in ?? () from /usr/lib/libQt5Widgets.so.5
#14 0x00007f2fa1f17be2 in ?? () from /usr/lib/libQt5Widgets.so.5
#15 0x00007f2fa1f1dd8d in ?? () from /usr/lib/libQt5Widgets.so.5
#16 0x00007f2fa55f5ada in QObject::event(QEvent*) () from /usr/lib/libQt5Core.so.5
#17 0x00007f2fa1f2c4ac in QGraphicsScene::event(QEvent*) () from /usr/lib/libQt5Widgets.so.5
#18 0x00007f2fa24224f3 in sipQGraphicsScene::event(QEvent*) () from /home/cromos/virtualenvs/inkcut/lib/python3.8/site-packages/PyQt5/QtWidgets.abi3.so
#19 0x00007f2fa1be1352 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
#20 0x00007f2fa1bea829 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
#21 0x00007f2fa24fa10e in sipQApplication::notify(QObject*, QEvent*) () from /home/cromos/virtualenvs/inkcut/lib/python3.8/site-packages/PyQt5/QtWidgets.abi3.so
#22 0x00007f2fa55c84f2 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/libQt5Core.so.5
#23 0x00007f2fa55cad56 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /usr/lib/libQt5Core.so.5
#24 0x00007f2fa5621244 in ?? () from /usr/lib/libQt5Core.so.5
#25 0x00007f2fa496f6be in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#26 0x00007f2fa4971531 in ?? () from /usr/lib/libglib-2.0.so.0
#27 0x00007f2fa4971571 in g_main_context_iteration () from /usr/lib/libglib-2.0.so.0
#28 0x00007f2fa5620890 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Core.so.5
#29 0x00007f2fa55c705c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Core.so.5
#30 0x00007f2fa55cf066 in QCoreApplication::exec() () from /usr/lib/libQt5Core.so.5
#31 0x00007f2fa24f8bfd in meth_QApplication_exec_ () from /home/cromos/virtualenvs/inkcut/lib/python3.8/site-packages/PyQt5/QtWidgets.abi3.so
#32 0x00007f2fa681909f in PyCFunction_Call () from /usr/lib/libpython3.8.so.1.0
#33 0x00007f2fa680c3d2 in _PyObject_MakeTpCall () from /usr/lib/libpython3.8.so.1.0
#34 0x00007f2fa68c9c51 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.8.so.1.0
#35 0x00007f2fa68b6a9d in _PyFunction_Vectorcall () from /usr/lib/libpython3.8.so.1.0
#36 0x00007f2fa68c558e in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.8.so.1.0
#37 0x00007f2fa68b6a9d in _PyFunction_Vectorcall () from /usr/lib/libpython3.8.so.1.0
#38 0x00007f2fa68c558e in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.8.so.1.0
#39 0x00007f2fa68b6a9d in _PyFunction_Vectorcall () from /usr/lib/libpython3.8.so.1.0
#40 0x00007f2fa68c558e in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.8.so.1.0
#41 0x00007f2fa68b6a9d in _PyFunction_Vectorcall () from /usr/lib/libpython3.8.so.1.0
#42 0x00007f2fa68c514a in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.8.so.1.0
#43 0x00007f2fa68b58f4 in _PyEval_EvalCodeWithName () from /usr/lib/libpython3.8.so.1.0
#44 0x00007f2fa693cd73 in PyEval_EvalCode () from /usr/lib/libpython3.8.so.1.0
#45 0x00007f2fa693cdc8 in ?? () from /usr/lib/libpython3.8.so.1.0
#46 0x00007f2fa6941063 in ?? () from /usr/lib/libpython3.8.so.1.0
#47 0x00007f2fa67dbdf0 in PyRun_FileExFlags () from /usr/lib/libpython3.8.so.1.0
#48 0x00007f2fa67e5aa4 in PyRun_SimpleFileExFlags () from /usr/lib/libpython3.8.so.1.0
#49 0x00007f2fa694d81e in Py_RunMain () from /usr/lib/libpython3.8.so.1.0
[... 46 lines cut ...]
#39 0x00007f2232d7e05c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#40 0x00007f2232d86066 in QCoreApplication::exec() () at /usr/lib/libQt5Core.so.5
#41 0x00007f222fcafbfd in meth_QApplication_exec_ () at /home/cromos/virtualenvs/inkcut/lib/python3.8/site-packages/PyQt5/QtWidgets.abi3.so
#42 0x00007f2233fd009f in PyCFunction_Call () at /usr/lib/libpython3.8.so.1.0
#43 0x00007f2233fc33d2 in _PyObject_MakeTpCall () at /usr/lib/libpython3.8.so.1.0
#44 0x00007f2234080c51 in _PyEval_EvalFrameDefault () at /usr/lib/libpython3.8.so.1.0
#45 0x00007f223406da9d in _PyFunction_Vectorcall () at /usr/lib/libpython3.8.so.1.0
#46 0x00007f223407c58e in _PyEval_EvalFrameDefault () at /usr/lib/libpython3.8.so.1.0
#47 0x00007f223406da9d in _PyFunction_Vectorcall () at /usr/lib/libpython3.8.so.1.0
#48 0x00007f223407c58e in _PyEval_EvalFrameDefault () at /usr/lib/libpython3.8.so.1.0
#49 0x00007f223406da9d in _PyFunction_Vectorcall () at /usr/lib/libpython3.8.so.1.0
#50 0x00007f223407c58e in _PyEval_EvalFrameDefault () at /usr/lib/libpython3.8.so.1.0
#51 0x00007f223406da9d in _PyFunction_Vectorcall () at /usr/lib/libpython3.8.so.1.0
#52 0x00007f223407c14a in _PyEval_EvalFrameDefault () at /usr/lib/libpython3.8.so.1.0
#53 0x00007f223406c8f4 in _PyEval_EvalCodeWithName () at /usr/lib/libpython3.8.so.1.0
#54 0x00007f22340f3d73 in PyEval_EvalCode () at /usr/lib/libpython3.8.so.1.0
#55 0x00007f22340f3dc8 in  () at /usr/lib/libpython3.8.so.1.0
#56 0x00007f22340f8063 in  () at /usr/lib/libpython3.8.so.1.0
#57 0x00007f2233f92df0 in PyRun_FileExFlags () at /usr/lib/libpython3.8.so.1.0
#58 0x00007f2233f9caa4 in PyRun_SimpleFileExFlags () at /usr/lib/libpython3.8.so.1.0
#59 0x00007f223410481e in Py_RunMain () at /usr/lib/libpython3.8.so.1.0
#60 0x00007f2234104909 in Py_BytesMain () at /usr/lib/libpython3.8.so.1.0
#61 0x00007f2234274023 in __libc_start_main () at /usr/lib/libc.so.6
#62 0x0000562be5d7405e in _start ()
```

frmdstryr, 2020-05-07:
I'm also getting segfaults after updating and dragging the plot around... That code has been unchanged for several years now.  This seems like a bug in the latest Qt/PyQt5

You can try to downgrade to one of the older versions.

cg9999, 2020-05-08:
I'll make some experiments with the versions. I've already tried pyqt5 5.14.1 and 5.14.0, and crash is still there. It's a bit trickier to downgrade the system qt5 as the latest has been rebuilt with new ICU lib so version.

I got an interesting crash with pyqt5 5.14.0:
2020-05-08 13:15:00,290 | DEBUG   | inkcut | -> /dev/ttyUSB0 | PD8523,3197;
2020-05-08 13:15:00,298 | DEBUG   | inkcut | -> /dev/ttyUSB0 | PD8557,3154;
Traceback (most recent call last):
  File "/home/cromos/virtualenvs/inkcut/lib/python3.8/site-packages/pyqtgraph/graphicsItems/ViewBox/ViewBox.py", line 298, in prepareForPaint
  File "/home/cromos/virtualenvs/inkcut/lib/python3.8/site-packages/pyqtgraph/graphicsItems/ViewBox/ViewBox.py", line 886, in updateAutoRange
  File "/home/cromos/virtualenvs/inkcut/lib/python3.8/site-packages/pyqtgraph/graphicsItems/ViewBox/ViewBox.py", line 1366, in childrenBounds
  File "/home/cromos/virtualenvs/inkcut/lib/python3.8/site-packages/pyqtgraph/debug.py", line 505, in __new__
NameError: name 'len' is not defined
Unhandled Python exception
[1]    148274 abort (core dumped)  ./main.py

enyachoke, 2020-05-08:
I also was  having similar issues but the crashes were  random with the either a segfault  or thing like some function is not defined.

frmdstryr, 2020-05-08:
Atom and enaml were recently updated... can you try downgrading to `atom == 0.4.3` and `enaml==0.10.3`?

enyachoke, 2020-05-08:
@frmdstryr  Thanks let me try it out.

cg9999, 2020-05-08:
Sadly I can't test 0.10.4 easily, as arch is now on python 3.8.
I did try to downgrade to atom==0.4.3 and enaml==0.11.0 but it did not help

I'll try more downgrades tomorrow

frmdstryr, 2020-05-28:
There's a bug fix for this at https://github.com/nucleic/enaml/pull/417  The fix seems to be working fine here, can anyone else confirm?

cg9999, 2020-05-28:
@frmdstryr The crash here is gone with the bug fix. Thanks!

frmdstryr, 2020-06-05:
Thanks for verifying! `enaml >= 0.11.2` was released so any new installs / updates should not have this issue.

closed 2020-06-05 by frmdstryr