inkcut/inkcut #246: "No Qt bindings could be found" with pyqt5 installed

CoffinDodger, 2020-05-15:
I had this running on my laptop through WSL and now that i'm trying to run it on my workstation but I'm running into this mess:

```
dingle@IRISHCOFFEE:~$ inkcut
Logging to /home/dingle/.config/inkcut/logs/inkcut.txt
2020-05-14 20:44:42,570 | INFO    | inkcut | ========================================
2020-05-14 20:44:42,570 | INFO    | inkcut | Inkcut launched
2020-05-14 20:44:42,571 | INFO    | inkcut | Version: 2.1.2dev
2020-05-14 20:44:42,571 | INFO    | inkcut | Python: 3.8.2 (default, Apr 27 2020, 15:53:34)
[GCC 9.3.0]
2020-05-14 20:44:42,578 | INFO    | inkcut | System: uname_result(system='Linux', node='IRISHCOFFEE', release='4.4.0-18362-Microsoft', version='#836-Microsoft Mon May 05 16:04:00 PST 2020', machine='x86_64', processor='x86_64')
2020-05-14 20:44:42,578 | INFO    | inkcut | Executable: /usr/bin/python3
2020-05-14 20:44:42,579 | INFO    | inkcut | Args: ['/home/dingle/.local/bin/inkcut']
2020-05-14 20:44:42,584 | ERROR   | inkcut | No Qt bindings could be found
Traceback (most recent call last):
  File "/home/dingle/.local/lib/python3.8/site-packages/qtpy/__init__.py", line 204, in <module>
    from PySide import __version__ as PYSIDE_VERSION  # analysis:ignore
ModuleNotFoundError: No module named 'PySide'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dingle/.local/lib/python3.8/site-packages/inkcut/app.py", line 78, in log_debug_info
    from qtpy import QT_VERSION
  File "/home/dingle/.local/lib/python3.8/site-packages/qtpy/__init__.py", line 210, in <module>
    raise PythonQtError('No Qt bindings could be found')
qtpy.PythonQtError: No Qt bindings could be found
2020-05-14 20:44:42,590 | INFO    | inkcut | ========================================
2020-05-14 20:44:42,596 | ERROR   | inkcut | Traceback (most recent call last):
  File "/home/dingle/.local/lib/python3.8/site-packages/qtpy/__init__.py", line 204, in <module>
    from PySide import __version__ as PYSIDE_VERSION  # analysis:ignore
ModuleNotFoundError: No module named 'PySide'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dingle/.local/lib/python3.8/site-packages/inkcut/app.py", line 97, in main
    from inkcut.core.workbench import InkcutWorkbench
  File "/home/dingle/.local/lib/python3.8/site-packages/inkcut/core/workbench.py", line 17, in <module>
    enamlx.install()
  File "/home/dingle/.local/lib/python3.8/site-packages/enamlx/__init__.py", line 8, in install
    from enamlx.qt import qt_factories
  File "/home/dingle/.local/lib/python3.8/site-packages/enamlx/qt/qt_factories.py", line 8, in <module>
    from enaml.qt import qt_factories
  File "/home/dingle/.local/lib/python3.8/site-packages/enaml/qt/__init__.py", line 8, in <module>
    from qtpy import API as QT_API, PYQT5_API, PYQT4_API, PYSIDE_API, PYSIDE2_API
  File "/home/dingle/.local/lib/python3.8/site-packages/qtpy/__init__.py", line 210, in <module>
    raise PythonQtError('No Qt bindings could be found')
qtpy.PythonQtError: No Qt bindings could be found

Traceback (most recent call last):
  File "/home/dingle/.local/lib/python3.8/site-packages/qtpy/__init__.py", line 204, in <module>
    from PySide import __version__ as PYSIDE_VERSION  # analysis:ignore
ModuleNotFoundError: No module named 'PySide'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dingle/.local/bin/inkcut", line 8, in <module>
    sys.exit(main())
  File "/home/dingle/.local/lib/python3.8/site-packages/inkcut/app.py", line 97, in main
    from inkcut.core.workbench import InkcutWorkbench
  File "/home/dingle/.local/lib/python3.8/site-packages/inkcut/core/workbench.py", line 17, in <module>
    enamlx.install()
  File "/home/dingle/.local/lib/python3.8/site-packages/enamlx/__init__.py", line 8, in install
    from enamlx.qt import qt_factories
  File "/home/dingle/.local/lib/python3.8/site-packages/enamlx/qt/qt_factories.py", line 8, in <module>
    from enaml.qt import qt_factories
  File "/home/dingle/.local/lib/python3.8/site-packages/enaml/qt/__init__.py", line 8, in <module>
    from qtpy import API as QT_API, PYQT5_API, PYQT4_API, PYSIDE_API, PYSIDE2_API
  File "/home/dingle/.local/lib/python3.8/site-packages/qtpy/__init__.py", line 210, in <module>
    raise PythonQtError('No Qt bindings could be found')
qtpy.PythonQtError: No Qt bindings could be found
```
Verified my resources:

```
dingle@IRISHCOFFEE:~$ sudo apt-get install python3-pip python3-pyqt5 python3-setuptools libcups2-dev python3-pyqt5.qtsvg
[sudo] password for dingle:
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-setuptools is already the newest version (45.2.0-1).
python3-pip is already the newest version (20.0.2-5ubuntu1).
python3-pyqt5 is already the newest version (5.14.1+dfsg-3build1).
python3-pyqt5.qtsvg is already the newest version (5.14.1+dfsg-3build1).
libcups2-dev is already the newest version (2.3.1-9ubuntu1.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
Aaaaaaann.......d my Ubuntu version:

```
dingle@IRISHCOFFEE:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04 LTS
Release:        20.04
Codename:       focal
```
so really wondering what's up.  I'm suspecting something may need to be downgraded or some other issue, I just don't know what.

frmdstryr, 2020-05-15:
It looks like system and user libraries are being mixed, either run `pip3 install --user PyQt5` or install inkcut  into the system with `sudo pip3 install inkcut`

CoffinDodger, 2020-05-16:
I uninstalled inkcut, reinstalled with ```sudo pip3 install inkcut``` got the same errors with different paths.  

```
dingle@IRISHCOFFEE:~$ sudo inkcut
Logging to /root/.config/inkcut/logs/inkcut.txt
2020-05-16 16:09:14,626 | INFO    | inkcut | ========================================
2020-05-16 16:09:14,626 | INFO    | inkcut | Inkcut launched
2020-05-16 16:09:14,626 | INFO    | inkcut | Version: 2.1.1
2020-05-16 16:09:14,626 | INFO    | inkcut | Python: 3.8.2 (default, Apr 27 2020, 15:53:34)
[GCC 9.3.0]
2020-05-16 16:09:14,633 | INFO    | inkcut | System: uname_result(system='Linux', node='IRISHCOFFEE', release='4.4.0-18362-Microsoft', version='#836-Microsoft Mon May 05 16:04:00 PST 2020', machine='x86_64', processor='x86_64')
2020-05-16 16:09:14,633 | INFO    | inkcut | Executable: /usr/bin/python3
2020-05-16 16:09:14,633 | INFO    | inkcut | Args: ['/usr/local/bin/inkcut']
2020-05-16 16:09:14,641 | ERROR   | inkcut | No Qt bindings could be found
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/qtpy/__init__.py", line 204, in <module>
    from PySide import __version__ as PYSIDE_VERSION  # analysis:ignore
ModuleNotFoundError: No module named 'PySide'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/inkcut/app.py", line 78, in log_debug_info
    from qtpy import QT_VERSION
  File "/usr/local/lib/python3.8/dist-packages/qtpy/__init__.py", line 210, in <module>
    raise PythonQtError('No Qt bindings could be found')
qtpy.PythonQtError: No Qt bindings could be found
2020-05-16 16:09:14,649 | INFO    | inkcut | ========================================
2020-05-16 16:09:14,660 | ERROR   | inkcut | Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/qtpy/__init__.py", line 204, in <module>
    from PySide import __version__ as PYSIDE_VERSION  # analysis:ignore
ModuleNotFoundError: No module named 'PySide'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/inkcut/app.py", line 97, in main
    from inkcut.core.workbench import InkcutWorkbench
  File "/usr/local/lib/python3.8/dist-packages/inkcut/core/workbench.py", line 16, in <module>
    enamlx.install()
  File "/usr/local/lib/python3.8/dist-packages/enamlx/__init__.py", line 8, in install
    from enamlx.qt import qt_factories
  File "/usr/local/lib/python3.8/dist-packages/enamlx/qt/qt_factories.py", line 8, in <module>
    from enaml.qt import qt_factories
  File "/usr/local/lib/python3.8/dist-packages/enaml/qt/__init__.py", line 8, in <module>
    from qtpy import API as QT_API, PYQT5_API, PYQT4_API, PYSIDE_API, PYSIDE2_API
  File "/usr/local/lib/python3.8/dist-packages/qtpy/__init__.py", line 210, in <module>
    raise PythonQtError('No Qt bindings could be found')
qtpy.PythonQtError: No Qt bindings could be found

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/qtpy/__init__.py", line 204, in <module>
    from PySide import __version__ as PYSIDE_VERSION  # analysis:ignore
ModuleNotFoundError: No module named 'PySide'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/inkcut", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/inkcut/app.py", line 97, in main
    from inkcut.core.workbench import InkcutWorkbench
  File "/usr/local/lib/python3.8/dist-packages/inkcut/core/workbench.py", line 16, in <module>
    enamlx.install()
  File "/usr/local/lib/python3.8/dist-packages/enamlx/__init__.py", line 8, in install
    from enamlx.qt import qt_factories
  File "/usr/local/lib/python3.8/dist-packages/enamlx/qt/qt_factories.py", line 8, in <module>
    from enaml.qt import qt_factories
  File "/usr/local/lib/python3.8/dist-packages/enaml/qt/__init__.py", line 8, in <module>
    from qtpy import API as QT_API, PYQT5_API, PYQT4_API, PYSIDE_API, PYSIDE2_API
  File "/usr/local/lib/python3.8/dist-packages/qtpy/__init__.py", line 210, in <module>
    raise PythonQtError('No Qt bindings could be found')
qtpy.PythonQtError: No Qt bindings could be found
```

I'll admit now I'm a Linux virgin so I'm sure there's more commands I should run to thoroughly remove/clean the install.  But running through WSL at least had the advantage of uninstalling ubuntu easily and reinstalling a fresh build.  SO, did that and ran:

```
sudo apt install python3-pip
pip3 install --user PyQt5

// let Linux sort itself out for the rest of the dependencies
sudo apt install python3-pip python3-pyqt5 python3-pyqt5.qtsvg libcups2-dev 
```
and now it works on the following:
- Ubuntu 20.04 through WSL with VcXsrv.  (went this route due to issue #217)

It may also be worth noting that this problem did not persist with Ubuntu 18.04/WSL.  But i had to identitfy the path when starting inkcut as found in the [inkcut forum](https://inkcut.org/t/inkcut-not-working-on-ubuntu-1804/35/)

So now got Inkcut running a hacked Cricut Expression with FreeExpression FW.

Thanks for the help!

closed 2020-05-16 by CoffinDodger