inkcut/inkcut #247: Inkscape extension does not work in Ubuntu 20.04

bianchifan, 2020-05-17:
System: Ubuntu 20.04 upgraded from 19.10
Inkscape 1.0 for 0.92.5 didn't ran any extension

Inkcut is running fine standalone but you can't call it from inkscape, neither marked area, nor complete file
[image]
[image]

aaujon, 2020-05-17:
I get the same result with python >= 3.6, it's related to subprocess : 
`Changed in version 3.6: Popen destructor now emits a ResourceWarning warning if the child process is still running.`
I will try to have a look this week.

jisep32, 2020-05-18:
I'm getting a different error but would this be related?

/usr/lib/python3.8/subprocess.py:942: ResourceWarning: subprocess 9731 is still running
  _warn("subprocess %s is still running" % self.pid,
ResourceWarning: Enable tracemalloc to get the object allocation traceback

aaujon, 2020-05-18:
Yes it's the same issue, inkscape is now using python 3 with ubuntu 20 and fedora 32

bianchifan, 2020-05-25:
No resource warnings any more but it still does not run :(
After a few seconds a box flashes for less than 0.1 sec, so only 1 clear pic on a screen recording..
[image]
"Open current document..." is working, please wait

But nothing happens

aaujon, 2020-05-26:
Hmm it's working for me, I can have a look later this week

jisep32, 2020-05-26:
I'm having the same warning flash for less then 0.1 seconds but Inkcut never open's.  Inkcut will run fine through terminal in Ubuntu 20.04.

bianchifan, 2020-05-27:
> Hmm it's working for me, I can have a look later this week

I tried to catch what happens, maybe it can help.
Gnome System utilitie are too slow, so I tried with top
2 sceneries, cut selection and current document, both started with clean inkscape

 selection
- python call 
- three! more inkscape jobs
- python disappears, inkcut appears very, very short (hard to catch)
- 4 inkscape jobs do remain, 3 without noticable CPU usage
[image]

[image]

current document
- python call with a temp document
- python disappeares, inkcut never visible for me due to delays when running SSR
- 2 more inkscape jobs
- 3 inkscape jobs will stay with two of them almost doing nothing
[image]

The fastest video I could catch with both scenes and top/SSR
[simplescreenrecorder-2020-05-26_13.09.31.zip](https://github.com/inkcut/inkcut/files/4687724/simplescreenrecorder-2020-05-26_13.09.31.zip)

aaujon, 2020-05-27:
Hi,
Thanks for you report, can you check the inkcut log (probable under ~/.config/inkcut/logs/inkcut.log)?

bianchifan, 2020-05-27:
Oh.., I forgot to mention, no log was produced, no file inkcut.txt  created.

aaujon, 2020-06-09:
I was finally able to reproduce this issue (I hope), I have the same behaviour when the inkcut path is incorrect. There is now an error message with code 2 if the path is wrong.
I also replaced a deprecated call, can you try with https://github.com/inkcut/inkcut/pull/257 ?

bianchifan, 2020-06-09:
I just tried..

Cut selection:
[image]

Current document:
[image]

aaujon, 2020-06-09:
I just fixed the bad indentation, bad copy/paste .. sorry
For the 'Error while starting inkcut', how did you install inkcut ? if it's not using pip, 'inkcut' binary is probably not in the path (if you use an alias for inkcut it doesn't work) can you try using DEBUG = True and set correct value line 58 and 59 ?

bianchifan, 2020-06-09:
excuse me please but I don't see any difference between old line 72 and new...
inkcut should be in the path..
[image]
Originally it was installed with pip/pypi??
After the Ubuntu upgrade I only copied the changed extension scripts, inkscape itself is still just the same.

bianchifan, 2020-06-09:
Sorry;; I'm handicaped at the moment, I pressed the wrong button :((

aaujon, 2020-07-06:
Can you try the command : whereis inkcut ?

bianchifan, 2020-07-07:
[image]
[image]

commit a32771a0 (2020-05-25): Merge pull request #248 from aaujon/master

Fix #247 : set return code so ResourceWarning is not emited with pyth…
commit ba284a1f (2020-05-25): Fix #247 : set return code so ResourceWarning is not emited with python >3.5