inkcut/inkcut #252: Overcut filter is not working as expected

tangentmonger, 2020-06-05:
For example, selecting an overcut of 1mm and cutting a square:

```
2020-06-05 15:09:00,469 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2020-06-05 15:09:00,483 | INFO    | inkcut | {}
2020-06-05 15:09:00,508 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2020-06-05 15:09:05,610 | DEBUG   | inkcut | device | submit <inkcut.job.models.Job object at 0x7fccedeb4b00>
2020-06-05 15:09:05,615 | INFO    | inkcut | device | Initializing job
2020-06-05 15:09:05,615 | DEBUG   | inkcut | device | init <inkcut.job.models.Job object at 0x7fccedeb4b00>
2020-06-05 15:09:05,617 | DEBUG   | inkcut | device | Path length: 1881.3601591501845
2020-06-05 15:09:08,927 | INFO    | inkcut | device | Connecting to device
2020-06-05 15:09:08,927 | INFO    | inkcut | device | Processing job
2020-06-05 15:09:08,953 | DEBUG   | inkcut | device | connect
2020-06-05 15:09:08,968 | DEBUG   | inkcut | -> Silhouette-CAMEO-4 | H
2020-06-05 15:09:08,971 | DEBUG   | inkcut | Silhouette-CAMEO-4 | opened
2020-06-05 15:09:08,972 | INFO    | inkcut | device | Working...
2020-06-05 15:09:08,973 | DEBUG   | inkcut |  filter | Running <inkcut.device.filters.overcut.OvercutFilter object at 0x7fccf42095c0> on model
2020-06-05 15:09:08,974 | DEBUG   | inkcut |  filter | Running <inkcut.device.filters.overcut.OvercutFilter object at 0x7fccf42095c0> on polypath
2020-06-05 15:09:08,981 | DEBUG   | inkcut | -> Silhouette-CAMEO-4 | M199,199
2020-06-05 15:09:08,982 | DEBUG   | inkcut | -> Silhouette-CAMEO-4 | D199,599
2020-06-05 15:09:08,984 | DEBUG   | inkcut | -> Silhouette-CAMEO-4 | D599,599
2020-06-05 15:09:08,986 | DEBUG   | inkcut | -> Silhouette-CAMEO-4 | D599,199
2020-06-05 15:09:08,987 | DEBUG   | inkcut | -> Silhouette-CAMEO-4 | D199,199
2020-06-05 15:09:08,988 | DEBUG   | inkcut | -> Silhouette-CAMEO-4 | D199,203
2020-06-05 15:09:08,990 | DEBUG   | inkcut | -> Silhouette-CAMEO-4 | M0,0
2020-06-05 15:09:08,991 | DEBUG   | inkcut | device | finish
2020-06-05 15:09:08,991 | DEBUG   | inkcut | device | disconnect
2020-06-05 15:09:08,991 | DEBUG   | inkcut | Silhouette-CAMEO-4 | closed
```

Cutting the same shape with overcut disabled:
```
2020-06-05 15:10:45,924 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2020-06-05 15:10:45,936 | INFO    | inkcut | {}
2020-06-05 15:10:45,963 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2020-06-05 15:10:49,685 | DEBUG   | inkcut | device | submit <inkcut.job.models.Job object at 0x7fccedeb4b00>
2020-06-05 15:10:49,691 | INFO    | inkcut | device | Initializing job
2020-06-05 15:10:49,692 | DEBUG   | inkcut | device | init <inkcut.job.models.Job object at 0x7fccedeb4b00>
2020-06-05 15:10:49,694 | DEBUG   | inkcut | device | Path length: 1881.3601591501845
2020-06-05 15:10:52,809 | INFO    | inkcut | device | Connecting to device
2020-06-05 15:10:52,810 | INFO    | inkcut | device | Processing job
2020-06-05 15:10:52,811 | DEBUG   | inkcut | device | connect
2020-06-05 15:10:52,824 | DEBUG   | inkcut | -> Silhouette-CAMEO-4 | H
2020-06-05 15:10:52,826 | DEBUG   | inkcut | Silhouette-CAMEO-4 | opened
2020-06-05 15:10:52,827 | INFO    | inkcut | device | Working...
2020-06-05 15:10:52,835 | DEBUG   | inkcut | -> Silhouette-CAMEO-4 | M199,199
2020-06-05 15:10:52,838 | DEBUG   | inkcut | -> Silhouette-CAMEO-4 | D199,599
2020-06-05 15:10:52,839 | DEBUG   | inkcut | -> Silhouette-CAMEO-4 | D599,599
2020-06-05 15:10:52,840 | DEBUG   | inkcut | -> Silhouette-CAMEO-4 | D599,199
2020-06-05 15:10:52,841 | DEBUG   | inkcut | -> Silhouette-CAMEO-4 | D199,199
2020-06-05 15:10:52,845 | DEBUG   | inkcut | -> Silhouette-CAMEO-4 | M0,0
2020-06-05 15:10:52,845 | DEBUG   | inkcut | device | finish
2020-06-05 15:10:52,845 | DEBUG   | inkcut | device | disconnect
2020-06-05 15:10:52,846 | DEBUG   | inkcut | Silhouette-CAMEO-4 | closed
```

Expected outcome: each edge of the square would be a separate cut and extend either side of the (199, 599) points, resulting in a square with nice sharp corners.

Actual outcome: the move and draw commands are unaltered except for the addition of one section of overcut at the very end ("D199,203"). The physical square has rounded corners.

frmdstryr, 2020-06-05:
I think you want to use blade offset? Overcut only modifies the node of each closed path..

[image]

Also I suppose Inkcut should do an initial alignment cut so it knows the starting position of the blade.

tangentmonger, 2020-06-08:
Blade offset helped a lot, thanks! However, I think the issue with overcut still stands; I was expecting something like this: 
[image] 

Currently only the last cut of the shape is being affected by this filter.

Image is from https://www.rolanddga.com/products/vinyl-cutters/camm-1-gr-large-format-cutters/features) and they describe this feature as "Overcutting corners is the process in which the blade travels past the cut line before the blade is lifted, turned and rotated toward the new cut direction in advance of the new cut line. The result is sharp, clean cuts with easy to weed corners."

frmdstryr, 2020-06-08:
I see, thanks for sharing that, it's quite interesting the length Roland goes to get clean cuts like that.

As you found, Inkcut doesn't do it that way. Inkcut's algorithm is mainly to try and fix the start/end points of a closed path from not being completely cut through.

In order to do something like this a new filter needs added. Inkcut also needs support to send commands to rotate the blade and a way to detect which part of the plot is the discard area that can be cut into.  Contributions are welcome!

aaujon, 2020-06-09:
That would be a nice feature but I don't think we can detect the discard area, the only way would be to have the user select them all, which can be really painfull.
@tangentmonger do you know how other software handle that ?
I guess we can code the overcut to always cut into the inner or outer area for all the shapes, of course it doesn't work for all drawing but it can help.

frmdstryr, 2020-06-24:
I was thinking about maybe adding this at some point. I'm curious as to what the use case is? Are you trying to make PCB bolder mask stencils or something?

tangentmonger, 2020-07-04:
I'm making solder paste stencils. The parts I'm working with have pads around 0.5mm across (APA102–2020 LEDs) so anything to increase the accuracy and corner sharpness is helpful. That said, too much overcut would slice through the walls of the stencil :(

frmdstryr, 2020-07-04:
Interesting you mention that, I just tried to make a stencil last week on some transparency film and had issues as well. What material are you using?

 In my case I think the biggest issue is the transparency sheet folds up when cutting in the push direction leading to diagonal bottom corners. Its a Roland Stika SV-15.

[image]

The ic is a htssop-24 with 0.65 pitch leads so I don't think there's a chance it'll work on those..

[image]