inkcut/inkcut #255: testing by layer filtering

Franck78, 2020-06-06:
```
  File "/usr/lib64/python3.6/site-packages/enaml/core/expression_engine.py", line 186, in read
    return pair.reader(owner, name)
  File "/usr/lib64/python3.6/site-packages/enaml/core/standard_handlers.py", line 101, in __call__
    return call_func(func, (tr,), {}, scope)
  File "/usr/lib/python3.6/site-packages/inkcut-2.1.2.dev0-py3.6.egg/inkcut/job/view.enaml", line 389, in <module>
    icon << color_icon(color_filter.color)
  File "/usr/lib/python3.6/site-packages/inkcut-2.1.2.dev0-py3.6.egg/inkcut/core/utils.py", line 82, in color_icon
    pixmap.fill(get_cached_qcolor(color))
  File "/usr/lib64/python3.6/site-packages/enaml/qt/q_resource_helpers.py", line 200, in get_cached_qcolor
    qcolor = color._tkdata
AttributeError: 'NoneType' object has no attribute '_tkdata'
Abandon (core dumped)

```

Hello,
I commented line 82 and it's working same as what I did (except shifting) . Good.
And on my Opensuse leap, still the qtreactor not found error

```
    #: Install twisted support into the Qt event loop
    import qt5reactor
    qt5reactor.install()

```

frmdstryr, 2020-06-06:
Thanks for testing it out. I pushed a change that should prevent the error. It seems like the fill or stroke couldn't be parsed. 

Could you paste the style attributes or attach the file that cause the error?

Franck78, 2020-06-06:
[test.zip](https://github.com/inkcut/inkcut/files/4740035/test.zip)

frmdstryr, 2020-06-06:
Ahh, `fill:none` so the error make sense as it needs to handle "transparent".  

Thanks!

Franck78, 2020-06-06:
Ok, fix applied.

As you have 'test.zip', you see two layers. "Calque 1" & "Fond".
"Fond" shows nothing because Inkcut does not uses what is in that layer. An image.

This is another filter to add, when no items are of use to Inkcut. But I don't really know what I'm looking for to write it.

Franck78, 2020-06-06:
2020-06-06 22:24:52,228 | DEBUG   | inkcut | Subpath count: 5039
2020-06-06 22:24:54,230 | WARNING | inkcut | Shortest path search aborted (time limit reached)
2020-06-06 22:24:54,237 | DEBUG   | inkcut | Shortest path search: Saved 608.4707871688737 in of movement in -2.001934289932251
2020-06-06 22:24:54,249 | INFO    | inkcut | {'type': 'update', 'object': <inkcut.job.models.Job object at 0x7fbd3b42cd08>, 'name': 'model', 'oldvalue': <PyQt5.QtGui.QPainterPath object at 0x7

duration = time() - now is better ;)
& round number ?

frmdstryr, 2020-06-07:
It now shows "none" as transparent and thanks for spotting the error in the duration!

As far as filtering out unsupported nodes I'm not sure what you mean here? It already ignores them so filtering them won't really do anything.

Franck78, 2020-06-07:
> As far as filtering out unsupported nodes I'm not sure what you mean here?

In my example (test.zip), the Layer "Fond" (visible or not) contains an image, nothing Inkcut can do something with. Just append the Layers if it has a supported node.

Franck78, 2020-09-10:
OK,
starting retesting Inkcut with the 'origin/master' code.

* I miss my "shift X or Y" layers functionality

closed 2020-09-10 by Franck78

frmdstryr, 2020-09-10:
See #270