inkcut/inkcut #256: Quick switch between plotter/cutter

Franck78, 2020-06-06:
Hi frmdstryr ,

Sound silly but probably  you some can (I do) use the plotter to plot or cut.

There is one absolute mandatory setting to change every time, the offset of the blade (to have square corners).

Because it is hidden, not easily memorized (the value), etc etc etc it would be cool if:

A big always visible switch allows changing setup from "Cutter<=>Plotter" profile

Perhaps also some other settings can be swapped (speed?), tool to use...

frmdstryr, 2020-06-08:
If I remember correctly the code is already there for this but the user interface needs changed to support this.

frmdstryr, 2020-06-08:
Actually I think its already possible to so this using the dropdown in the monitor tab (though there probably should be a more obvious way to change devices)

Franck78, 2020-06-08:
I think so, but not sure.
The essential param for my problem is in a Filter

There is Class for the parameters (Offset,Angle) , Instanciated  (config=Intanciate(params) )
in another Filter class, filter registered with the Devive class.....

Problem, I don't understant of to pass a param from to the filter to select between some config.

frmdstryr, 2020-07-25:
@Franck78 I added this. The device setup now supports saving multiple device configurations and you can select one by either choosing it in the settings or selecting it in the Control or Monitor rabs. Please give it a try.

[image]

frmdstryr, 2020-08-04:
Please create a new issue if you find any with this. Thanks for the suggestion!

closed 2020-08-04 by frmdstryr

Franck78, 2020-08-04:
I will use inkcut soon. Just ordering tasks ;)

Franck