inkcut/inkcut #259: Kubuntu 20.04 plug-in inkscape>Inkcut error

Danieleeffe1, 2020-06-23:
Traceback (most recent call last):
  File "inkcut_cut.py", line 27, in <module>
    optparse_spec = importlib.util.find_spec("optparse")
AttributeError: module 'importlib' has no attribute 'util'

sorry for my english
if you need more information please tell me how to get it

frmdstryr, 2020-06-23:
If the Inkscape version is less than 1.0 you need to use the old plugin from https://inkscape.org/gallery/item/12796/inkcut-for-inkscape-2.1.1.tar.gz

aaujon, 2020-07-06:
Actually, it should work, what is your python version ? (python -V from a terminal)
I found this issue that can be related https://bugs.python.org/issue31318, it refers to windows but I guess this might be the same problem

raboof, 2020-09-24:
> If the Inkscape version is less than 1.0 you need to use the old plugin

I see this error even under Inkscape 1.0 - #260 fixes it for me, too, though.

PR #260 (2020-09-24): Fix 259 : import submodule directly
Try to fix #259
closed 2020-09-24 by frmdstryr