inkcut/inkcut #261: Serial Port difficulties

ithinkido, 2020-07-08:
I would like to connect to a HP pen plotter that uses HPGL. I am using Ubuntu 20.04 and have installed Inkcut using the instructions as per https://www.codelv.com/projects/inkcut/download/ 
I have used the plotter over Putty, sending HPGL commands from the same machine. I am fairly confident that I do not have any cable / hardware / port issues as this same setup works when sending over Putty.
My issue is that when trying to connect using Inkcut I am unable to make a connection to the plotter. The correct port is discovered,

[image]

But the device remains diconnected as can be seen in the Monitor window

[image]

The device is not listed in the driver list, so I am using the Generic Inkcut driver and sending HPGL protocal

[image]

Could you tell me if there is something else I need to do to set up a connection here

frmdstryr, 2020-07-09:
Can you please attach a log from Under Help -> Logs?

Does anything happen when you click the connect button in the monitor tab?  Are you sure the port is not in use?

ithinkido, 2020-07-09:
Sure, 
[inkcut.txt](https://github.com/inkcut/inkcut/files/4896648/inkcut.txt)

> 
> Does anything happen when you click the connect button in the monitor tab? 
>
On the GUI, no, but there are some error messages being thrown in terminal, and I think they are the same as what is in the logs
>
>Are you sure the port is not in use?
>
Yes, I have double checked that

frmdstryr, 2020-07-09:
It looks like the configuration is not being saved/set properly so it's attempting to connect to an empty device path. 

Can you go to the Device setup, change the connection type to something else, then change it back to the serial port (to try and force it to update)?

If that doesn't work please attach the `inkcut.device.json` config file (one folder up from the log folder).

ithinkido, 2020-07-09:
By switching between serial and parallel and back it now connects perfect. Thanks for the help

closed 2020-07-09 by ithinkido