inkcut/inkcut #265: Forum email activation broken

risacher, 2020-08-10:
Sorry this is the wrong place to post this, but can't post to the forum.

I tried to register for the forum, but I never get the activation email.  My mail server (exim4) logs the following: 

2020-08-10 16:02:39 TLS error on connection from codelv.com (mail.codelv.com) [144.202.2.236] (gnutls_handshake): A TLS fatal alert has been received.

frmdstryr, 2020-08-10:
Thanks for reporting this. I manually activated your account. 

Still seems to be a problem with the mail server I need to look into.

moaoci, 2020-12-26:
Hello,
I also found no place to report an activation problem on Inkcut Forum page. On the request activation link page, the Send link button produces this error

'Unknown error has occured'.

moaoci, 2020-12-31:
Hello,  "send link" button is responding "Activation link was sent to (my email)" but no email was sent,  many tries on different time.

frmdstryr, 2020-12-31:
Sorry for the delay... I activated your account.