inkcut/inkcut #266: Add Support for: KnK Zing Air

DeeEmm, 2020-08-13:
Manufacturer : KnK (Klic-n-Kut)
Model name : Zing 
Model Number: Z1450
Supported protocols : HPGL
Supported connection types : Serial - Baud 57600
Connection Command : ZG;
Width and length of the plotting area : W 450mm / L Unlimited

I have managed to get the printer working however driver installation needs to be carried out manually via CUPs

- https://inkcut.org/t/knk-zing-compatibility/40/#post-253
- https://deeemm.com/index.php/entry/general/using-inkcut-to-control-a-klicknkut-zing-air-vinyl-cutter

/DM

DeeEmm, 2020-08-15:
Still trying to work out device output settings for correct orientation of preview / live / job output.  Preview & live displays do not match each other and neither match the device output. Will raise a seperate issue for this.

pReya, 2021-02-17:
@DeeEmm Any progress or results in this area? Have the same plotter and would really love to use Inkcut.

rockumk, 2021-02-18:
If and when this works, will it also work for the original "non-air" version?

DeeEmm, 2021-03-23:
Hi all, sorry for the delay in replying, the notification ended up in my junk folder. :(

I've had the machine working for some time now. It will work without the CUPS driver by using serial direct + the generic HPGL / Inkcut driver. 

@rockumk  I'm not sure about a non-air version as I only have the air machine, but if it shares controllers then I cannot see why not.

The most important aspect of getting the machine working was the ZG; command after connection. Then it appears to use standard HPGL commands.

The issue I reported in the linked material above relating to preview / orientation issues may be simple because I was trying to emulate the original MTC software which has the origin at the bottom right of the work. This is non-standard. I tried to emulate the MTC layout as my workflow was setup for this. In retrospect it's very likely possible to overcome the orientation / preview issue I found by simply changing the origin of workpiece and then setting up the machine layout again.

I have not tried this change myself, but I cannot see why it would not work.

/DM

pReya, 2021-03-23:
This issue ranks pretty high on Google regarding the KNK Zing Air. So just as an FYI: You can use the KNK Cutters like the Air with Inkscape directly via the included "Extensions -> Export -> Plot" plugin. This does not offer as many possibilities and options as Inkcut, but it is a nice way to cut very basic things.