inkcut/inkcut #267: No File / Device menus on OSX

DeeEmm, 2020-08-13:
The File / Device menus are missing on OSX this makes it very difficult to use. 

This issue was also present on the beta version as well. It is also present on both machines I own

Running Catalina 10.15.5 / InkCut 2.1.1 / Python 3.8.5 

I do get a warning in the terminal on startup...

`Warning: QApplication was created before pyqtgraph was imported; there may be problems (to avoid bugs, call QApplication.setGraphicsSystem("raster") before the QApplication is created).`

[image]

/DM

frmdstryr, 2020-08-16:
Can you please try a potential fix for me:

1. In Inkcut, open the "Console" tab
2. Paste in the following code

```python
ui = workbench.get_plugin('enaml.workbench.ui')
ui.window.children[0].proxy.widget.setNativeMenu(False)
```
3. Do the menu's now show?

DeeEmm, 2020-08-16:
I get the following error in response:

`AttributeError: 'QMenuBar' object has no attribute 'setNativeMenu'`

frmdstryr, 2020-08-16:
Oh, sorry it should be 

```python
ui = workbench.get_plugin('enaml.workbench.ui')
ui.window.children[0].proxy.widget.setNativeMenuBar(False)
```

DeeEmm, 2020-08-16:
Yes, that works 👍

frmdstryr, 2020-08-16:
Does it show the menus?

DeeEmm, 2020-08-16:
Yes, the menus now show...

[image]

frmdstryr, 2020-08-16:
Great! Can you please try out #269, then i'll merge it.

DeeEmm, 2020-08-16:
No, that doesn't work. The menu is not shown.

If I issue the command above it then shows.

Version os showing as '2.1.2dev' in terminal when I launch. I'm not 100% sure if I've managed to install the branch correctly. I used the following command...

`sudo pip3 install git+https://github.com/codelv/inkcut.git@fix-osx-menubar
`

frmdstryr, 2020-08-16:
Hmm, the command looks correct. Can you open a terminal and run:

```bash
python -c "import inspect; import inkcut.ui.plugin; print(inspect.getsource(inkcut.ui.plugin))"
```

And see if it has the change?

DeeEmm, 2020-08-16:
I get the following error

`Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named inkcut.ui.plugin`

frmdstryr, 2020-08-16:
Try using `python3` instead of `python`

DeeEmm, 2020-08-16:
This is the output. It appears to end in an exception.

`/usr/local/lib/python3.8/site-packages/atom/scalars.py:228: FutureWarning: Unicode is deprecated and will be removed in atom 0.6.0, use Str instead.
  warnings.warn(FutureWarning(msg))
"""
Copyright (c) 2017-2019, Jairus Martin.

Distributed under the terms of the GPL v3 License.

The full license is in the file LICENSE, distributed with this software.

Created on Jul 12, 2015

@author: jrm
"""
import enaml
import pkg_resources
from datetime import datetime
from atom.api import Atom, Int, List, Unicode, Instance, Bool, Enum, Dict
from enaml.qt.q_resource_helpers import get_cached_qicon
from enaml.layout.api import AreaLayout, DockBarLayout, HSplitLayout
from enaml.application import timed_call

from inkcut.core.api import Plugin, DockItem, log
from inkcut.core.utils import load_icon
from . import extensions

with enaml.imports():
    from enaml.stdlib.dock_area_styles import available_styles

ALL_STYLES = sorted(['system']+available_styles())

class Clock(Atom):
    """ A clock so widgets can observe each field as required. """
    year = Int()
    month = Int()
    day = Int()
    hour = Int()
    minute = Int()
    second = Int()
    running = Bool(True)
    now = Instance(datetime, factory=lambda: datetime.now())

    def _observe_running(self, change):
        if self.running:
            timed_call(0, self.tick)

    def _observe_now(self, change):
        t = self.now
        self.year, self.month, self.day = t.year, t.month, t.day
        self.hour, self.minute, self.second = t.hour, t.minute, t.second
        if self.running:
            timed_call(1000, self.tick)

    def tick(self):
        self.now = datetime.now()

class InkcutPlugin(Plugin):
    #: Project site
    wiki_page = Unicode("https://www.codelv.com/projects/inkcut")

    #: For anything that needs to update every second
    clock = Instance(Clock, ())

    #: Dock items to add
    dock_items = List(DockItem)
    dock_layout = Instance(AreaLayout)
    dock_style = Enum(*reversed(ALL_STYLES)).tag(config=True)

    #: Settings pages to add
[... 478 lines cut ...]
    # Utility methods
    # -------------------------------------------------------------------------
    def set_app_name(self):
        """ Set the application name

        """
        ui = self.workbench.get_plugin('enaml.workbench.ui')
        try:
            qt_app = ui._application._qapp
            qt_app.setApplicationName('inkcut')
            if hasattr(qt_app, 'setApplicationDisplayName'):
                qt_app.setApplicationDisplayName('Inkcut')
        except Exception as e:
            log.error('Failed to set app name: {}'.format(e))

    def set_window_icon(self):
        """ Set the main application window icon

        """
        ui = self.workbench.get_plugin('enaml.workbench.ui')
        try:
            icon = get_cached_qicon(load_icon('logo'))
            ui.window.proxy.widget.setWindowIcon(icon)
        except Exception as e:
            log.error('Failed to set window icon: {}'.format(e))`

frmdstryr, 2020-08-16:
Looks like it didn't install it.  Can you try using `sudo python3 -m pip install https://github.com/inkcut/inkcut/archive/fix-osx-menubar.zip`

DeeEmm, 2020-08-16:
Nope, It says "Successfully built inkcut" but when I run inkcut, the menu is still not showing.

DeeEmm, 2020-08-16:
Are there any caches that need clearing?

frmdstryr, 2020-08-16:
Try using `python3 -m inkcut`, it's possible there's multiple versions installed. (eg in the user packages or something). I suppose I should add info with this somewhere in the UI.

DeeEmm, 2020-08-16:
Good call, yes I seem to have several instances of python installed, it's possible I already had python installed when I followed the installation instructions on the website (does OSX ship with python?) 

However, this was not the issue.

I did manage to get it to work but I had to do the following:

`
cd /usr/local/lib/python3.8/site-packages/
sudo pip3 uninstall inkcut  
python3 -m pip install https://github.com/inkcut/inkcut/archive/fix-osx-menubar.zip
`

I think that maybe it was not pulling in the changed file as I was getting 'requirement already satisfied messages when trying to install

DeeEmm, 2020-08-17:
Mac OS X ships with Python 2.7, although it looks like this may stop sometime in the future as it is only included for legacy support.

frmdstryr, 2020-08-17:
I see, thanks for investigating. So was the menu fixed after it updated?

DeeEmm, 2020-08-17:
Yes, inkcut now shows the menus as expected:

[image]

frmdstryr, 2020-08-17:
Great thanks for the help!

DeeEmm, 2020-08-17:
Most welcome. Thanks for the awesome program.

closed 2020-08-17 by DeeEmm

JS3910, 2020-10-24:
> Looks like it didn't install it. Can you try using `sudo python3 -m pip install https://github.com/inkcut/inkcut/archive/fix-osx-menubar.zip`

This link is now a 404 error.