inkcut/inkcut #268: Preview display / Live display / Machine output do not match

DeeEmm, 2020-08-15:
The preview & live displays do not match each other and neither match the device output.

[image]

As you can see the image is aligned to the left of the material in the preview (incorrect for my machine), however the output is aligned to the right (which is correct.) The machine output was correct.

After playing with the various permutations of output orientation in the device setup, I have come to the conclusion that it is not possible to get the preview / live view and output to match and it's not possible to get the software to represent my machine. I can fool it into producing the correct output, but this is not what is represented on the screen

Some observations:

 - Device > Output > Orientation: Mirror X, Mirror Y, Swap X / Y - Changes are not shown in preview but are shown in Live view
 - Device > Output > Scale: Changing scale to negative value (as shown on your webpage) does flip preview image, however the live view shows job progress outside of material boundary.

I can get my machine to plot correctly by using the following settings:

 - Swap x/y: yes
 - Mirror x: yes
 - Mirror y: no
 - Scale x: -1
 - Scale y: 1

Whilst the preview does have the correct page orientation and each item is in the correct location on the page, the orientation of each item is incorrect. The live output, whilst correct, shows the material location incorrectly:

[image]

So I think this is a combination of a few issues with the interface:

 - Orientation changes to the setup are not reflected on the preview but are shown correctly in the live view
 - Scale changes made in the setup are shown correctly on the preview but not in the live view
 - Material location is not correctly shown in the live view (as a result of scale change issue)

The machine settings are also not correctly implemented in other aspects of the user interface:

The margin settings are also not correct: For example with my machine setup the bottom margin changes the left margin in the preview / output...

[image]

The same also applies to the machine control settings: For example with my machine setup, move left feeds paper through the machine.

The device setup needs to be correctly applied to all aspects of device control 

/DM

frmdstryr, 2020-08-16:
Thanks for the detailed report!  

I think the issue is the preview output does not reflect the current device settings.

lawrence-jeff, 2021-01-05:
I am seeing this also
"The same also applies to the machine control settings: For example with my machine setup, move left feeds paper through the machine."

The swap X/Y option does not seem to change the behavior of the control tab jog buttons. They remain inverted regardless of this setting.