inkcut/inkcut #271: PD / PD not working (pen force too weak?)

DeeEmm, 2020-08-29:
Since fixing the menus #267 I have not used my machine. I went to use it today and noticed that the pen-up/pen-down is not working. Assuming it was related to #267 I uninstalled and reverted back to the previous version of Inkcut but found that this was the same. Thinking it might be the machine itself I tried the original MTC software and the PU/PD is working fine.

I tried reinstalling again from master and it's no different. The job runs fine but the cutter does not move down. Occasionally it does move down but it seems almost random. I cannot figure out what's happening. The job has cut fine before.

The cutter does work manually but the force seems very weak, even if I enable the force override and set it to the maximum allowable setting (100.00 g).  It almost seems as if the force is nowhere near strong enough. Maybe this is related.

Pen force in the MTC software is strong, it feels like it might crush your finger. I can happy place my finger under the pen carrier, it easily deflects it.

I'm not entirely sure that this is a bug, perhaps just more a settings issue.  Can I get the pen force limits removed so that I can try and see if this is the issue? Is there another way to input more than 100?

/DM

BTW I'm using the inkcut generic driver with HPGL protocol

DeeEmm, 2020-08-29:
Did some experimenting issuing HPGL commands directly through the serial monitor i.e.

`PF120;` 

Where PF = Pen Force

Seems that for my printer I get a noticeable difference in frequency (PWM noise) and pen force up to a value of 160. Above this it seems that there is no difference. So it would appear that the max for the KNK Zing is a value of 160.

In this case I'm not sure if Pen Force is actually in grams, as the interface suggests, but may be some arbitrary denomination. But at any rate, is there any chance that wan we get the interface limit changed ;) :) 

Alas however it appears that the pen force is not the cause of my issue as manually issuing the command via the serial monitor whilst the job is running does not have any effect. For the moment, the issue still remains a mystery.

Some further info:

I can issue pen up (PU) and pen down (PD) commands and the pen goes up and down. I can issue pen force commends and the force increases. I can see PD & PU commands being sent tot he plotter whilst the job is running, but they seem to have no effect. Something odd is going on.

/DM

frmdstryr, 2020-08-29:
Thanks for investigating. Added in 52f5cf6

frmdstryr, 2020-08-29:
Also Inkcut uses "FS" if the HPGL protocol is used (see https://github.com/inkcut/inkcut/blob/master/inkcut/device/protocols/hpgl.py#L32), I suppose each cutter just uses whatever, you can change it there if you want.

Also (what I do if needed) you can add whatever in the "before job commands" section.

DeeEmm, 2020-08-30:
Sorry I was mistaken, 'FS' is correct (was late last night when I was trying to summarise my findings from memory).

Thanks for the additional info, very useful. I haven't looked at the source (yet), just the interface. Obviously better to get it working without resorting to hacking if possible. I'm sure that's possible, just haven't figured it out yet.

The reference I used was this one - [http://paulbourke.net/dataformats/hpgl/](url) (I'm sure you're already aware of it) . Many of the commands work, some don't, but then I suspect that not all are applicable to my plotter. 

Would be good to get the limit for the force override setting raised or removed. I suspect that the limit of 100 may be incorrect for many machines using HPGL.

I'm still at a loss as to why my pen up/down commands seem to have no effect whilst actually printing, and why they worked fine before, but not now. At the moment I'm trying to set up a driver so that I can send a RAW text file to the printer so that I can experiment with the command set / delimiters. Alas this does not seem to be as easy as I hoped.

Also. one thing that has puzzled me from the start is that the manual controls do not really work as I would expect. They seem to be absolute: i.e. for a step size of 160, it will only move 160 once. I wonder if this is related?

Will keep chipping away at getting to the bottom of this.

/DM

DeeEmm, 2020-08-30:
Another observation is that as I test, printing more and more jobs, the movement has ever increasing pauses in it, much like there is some kind of buffer issue. Power cycling the printer has no effect but when I close and then reopen inkscape the issue resolves and movement is back to normal. I did notice this before when I had a large job set up, towards the end of the job the movement was very jerky. I've noted the comments relating to making sure that the cutting speed does not exceed the  machines capability and this is set okay.

After some research it appears that only the newer machines (orbit) have communications handshaking. The following is taken from the KnK website - 

`Unlike previous Zing models, the ZIng Orbit uses RTS/CTS flow control signals. This means that instead of relying on software to send packets (or chunks) of data at a time, the Zing Orbit asserts when it is ready to receive more data. This creates a faster / more consistent form of communication. As a result, the Zing Orbit can also perform extremely complex jobs without experiencing buffer overruns.`

I'm guessing that the older machines like mine don't have that. I wonder if this is related?

DeeEmm, 2020-08-30:
So after an afternoon of playing with various apps, drivers and trying to sniff serial data I discovered that the issue was a broken head cable. It's a FPC style flat plastic ribbon cable that had simply gone brittle and snapped inside. D'OH! Ahh well. At least the machine is (kinda) fixed now. The pen up / down is working again and I can once again cut. 

It's been an interesting journey and has at least broadened my knowledge a little. Still not sure what the buffer / jerky motion issue is. I'll raise another issue if I can figure out any more info on that.

I've reached out to a fella called Rob Hawkes, who played with the KnK machines before. I'm hoping he might have some more info on the flavour of HPGL that KNK use.  Will update if I find out anything further.

closed 2020-08-30 by DeeEmm

bleughb, 2022-07-02:
> Did some experimenting issuing HPGL commands directly through the serial monitor i.e.
> 
> `PF120;`
> 
> Where PF = Pen Force
> 
> Seems that for my printer I get a noticeable difference in frequency (PWM noise) and pen force up to a value of 160. Above this it seems that there is no difference. So it would appear that the max for the KNK Zing is a value of 160.
> 
> In this case I'm not sure if Pen Force is actually in grams, as the interface suggests, but may be some arbitrary denomination. But at any rate, is there any chance that wan we get the interface limit changed ;) :)
> 
> Alas however it appears that the pen force is not the cause of my issue as manually issuing the command via the serial monitor whilst the job is running does not have any effect. For the moment, the issue still remains a mystery.
> 
> Some further info:
> 
> I can issue pen up (PU) and pen down (PD) commands and the pen goes up and down. I can issue pen force commends and the force increases. I can see PD & PU commands being sent to the plotter whilst the job is running, but they seem to have no effect. Something odd is definitely going on.
> 
> /DM

may be too late to the party here, but the interface in MTC (Make the cut) goes from 1 to 160....The force is adjustable in grams up to around 750g i think....No idea about the linearity or how 1-160 translates to 1-750