inkcut/inkcut #273: Merge point

Franck78, 2020-09-10:
Just noticed a bad thing on the Cutter movements.

It does a lot of "pen UP; pen Down" that probably Inkcut should remove.

It starts within Inkscape. You can draw a path with segment, the ending point of one segment  being the start of the next segment.

Almost invisible in Inkscape. When you spot a 'duo' you can merge the point.

The result of this .svg inside inkcut will be:

* trace to point A
* pen UP to point B
* trace to point C

with a zero displacement between A and B (they are identical) but a real PU PD

IN;PU12146,3849;PD11632,3217;PD11567,3187;PD11508,3176;PD11454,3179;PD11407,3191;PD11334,3226;PD11293,3245;PD11298,2523;PD11298,2479;PD11315,2443;PD11328,2432;PD11345,2429;PD11366,2436;PD11389,2454;PD12401,3555;PD13397,2230;PD13418,2223;PD13437,2223;PD13470,2239;PD13493,2262;PD13503,2279;PD13469,3256;PD13385,3203;PD13341,3183;PD13295,3170;PD13247,3169;PD13197,3182;PD13143,3214;PD13085,3266; ** **PD12620,3849;PU12620,3849;** ** PD13209,4541;PD13252,4576;PD13291,4594;PD13326,4600;PD13357,4596;PD13406,4577;PD13437,4564;PD13433,5269;PD13425,5297;PD13409,5304;PD13377,5293;PD12365,4144;PD11333,5447;PD11299,5458;PD11278,5450;PD11263,5420;PD11261,4462;PD11354,4511;PD11406,4528;PD11461,4535;PD11518,4532;PD11577,4514;PD11636,4480;PD11696,4427;PD12146,3849;

Inkcut should optionally remove those 0 zero moves.

Franck78, 2020-09-10:
I found a way in inkscape to remove the duo points. 

* path union ctrl++
* path separate ctrl+shit+K

assuming you can lost existing combined paths

Franck78, 2020-09-19:
there is something else somewhere !

I'm sure there is no more 'duo' point inside the .svg

Despite that, a lot of them are inserted. See the screen capture. It is just the beginning of the design. It starts ok then bam, tac tac tac tac tac tac.

The debug shows one PDxxx but the Monitor shows very well duplicated PD.

Inkcut says 28 seconds to send (or cut?) the design which is clearly not true with the settings in the cutter machine. Could this be root of a problem ? Some handshaking not good ?

[image]

frmdstryr, 2020-09-19:
Unless the cutters firmware is horrible duplicate points should not have any effect.

Franck78, 2020-09-19:
"Unless the cutters firmware is horrible"
It is safe to assume chineese machine, rapid dumb minimalist software.

Tell me where , in what file I can do the dup check/removal. I have to try.

Franck78, 2020-09-19:
do you think Horizon is KiCad killer app ?

frmdstryr, 2020-09-19:
https://github.com/inkcut/inkcut/blob/master/inkcut/device/plugin.py#L581

Could check for duplicates there or in the process function.

Franck78, 2020-09-19:
At this point, it compares 'real' and this is exceptional when R1==R2

Consequently, my code never detected a duo. The real problem is the speed of sending PDxxxxx
It drops to 4PD/s and after console log clear, climbs to 35PD/s (#274)

``` 
 def move(self, position, absolute=True):
   ......... 
        log.debug(self.position)
        if (self.position == self.last_position):
            log.debug("DUPLICATE move")
            return None

        self.last_position = self.position
        result = self.connection.protocol.move(*position, absolute=absolute)

some debugs
020-09-19 21:46:20,053 | DEBUG   | inkcut | [516.6175649586036, 1321.6615150794687, 1]
2020-09-19 21:46:20,053 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD5860,14993;
2020-09-19 21:46:20,321 | DEBUG   | inkcut | [514.8163115137288, 1323.591533411154, 1]
2020-09-19 21:46:20,321 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD5840,15015;
2020-09-19 21:46:20,596 | DEBUG   | inkcut | [513.4071692893249, 1326.3717394888988, 1]
2020-09-19 21:46:20,597 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD5824,15046;
2020-09-19 21:46:20,865 | DEBUG   | inkcut | [510.032848607744, 1335.8642381355144, 1]
2020-09-19 21:46:20,865 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD5786,15154;

2020-09-19 21:48:15,016 | DEBUG   | inkcut | [1849.4272391835846, 603.5700724684101, 1]
2020-09-19 21:48:15,016 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD20980,6847;
2020-09-19 21:48:15,044 | DEBUG   | inkcut | [1849.9762688901055, 602.5479411130358, 1]
2020-09-19 21:48:15,044 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD20986,6835;
2020-09-19 21:48:15,071 | DEBUG   | inkcut | [1850.9125772874243, 601.9001126200708, 1]
2020-09-19 21:48:15,072 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD20997,6828;
2020-09-19 21:48:15,099 | DEBUG   | inkcut | [1852.1834365655368, 601.6613059145537, 1]
2020-09-19 21:48:15,100 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21011,6825;
2020-09-19 21:48:15,128 | DEBUG   | inkcut | [1853.7361189144387, 601.8662399215231, 1]
2020-09-19 21:48:15,128 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21029,6827;
2020-09-19 21:48:15,156 | DEBUG   | inkcut | [1861.9616373166366, 603.8411869181183, 1]
2020-09-19 21:48:15,156 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21122,6850;
2020-09-19 21:48:15,184 | DEBUG   | inkcut | [1871.4713581440567, 605.4479152670102, 1]
2020-09-19 21:48:15,184 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21230,6868;
2020-09-19 21:48:15,214 | DEBUG   | inkcut | [1876.1504682108211, 605.8028108863169, 1]
2020-09-19 21:48:15,214 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21283,6872;
2020-09-19 21:48:15,242 | DEBUG   | inkcut | [1880.481837376206, 605.6931879689973, 1]
2020-09-19 21:48:15,242 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21333,6871;
2020-09-19 21:48:15,270 | DEBUG   | inkcut | [1884.2425351376498, 604.9948918901512, 1]
2020-09-19 21:48:15,270 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21375,6863;
2020-09-19 21:48:15,298 | DEBUG   | inkcut | [1887.2096309925912, 603.583768024878, 1]
2020-09-19 21:48:15,299 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21409,6847;
2020-09-19 21:48:15,326 | DEBUG   | inkcut | [1916.9349712429798, 579.8020370745855, 1]
2020-09-19 21:48:15,326 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21746,6577;
2020-09-19 21:48:15,364 | DEBUG   | inkcut | [1921.248150125391, 594.0410790032934, 1]
2020-09-19 21:48:15,364 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21795,6739;
2020-09-19 21:48:15,392 | DEBUG   | inkcut | [1925.5866563390257, 603.5180295003158, 1]
2020-09-19 21:48:15,392 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21844,6846;
2020-09-19 21:48:15,421 | DEBUG   | inkcut | [1927.7342949784945, 606.5626442011736, 1]
2020-09-19 21:48:15,422 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21869,6881;
2020-09-19 21:48:15,453 | DEBUG   | inkcut | [1929.8509307909005, 608.5270445311844, 1]
2020-09-19 21:48:15,453 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21893,6903;
2020-09-19 21:48:15,483 | DEBUG   | inkcut | [1931.9241188896208, 609.4479999860393, 1]
2020-09-19 21:48:15,483 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21916,6913;
2020-09-19 21:48:15,517 | DEBUG   | inkcut | [1933.9414143880324, 609.3622800614298, 1]
2020-09-19 21:48:15,518 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21939,6912;
2020-09-19 21:48:15,546 | DEBUG   | inkcut | [1935.8903723995122, 608.3066542530474, 1]
2020-09-19 21:48:15,546 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21961,6900;
2020-09-19 21:48:15,578 | DEBUG   | inkcut | [1937.758548037437, 606.3178920565836, 1]
2020-09-19 21:48:15,579 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD21982,6878;
2020-09-19 21:48:15,608 | DEBUG   | inkcut | [1941.202772646131, 599.6880364821768, 1]
2020-09-19 21:48:15,608 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD22021,6803;
2020-09-19 21:48:15,638 | DEBUG   | inkcut | [1944.174529121131, 589.7668693037408, 1]
2020-09-19 21:48:15,638 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD22055,6690;
2020-09-19 21:48:15,668 | DEBUG   | inkcut | [1946.574258369453, 576.8485464868066, 1]
2020-09-19 21:48:15,668 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD22082,6544;
2020-09-19 21:48:15,698 | DEBUG   | inkcut | [1954.4190975772294, 569.3037126951712, 1]
2020-09-19 21:48:15,698 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD22171,6458;
2020-09-19 21:48:15,728 | DEBUG   | inkcut | [1961.5652195086554, 561.409592775125, 1]
2020-09-19 21:48:15,728 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD22252,6368;
2020-09-19 21:48:15,758 | DEBUG   | inkcut | [1967.3585904805668, 552.8391052854777, 1]
[... cut ...]
2020-09-19 21:48:15,911 | DEBUG   | inkcut | [1995.3704778594827, 524.5118980619027, 1]
2020-09-19 21:48:15,911 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD22636,5950;
2020-09-19 21:48:15,940 | DEBUG   | inkcut | [1995.6575108758848, 522.2190253002711, 1]
2020-09-19 21:48:15,940 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD22639,5924;
2020-09-19 21:48:15,968 | DEBUG   | inkcut | [1990.4487517496332, 520.1830239282247, 1]
2020-09-19 21:48:15,968 | DEBUG   | inkcut | -> /dev/usb/lp0 | PD22580,5901;
2020-09-19 21:48:15,998 | DEBUG   | inkcut | [1983.5823308522802, 519.297844038257, 1]

```